jneilliii/OctoPrint-TPLinkSmartplug #99: temperature failsafe  shutting down for too low temp

overling, 2019-04-06:
Running smoothieware and 2 hotend thermistors. 

tested failsafe to alarm when temp falls below 5 and tested at 10.
Both times it fails with message <5 or <10 and stops print.

set to 0 and printer runs ok.  
Temp inside it ambient reading around 24c.

jneilliii, 2019-04-06:
@overling, what is your issue/request?  Do you want to add the temperature failsafe feature to the plugin so it will automatically turn off the plug when the temperature is below some pre-defined setting?

overling, 2019-04-06:
HI, 
there are two settings, an over temp and an undertemp.

For some reason the undertemp is tripping when its not actually under temp.
for example when i set undertemp to stop process <10c and i start the print it will immediately stop the print in alarm.
But now i set undertemp setting to 0 and the print will run just fine.   
[image]

This started after updating to the previous octoprint release candidate.

jneilliii, 2019-04-06:
My plugin doesn't have those settings, are those in some other plugin you are using to integrate with mine?

overling, 2019-04-06:
OH man, i got linked into your git from a thread in another plugin page by accident.  Go ahead and close this please.  Also, I had to stop using your plugin some months back because it was interfering with another plugin or another plugin was interfering with yours.  I had messaged about it but you i don't know if it ever got fixed. I forgot what it was now.  anyway sorry for the mix up.

PR #107 (2019-05-11): 0.9.16
### Added
- Added HS107 and HS300 multiple plug support, resolves #81.
- Energy data logging and graphing via new tab, resolves #17.
- Temperature runaway protection, resolves #99.
### Fixed
- Change socket receive to grab 1024 chunks until full response is received for larger responses.
- Fixed settings migration issue, resolves #102.
- CSS changes to integrate better with TouchUI, potentially resolves #79.
- Status checks on settings save if plug list changes in anyway.
- Status checks no longer saves the settings in favor of real-time updates to current state, resolves #93.
closed 2019-04-06 by jneilliii