jneilliii/OctoPrint-TPLinkSmartplug #100: Idle timer?

dude2k5, 2019-04-30:
Any chance you could add an idle timer or something similar? I leave prints at work and they finish when im at home, but sometimes other people use the machine after me, so I dont want to put a M code to shut it down every time. However if there was an idle timer that could detect if the printer was in "operational or error/disconnected" and turn off after 30 min of "idle", if it's still in this mode (without any interaction), that would be awesome. Pretty cool plugin regardless. I like.

jneilliii, 2019-04-30:
If you enable the warn while printing option, the gcode triggers will be ignored and any ongoing print will not be interrupted.

jneilliii, 2019-04-30:
If I remember correctly...

closed 2019-04-30 by jneilliii

jneilliii, 2020-03-07:
I know this issue has been closed for some time, but I just pushed updates to my current development branch the enables an idle timeout option. You can install using the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.19.zip

PR #149 (2020-03-15): 0.9.19
Hide tab if there are no energy monitoring capable plugs configured, resolves #145 
Fix countdown related issues, resolves #120 
Fix gcode processing related issues, resolves #148 
Abort power off now clears countdown timers, resolves #144 
Polling moved to server side code, resolves #141, resolves #151 
Added German translations, thanks to @TAOGde
Changed automatic power off option to an idle timeout, resolves #100, resolves #152
Added cooldown option, resolves #146 
Added calculated cost per print job, **highly experimental, and possibly inaccurate**, resolves #150