jneilliii/OctoPrint-TPLinkSmartplug #102: 0.9.16 crash octoprint startup

devildant, 2019-05-01:
Hello, 
after the update of tp link, impossible to restart octoprint, there is a stacktrace at startup
```
2019-05-01 23:25:18,803 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 11, in <module>
    load_entry_point('OctoPrint==1.3.11rc3', 'console_scripts', 'octoprint')()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/__init__.py", line 609, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/cli/server.py", line 206, in serve_command
    ignore_blacklist)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/cli/server.py", line 118, in run_server
    octoprint_server.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/__init__.py", line 478, in run
    extra=dict(plugin=plugin._identifier))
UnboundLocalError: local variable 'plugin' referenced before assignment

```
I try several times, without success, I have downgrader in 0.9.13

devildant, 2019-05-01:
octoprint version 1.3.11rc3

jneilliii, 2019-05-01:
Did you try deleting plugin's settings in config.yaml? None of the error log is referencing my plugin files, so it's hard to debug this one.

devildant, 2019-05-01:
I tried, but nothing to do impossible to start octoprint, I am forced to go into safe mode.
remove tplink 0.9.16, and then restart normally and reinstall 0.9.13.
with the 0.9.13 no problem

devildant, 2019-05-02:
I doubt that it will be easy to fix, I have no more info unfortunately.

jneilliii, 2019-05-02:
Ok, I'm able to reproduce on my end with the upgrade from 0.9.13 to 0.9.16.  I have reverted that release for now and will investigate further.

devildant, 2019-05-02:
I wish you good luck

jneilliii, 2019-05-02:
@devildant, can you try installing the upgrade from the url below and let me know if it gets through for you or not.  It seems to be working for me in my testing.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16a.zip

devildant, 2019-05-02:
hello, 
it's ok good jobs.

I have now detected a JS problem with the latest version of tplink : 
[image]

problem with lib :
 // source: plugin/errortracking/js/sentry.min.js
/*! @sentry/browser 5.0.5 (913eafc4) | https://github.com/getsentry/sentry-javascript */

jneilliii, 2019-05-02:
During upgrade ther plugin was trying to create the sqlite3 database using a getplugindatafolder call,  which isn't available I guess during that phase of loading, so I moved that code to a different section. 

Sentry is part of the new error tracking, not my plugin.

devildant, 2019-05-02:
> During upgrade ther plugin was trying to create the sqlite3 database using a getplugindatafolder call, which isn't available I guess during that phase of loading, so I moved that code to a different section.

ok i see, thx

> Sentry is part of the new error tracking, not my plugin.
yes i know that sentry is not inside tplink plugin but with the 0.9.13 i don't have any error inside the console.
this error only appears if 0.9.16 is installed

jneilliii, 2019-05-02:
Ok, I'll check it out to see if it's the same on my end. May be something that will require @foosel assistance.

jneilliii, 2019-05-03:
Ok, I see what's going on.  Not sure what the sentry reference you posted is, but the errors are on the initial load of the page as the select list of the tab is getting populated. It triggers a change event of the list, which in turn is trying to graph the data.  I just need to add some error checking within that part of the graphing and it should resolve the issue.

jneilliii, 2019-05-03:
Took a slightly different approach on the graphing issue.  In lieu of using change events, added an update button to the tab. This should resolve the js errors.  Give it a try from this link and let me know how it goes. Makes more sense that way as it won't make as many calls back to the database/api side. I'm sure there could still be some improvement to the tab interface in some way.
  
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16.zip

devildant, 2019-05-03:
good job, it works, for the automatic update of the graph, it might be interesting to rely on polling if it is enabled, right?

I also take the opportunity to point out another error that seems to have occurred at random with the previous version, and that I still have today

```
packed_plugins.js?70109bc7:9424 Uncaught TypeError: Cannot read property 'plugins' of undefined
    at tplinksmartplugViewModel.self.onEventSettingsUpdated (packed_plugins.js?70109bc7:9424)
    at packed_core.js?79f3153d:13843
    at Pn (packed_libs.js?ea9fbc9a:11208)
    at Function.<anonymous> (packed_libs.js?ea9fbc9a:11229)
    at packed_core.js?79f3153d:13841
    at DataUpdater.self._ifInitialized (packed_core.js?79f3153d:13886)
    at DataUpdater.self._onEvent (packed_core.js?79f3153d:13746)
    at packed_client.js?68a8d0b0:390
    at Pn (packed_libs.js?ea9fbc9a:11208)
    at Function.<anonymous> (packed_libs.js?ea9fbc9a:11229)
    at OctoPrintSocketClient.propagateMessage (packed_client.js?68a8d0b0:389)
    at packed_client.js?68a8d0b0:484
    at packed_libs.js?ea9fbc9a:11225
    at _t (packed_libs.js?ea9fbc9a:11213)
    at packed_libs.js?ea9fbc9a:11224
    at Function.<anonymous> (packed_libs.js?ea9fbc9a:11229)
    at SockJS.onMessage [as onmessage] (packed_client.js?68a8d0b0:483)
    at SockJS.EventTarget.dispatchEvent (packed_libs.js?ea9fbc9a:22585)
    at packed_libs.js?ea9fbc9a:23298
    at Array.forEach (<anonymous>)
    at SockJS._transportMessage (packed_libs.js?ea9fbc9a:23296)
    at WebSocketTransport.EventEmitter.emit (packed_libs.js?ea9fbc9a:22501)
    at WebSocketTransport.ws.onmessage (packed_libs.js?ea9fbc9a:25372)
```
the error is here : 

```
self.onEventSettingsUpdated = function(payload) {
        			   self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
        		        }
```

jneilliii, 2019-05-03:
If I remember correctly, yes if polling is enabled the graph will update,  also on every percent complete change while printing. 

As for this other error, that's standard function calls documented all over the development examples. I'm not doing anything out of the ordinary there. When the  error happens does it effect functionality in any way?

devildant, 2019-05-03:
> If I remember correctly, yes if polling is enabled the graph will update, also on every percent complete change while printing.
the graph only updates if you click on update, even if polling is enabled

> As for this other error, that's standard function calls documented all over the development examples. I'm not doing anything out of the ordinary there. When the error happens does it effect functionality in any way?

I did not detect any anomaly,  may be just put an if (it's my cracked side of dev :p)
```
self.onEventSettingsUpdated = function(payload) {
          if (self.settings.settings.hasOwnProperty("plugins") && self.settings.settings.plugins.hasOwnProperty("tplinksmartplug") ) 
             self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
}
```

I also found a bug with the graph menu and the octoprint tab (see photo)
[image]

for fix add this just after your div : <div class="row-fluid" id="tplinksmartplug_energy_graph" data-bind="visible: loginState.isUser()"></div>
```
<style>
.js-plotly-plot .plotly .modebar {
 z-index: 100!important;
}
</style>
```

last point promised,
I discovered a little problem with polling.
I wanted to put 0.5 in value, and it is not a good idea because it makes a fork bomb :p
for fix : 
```
if (parseInt(self.settings.settings.plugins.tplinksmartplug.pollingEnabled(),10)) {
				setTimeout(function() {self.checkStatuses();}, (parseInt(self.settings.settings.plugins.tplinksmartplug.pollingInterval(),10) * 60000));
			};

```

after I would like to be able to poll every 10 second :p

I use a lot of your plugin, which is why I bother you a bit.
and then my plugin (shutdown printer) uses tplink :)
in any case thanks for the work

jneilliii, 2019-05-03:
Thanks, I'll look into this tonight.

jneilliii, 2019-05-03:
Thanks for the heads up on the modebar style.  I added that to the css file.  I wouldn't have caught that one because I use my TabOrder plugin and change all the tabs to icons.

[image]

Anyway, I just updated the pre-release 0.9.16 version. The polling and print percentage increase should update the graph now.  It was recording the data but not telling the UI to update.  Install using the below url.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16.zip

devildant, 2019-05-03:
i will test that :) 
you could look for the fork bomb?

devildant, 2019-05-03:
almost, it's plugins which is undefined
try this : 
```
self.onEventSettingsUpdated = function(payload) {
        			if (self.settings.settings.hasOwnProperty("plugins")) {
        				self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
        			}
        		}
```

jneilliii, 2019-05-04:
> you could look for the fork bomb?
Not sure what this is, but I added in your proposed fix.

jneilliii, 2019-05-04:
Just updated the pre-release version again with the onEventSettingsUpdated fix.

devildant, 2019-05-04:
Hi,
The problem of error is more complex than I thought.
it's a problem of asynchronicity.
the onEventSettingsUpdated method must be called just after the method that ends "self.setting ...".
this last method must be asynchronous.
I am looking for a solution, I will send you the final fix

devildant, 2019-05-04:
here below, the implementation that works (tested)
```
		self.onEventSettingsUpdated = function(payload) {
			if (self.settings.settings !== undefined && self.settings.settings.plugins !== undefined) {
				self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
			} 
		}
```

jneilliii, 2019-05-04:
Thanks. I'll update later today as it seems the update graph on percent complete increases aren't working either and I need to figure out why.

devildant, 2019-05-04:
on my side the chart, seems to be updated, each polling

jneilliii, 2019-05-04:
Yeah it's working for polling, but not for print progress.

devildant, 2019-05-04:
ha ok I did not try

jneilliii, 2019-05-04:
Ok, I have the code updated and everything seems to be working now.  I ended up changing the knockout binding around a little, so please give some thorough testing if you get a chance and let me know how it goes.  

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16.zip

devildant, 2019-05-04:
ok, i detected several problem sorry.
1) 
```
packed_core.js?79f3153d:4194 Uncaught TypeError: Cannot read property 'is' of undefined
    at SettingsViewModel.self.isDialogActive (packed_core.js?79f3153d:4194)
    at SettingsViewModel.self.onEventSettingsUpdated (packed_core.js?79f3153d:4733)
    at packed_core.js?79f3153d:13843
    at Pn (packed_libs.js?ea9fbc9a:11208)
    at Function.<anonymous> (packed_libs.js?ea9fbc9a:11229)
    at packed_core.js?79f3153d:13841
    at DataUpdater.self._ifInitialized (packed_core.js?79f3153d:13886)
    at DataUpdater.self._onEvent (packed_core.js?79f3153d:13746)
    at packed_client.js?68a8d0b0:390
    at Pn (packed_libs.js?ea9fbc9a:11208)
    at Function.<anonymous> (packed_libs.js?ea9fbc9a:11229)
    at OctoPrintSocketClient.propagateMessage (packed_client.js?68a8d0b0:389)
    at packed_client.js?68a8d0b0:484
    at packed_libs.js?ea9fbc9a:11225
    at _t (packed_libs.js?ea9fbc9a:11213)
    at packed_libs.js?ea9fbc9a:11224
    at Function.<anonymous> (packed_libs.js?ea9fbc9a:11229)
    at SockJS.onMessage [as onmessage] (packed_client.js?68a8d0b0:483)
    at SockJS.EventTarget.dispatchEvent (packed_libs.js?ea9fbc9a:22585)
    at packed_libs.js?ea9fbc9a:23298
    at Array.forEach (<anonymous>)
    at SockJS._transportMessage (packed_libs.js?ea9fbc9a:23296)
    at WebSocketTransport.EventEmitter.emit (packed_libs.js?ea9fbc9a:22501)
    at WebSocketTransport.ws.onmessage (packed_libs.js?ea9fbc9a:25372)
```
```
        self.isDialogActive = function() {
            return self.settingsDialog.is(":visible");
        };

```

2) Current power stay to 0
[image]

3) if we reload the page (by removing the cache) the status of the plugin is off even if it is on.

devildant, 2019-05-05:
little gift for you, for compatibility with touchui plugin

for this render 
[image]

use this css
```
#touch body #tabs #tab_plugin_tplinksmartplug_link a:before, #touch body #navbar #all_touchui_settings #tab_plugin_tplinksmartplug_link a:before {
    content: "\f080";
}

#touch body #tabs #sidebar_plugin_tplinksmartplug_wrapper_link a:before, #touch body #navbar #all_touchui_settings #sidebar_plugin_tplinksmartplug_wrapper_link a:before {
    content: "\f1e6";
}

#touch body #sidebar_plugin_tplinksmartplug > div.accordion-inner > #tplinksmartplugs > div > div.on {
    color: #FFFFFF;
}

```

for this render : 
[image]

```
#touch body #tabs #tab_plugin_tplinksmartplug_link a:before, #touch body #navbar #all_touchui_settings #tab_plugin_tplinksmartplug_link a:before {
    content: "\f1e6\f080";
}

#touch body #tabs #sidebar_plugin_tplinksmartplug_wrapper_link a:before, #touch body #navbar #all_touchui_settings #sidebar_plugin_tplinksmartplug_wrapper_link a:before {
    content: "\f1e6";
}

#touch body #sidebar_plugin_tplinksmartplug > div.accordion-inner > #tplinksmartplugs > div > div.on {
    color: #FFFFFF;
}
```

devildant, 2019-05-05:
and for button plug
add the span with name plug

```
<a class="pull-right" href="#&quot;" data-toggle="tooltip" data-html="true" data-bind="click: $root.toggleRelay, visible: $root.loginState.loggedIn(), tooltip: {title: label, placement: 'bottom'}" style="float: left;" data-original-title="" title="">
<i class="icon off icon-print" data-bind="css: [currentState(), icon(),(($root.processing().indexOf(ip()) > -1) ? 'icon-spin' : '')].join(' ')"></i>
**<span>plug name</span>**
</a>
```

and add this css :
```
#touch body #navbar .navbar-inner .container .nav li.dropdown#all_touchui_settings #navbar_plugin_tplinksmartplug a:before {
	display: none;
}

#touch body #navbar_plugin_tplinksmartplug > a > i {
	padding-left: 4px;
}

body #navbar_plugin_tplinksmartplug > a > span {
	display:none;
}

#touch body #navbar_plugin_tplinksmartplug > a > span {
	padding-left: 17px;
	display:inline;
}
```

I did not really test, only by hacking the HTML code.
but I can test, if you decide to implement it

jneilliii, 2019-05-05:
Nice @devildant, I'll give that a try.  In the meantime I update the release that reverted the previous changes and should still fix the issue I was trying to solve.  Give it a go and let me know.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16.zip

devildant, 2019-05-05:
it's better, but I still have a problem with the plug state when loading the page.
reproduce:
1) activate the plug
2) refresh the page

the state of the plug changes from green to red.
the ajax checkstatus request returns the current state to "on".
I think the web-based socket has the status "off"

jneilliii, 2019-05-05:
Does the status update properly once a polling interval is hit?

jneilliii, 2019-05-05:
So integrated the touchui changes, but I don't have a 1.3.10 octoprint and as you already know I can't get touchui to load in 1.3.11rc3.  Install from here...

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16c.zip

devildant, 2019-05-05:
> Does the status update properly once a polling interval is hit?
with polling with interval after 1 min  it's ok

devildant, 2019-05-05:
> So integrated the touchui changes, but I don't have a 1.3.10 octoprint and as you already know I can't get touchui to load in 1.3.11rc3. Install from here...
> 
> https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16c.zip

yes i know, i also use the #touch for the moment
the creator of the plugin is not very active, it's been a long time since he does not correct his plugin.
I will test your version right away

devildant, 2019-05-05:
there seems to be a problem with the plug button (it is no longer visible)
```
packed_core.js?79f3153d:15654 Error instantiating tplinksmartplugViewModel : TypeError: ko.observableDictionary is not a function
    at new tplinksmartplugViewModel (https://192.168.1.47/static/webassets/packed_plugins.js?55295a52:11771:36)
    at _createViewModelInstance (https://192.168.1.47/static/webassets/packed_core.js?79f3153d:15569:20)
    at HTMLDocument.<anonymous> (https://192.168.1.47/static/webassets/packed_core.js?79f3153d:15652:41)
    at mightThrow (https://192.168.1.47/static/webassets/packed_libs.js?ea9fbc9a:3535:29)
    at process (https://192.168.1.47/static/webassets/packed_libs.js?ea9fbc9a:3603:12)
```

devildant, 2019-05-05:
otherwise for the icons of the other sidebar / tab it's ok
[image]

devildant, 2019-05-05:
mmm, it seems that the whole viewmodel is in error

devildant, 2019-05-05:
ok, you accidentally delete the file: ko.observableDictionary.js
and I think I found the cause of the plug button status problem, it's this callback : 
```
		self.onEventSettingsUpdated = function(payload) {
			if (self.settings.settings !== undefined && self.settings.settings.plugins !== undefined) {
				self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
			} 
		}
```
I wanted to use it on my plugin to see, and it's a disaster, in this reminder, my self is partially fed.

devildant, 2019-05-05:
I confirm that it is onEventSettingsUpdated which poses problem.
just delete it

devildant, 2019-05-05:
it misses the name in the "span" of the plug (icon)
```
<a class="pull-right" href="#" data-toggle="tooltip" data-html="true" data-bind="click: $root.toggleRelay, visible: $root.loginState.loggedIn(), tooltip: {title: label, placement: 'bottom'}" style="float: left;" data-original-title="" title="">
<i class="icon on icon-print" data-bind="css: [currentState(), icon(),(($root.processing().indexOf(ip()) > -1) ? 'icon-spin' : '')].join(' ')"></i>

<span data-bind-"text:="" label"=""></span>

</a>
```

devildant, 2019-05-05:
here is a small improvement css
```
#touch body #navbar_plugin_tplinksmartplug > a > i {
    padding-left: 1px;
    font-size: 1.25rem!important;
}
```

devildant, 2019-05-05:
if you want to test by yourself, on the page of octoprint F12 and console
and enter this cmd for force open  touch panel:
`
$("#touchui_settings_dialog").modal('show')`

devildant, 2019-05-05:
and i detect 
[image]
I also detected a small problem of responsiveness with the bottom button, but it's not much

jneilliii, 2019-05-05:
Ok, I reverted the onEventSettingsUpdated function back to it's original state because that should work, and I'm not sure why it would be causing errors for you in the first place.  I can't reproduce the error.  I've updated the css a bit for touchui and fixed the span issue.  The graph still doesn't work great on small screens, but it is there.  If the graph ends up being too narrow, it throws a lot of errors from plotly side.  Not much I can do about that, and would actually consider hiding the tab altogether in touchui rather than trying to resolve the issue.

devildant, 2019-05-05:
hello @jneilliii 

I think I have found a better solution.
replace the callback "onEventSettingsUpdated" with "onSettingsBeforeSave"

```
		self.onEventSettingsUpdated = function(payload) {
			if (self.complete == true && self.settings.settings !== undefined && self.settings.settings.plugins !== undefined) {
				self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
			} 
		}
```
become :
```
		self.onSettingsBeforeSave = function(payload) {
			self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
                        self.checkStatuses(); // call check status for get current stat
		}
```

this has several advantages : 
1) this callback, is not calling the loading of the page, the get status is not override
2) this callback is called at the time of the setting's know, so it behaves like onEventSettingsUpdated but without the disadvantages
3) with the call to the checkstatus, it does not seem to be necessary to refresh the page after adding a plug

note: self.settings.settings.plugins.tplinksmartplug.arrSmartplugs() seems to have a plug status always off, that's what was wrong with "onEventSettingsUpdated". except that with this callback it was impossible to call the check status, because this callback was issued several times

devildant, 2019-05-05:
I am currently testing this solution and it seems very good (and error free)

devildant, 2019-05-05:
and if you want hide chart on touchui view use this css :
```

#touch body #tab_plugin_tplinksmartplug_link {
display:none;
}

#touch body #tab_plugin_tplinksmartplug_link2 {
display:none;
}

#touch body #tab_plugin_tplinksmartplug {
display:none;
}
```

jneilliii, 2019-05-05:
Can you do me a favor and test this function instead and let me know how it goes.
```
		self.onSettingsBeforeSave = function(payload) {
                        self.checkStatuses(); // call check status for get current state
			self.settings.settings.plugins.tplinksmartplug.arrSmartplugs(self.arrSmartplugs()); // save settings with current states updated.
		}
```

devildant, 2019-05-05:
hello, 
juste switch position of self.checkStatuses(); ?

devildant, 2019-05-05:
ha ok i see, I think that this poses a problem when updating settings (add plug).
but i will try

jneilliii, 2019-05-05:
It might not work the way I think it should, because of asynchronous calls of checkStatuses.

devildant, 2019-05-05:
it's good that I thought, there is a problem if we add a plug.
just onSettingsBeforeSave  instead of onEventSettingsUpdated  work well

```
		self.onSettingsBeforeSave = function(payload) {
			self.arrSmartplugs(self.settings.settings.plugins.tplinksmartplug.arrSmartplugs());
                        self.checkStatuses(); // call check status for get current stat
		}
```
with this callback the addition of a plug, the status changes to red and then green if the plug is on.
but nothing shocking

and there is no problem loading the page

devildant, 2019-05-05:
in addition this change does not impact the logic of your code, and it is much more stable

devildant, 2019-05-05:
in any case on my side it works really well

jneilliii, 2019-05-05:
I ended up integrating your method with one slight change.  To keep the status checks down I am comparing the length of the two arrays and only checking statuses if they are not the same.  This should pick up on the addition of new plugs and prevent unneeded calls to the server during save operations.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16c.zip

Let me know if you foresee any drawback to that approach.

devildant, 2019-05-05:
perfect.
i changed the css,
so that the size of the icon does not affect the positioning
```
#touch body #navbar_plugin_tplinksmartplug > a > span {
    display: inline;
    position: absolute;
    left: 33px;
    line-height: 34px;
}
```

[17 more replies]

PR #107 (2019-05-11): 0.9.16
### Added
- Added HS107 and HS300 multiple plug support, resolves #81.
- Energy data logging and graphing via new tab, resolves #17.
- Temperature runaway protection, resolves #99.
### Fixed
- Change socket receive to grab 1024 chunks until full response is received for larger responses.
- Fixed settings migration issue, resolves #102.
- CSS changes to integrate better with TouchUI, potentially resolves #79.
- Status checks on settings save if plug list changes in anyway.
- Status checks no longer saves the settings in favor of real-time updates to current state, resolves #93.
closed 2019-05-11 by jneilliii