jneilliii/OctoPrint-TPLinkSmartplug #105: Serial/Print reset when starting

HuguesDug, 2019-05-05:
Hello,

Configuration OctoPrint 1.3.10, TPLinkSmartPlug plugin 0.96, Printer Anet A8 with Marlin 1.1.9 Bug fix.

If I tick the GCode Trigger in the configuration panel of the Plugin, the printer will reset the serial connection and the printer at the beginning of the print (it heats the beg, run home X/Y  and then lost TTY during Z).

If I do not tick the box, the printer will not have problem... however, it will then not turn off when print complete.

What is wrong either in my configuration or in the plugin.

---- Dialog with the printer -----

Send: N8 M107*45
Recv: ok
Send: N9 G28 X0 Y0*27
[...]
Recv: X:-31.00 Y:-8.00 Z:14.14 E:119.43 Count X:-3100 Y:-800 Z:5658
Recv: ok
Send: N10 G28 Z0*104
[...]
Connection closed, closing down monitor
Changing monitoring state from "Printing" to "Offline"
Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyUSB0']
Connecting to: /dev/ttyUSB0
Changing monitoring state from "Detecting serial port" to "Opening serial port"

jneilliii, 2019-05-05:
Do you mean version 0.9.16 of the plugin?  What you describe is unexpected, but the problem could be related to the thermal runaway feature that was added with that version, but it would power off the plug altogether.  The plugin monitors for the commands `M80` `M81` `@TPLINKON` and `@TPLINKOFF` and only reacts to those commands. Are there any errors in your octoprint,log related to the plugin?

HuguesDug, 2019-05-05:
Hello,

I unticked the box "Auto Connect" in the plugin and in Octoprint and it works. 

I read your source code.

 Problem is that you apply the reconnect, even if the printer is already connected. As a consequence, it would resets the TtyUSB if already connected. You should aad a test to do "Connect" only if disconnected. You can do that with GET /api/connection and checking current.state<>"operational".

jneilliii, 2019-05-05:
Thanks, I never encountered an issue with that as I use it to control my printer and not auxiliary devices like lights, etc. I'll investigate and incorporate the change in a future release.

jneilliii, 2019-05-09:
I think I have this worked out now, if you don't mind testing by installing using the url below in plugin manager to confirm that would be helpful.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.16d.zip

HuguesDug, 2019-05-10:
I will as soon as I can... I review your modifications, for me, should do the trick.

HuguesDug, 2019-05-10:
Installed and tested. It works !

jneilliii, 2019-05-11:
Thanks, I just finalized the official 0.9.16 version with a couple of minor changes.

closed 2019-05-11 by jneilliii