jneilliii/OctoPrint-TPLinkSmartplug #106: No longer seems to be working? Toggle switch stays red and just rotates

nirurin, 2019-05-08:
The toggle switch is staying red, and if I click the switch it just spins forever and nothing happens. If I turn the switch on manually (at the plug) and refresh the octoprint page, the toggle is still red. Nothing I do will turn it green. 

Restarted and rebooted the server several times. 

I've uninstalled the plugin and reinstalled it. Seems to be on version .16, even though it says the newest version is .13. Not sure why this is.

jneilliii, 2019-05-09:
That's my bad, development branch was inadvertantly merged into master. Try installing from the link below to get previous release. 

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.13.zip

nirurin, 2019-05-09:
I've only installed plugins through the octoprint menus... how would i
install the update from that zip file?

jneilliii, 2019-05-09:
In octoprint's plugin manager at the bottom of the list there's an option to install from a url. Paste the url into the field and press install.

nirurin, 2019-05-09:
Ahh got it. Installed now. I'm back on .13, but the toggle for the power
still isn't working.

jneilliii, 2019-05-09:
That sounds like maybe your plug's ip address has changed maybe? Occasionally I have some weird issues like this and end up having to unplug the tplink from the wall and plug it back in as well.

closed 2019-05-30 by stale[bot]