jneilliii/OctoPrint-TPLinkSmartplug #108: update to last version

mauwl, 2019-05-19:
I updated to the latest version of tplinksmartplug and now the button close in octoprint no longer works.

By disabling tplinksmartplug the button close works again and if I reactivate the plugin, it does not work again

jneilliii, 2019-05-19:
What button are you talking about?

mauwl, 2019-05-19:
in all menu same problem 
[image]

jneilliii, 2019-05-19:
That's weird, my code shouldn't effect that.  I wonder if something is getting broken due to upgrade and therefore breaking something else.  Could you try to ssh to your pi and run the following command?
```
nano ~/.octoprint/config.yaml
```
In that file, delete the entire tplinksmartplug section under plugins and save the file and then restart octoprint. You'll have to re-add your plug(s) back in.

rrauenza, 2019-05-20:
Ok, maybe unrelated -- but tonight I kept having the edit screens disappear while editing fields -- like the countdown field:

[image]

I'd do a screen record, but I need a screen recorder that shows keystrokes.  I reproduced it once more, and then can't again.  I keep thinking I hit ESC but that key is far away from the keys I'm hitting.  Or its a keyboard glitch.

But if you see it, you're not going crazy.  If it never happens for you, then I'm going crazy.

mauwl, 2019-05-20:
hi,
i delete all line tplinksmartplug in the yaml, i restart octoprint and add again plugin , but problem come again

but, i resolve this problem, i uninstall plugin and take previous version 0.9.13 and no more problem

jneilliii, 2019-05-22:
@mauwl, I'll try to take a look this weekend.  For reference, which Themeify variant are you using and what other plugins are installed?  There could be a conflict from that perspective as well.

hankchillx, 2019-05-23:
@jneilliii I am experiencing the same issue, I believe it may have something to do with the thermal runaway monitoring options, as when it's enabled and values are set, they disappear after a refresh. Something seems to be looking for them as an error appears in the console.

[image]

jneilliii, 2019-05-23:
Thanks for the details @hankchillx, I'll keep that in mind when I start investigating this weekend.

mauwl, 2019-05-23:
hi, here what plugin i have
[image]

closed 2019-06-13 by stale[bot]