jneilliii/OctoPrint-TPLinkSmartplug #109: hs300 isn't working

rrauenza, 2019-05-20:
I'm getting the spinning lightning bolt, can't seem to get logging to work, and not getting any power status.  tcpdump isn't helpful I guess because of the xor encryption.

I know this is a new feature and I'd like to help resolve.

I'm also a little worried that when I got it today I upgraded the firmware...

CC @romainrobotics

rrauenza, 2019-05-20:
The lightning bolt is spinning ... and the "plug to graph" dropdown list is empty.

```
[2019-05-19 20:15:56,649] DEBUG: Checking status of 10.13.54.90/0.
[2019-05-19 20:15:56,650] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-05-19 20:15:56,651] DEBUG: IP 10.13.54.90 is valid.
[2019-05-19 20:15:56,652] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-05-19 20:15:56,652] DEBUG: IP 10.13.54.90 is valid.
[2019-05-19 20:15:56,653] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.13.54.90
[2019-05-19 20:15:56,819] DEBUG: <AB>^@^D<A4>p"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"800671B8B6CC08249B39955E766F0C281B1C1245","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-76,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"800671B8B6CC08249B39955E766F0C281B1C124500","state":1,"alias":"Plug 1","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124501","state":1,"alias":"Plug 2","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124502","state":1,"alias":"Plug 3","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124503","state":1,"alias":"Plug 4","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124504","state":1,"alias":"Plug 5","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124505","state":1,"alias":"Plug 6","on_time":3666,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-19 20:15:56,825] DEBUG: get_device_id response: 800671B8B6CC08249B39955E766F0C281B1C124500
[2019-05-19 20:15:56,825] DEBUG: Sending command {'system': {'get_sysinfo': {}}, 'context': {'child_ids': [u'800671B8B6CC08249B39955E766F0C281B1C124500']}} to 10.13.54.90
[2019-05-19 20:15:56,963] DEBUG: <AB>^@^D<A4>p"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"800671B8B6CC08249B39955E766F0C281B1C1245","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-76,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"800671B8B6CC08249B39955E766F0C281B1C124500","state":1,"alias":"Plug 1","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124501","state":1,"alias":"Plug 2","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124502","state":1,"alias":"Plug 3","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124503","state":1,"alias":"Plug 4","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124504","state":1,"alias":"Plug 5","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124505","state":1,"alias":"Plug 6","on_time":3666,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-19 20:15:56,967] DEBUG: IP 10.13.54.90 is valid.
[2019-05-19 20:15:56,968] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-05-19 20:15:56,969] DEBUG: IP 10.13.54.90 is valid.
[2019-05-19 20:15:56,969] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 10.13.54.90
[2019-05-19 20:15:57,109] DEBUG: <AB>^@^D<A4>p"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"800671B8B6CC08249B39955E766F0C281B1C1245","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-76,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"800671B8B6CC08249B39955E766F0C281B1C124500","state":1,"alias":"Plug 1","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124501","state":1,"alias":"Plug 2","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124502","state":1,"alias":"Plug 3","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124503","state":1,"alias":"Plug 4","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124504","state":1,"alias":"Plug 5","on_time":3666,"next_action":{"type":-1}},{"id":"800671B8B6CC08249B39955E766F0C281B1C124505","state":1,"alias":"Plug 6","on_time":3666,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-19 20:15:57,114] DEBUG: get_device_id response: 800671B8B6CC08249B39955E766F0C281B1C124500
[2019-05-19 20:15:57,115] DEBUG: Sending command {'emeter': {'get_realtime': {}}, 'context': {'child_ids': [u'800671B8B6CC08249B39955E766F0C281B1C124500']}} to 10.13.54.90
[2019-05-19 20:15:57,302] DEBUG: <AB>^@^@l<BC>"emeter":{"get_realtime":{"voltage_mv":122365,"current_ma":118,"power_mw":8486,"total_wh":7,"err_code":0}}}
```

rrauenza, 2019-05-20:
[image]

rrauenza, 2019-05-20:
Linking to original implementation: https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/81

rrauenza, 2019-05-20:
Ok, that's odd.  I added a /1 and that one works.   Oh, ok, I'm an idiot -- it's the delay function on /0.

rrauenza, 2019-05-20:
Changed the delay off to 1 and it doesn't work after 2 minutes.  Looking through the code for 'countdown' ... I don't understand how it works, unless countdown is being sent to the tplink device itself?

btw -- I want the countdown to give the hot end a chance to cool down via the heatbreak fan.  An alternate feature might be a temperature level to wait for before turning off.

rrauenza, 2019-05-20:
When I click to power off neither plugs with countdowns or plugs without countdowns seem to do any logging.    Oh -- maybe this is an error at the js level.  Yes:

```
packed_core.js?8c230bbf:15214 Error calling onDataUpdaterPluginMessage on view model tplinksmartplugViewModel : TypeError: Cannot read property 'currentState' of null
    at tplinksmartplugViewModel.self.onDataUpdaterPluginMessage (http://octopi.private.shroop.net/static/webassets/packed_plugins.js?40a77695:994:20)
    at callViewModelIf (http://octopi.private.shroop.net/static/webassets/packed_core.js?8c230bbf:15201:35)
    at http://octopi.private.shroop.net/static/webassets/packed_core.js?8c230bbf:15150:13
    at Pn (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11208:530)
    at Function.<anonymous> (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11229:66)
    at callViewModelsIf (http://octopi.private.shroop.net/static/webassets/packed_core.js?8c230bbf:15148:7)
    at callViewModels (http://octopi.private.shroop.net/static/webassets/packed_core.js?8c230bbf:15142:5)
    at http://octopi.private.shroop.net/static/webassets/packed_core.js?8c230bbf:13883:13
    at DataUpdater.self._ifInitialized (http://octopi.private.shroop.net/static/webassets/packed_core.js?8c230bbf:13907:13)
    at DataUpdater.self._onPluginMessage (http://octopi.private.shroop.net/static/webassets/packed_core.js?8c230bbf:13882:14)
    at http://octopi.private.shroop.net/static/webassets/packed_client.js?68a8d0b0:390:17
    at Pn (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11208:530)
    at Function.<anonymous> (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11229:66)
    at OctoPrintSocketClient.propagateMessage (http://octopi.private.shroop.net/static/webassets/packed_client.js?68a8d0b0:389:15)
    at http://octopi.private.shroop.net/static/webassets/packed_client.js?68a8d0b0:484:22
    at http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11225:23
    at _t (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11213:392)
    at http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11224:369
    at Function.<anonymous> (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:11229:73)
    at SockJS.onMessage [as onmessage] (http://octopi.private.shroop.net/static/webassets/packed_client.js?68a8d0b0:483:15)
    at SockJS.EventTarget.dispatchEvent (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:22585:20)
    at http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:23298:16
    at Array.forEach (<anonymous>)
    at SockJS._transportMessage (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:23296:17)
    at WebSocketTransport.EventEmitter.emit (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:22501:18)
    at WebSocketTransport.ws.onmessage (http://octopi.private.shroop.net/static/webassets/packed_libs.js?ea9fbc9a:25372:10)
callViewModelIf @ packed_core.js?8c230bbf:15214
(anonymous) @ packed_core.js?8c230bbf:15150
Pn @ packed_libs.js?ea9fbc9a:11208
(anonymous) @ packed_libs.js?ea9fbc9a:11229
callViewModelsIf @ packed_core.js?8c230bbf:15148
callViewModels @ packed_core.js?8c230bbf:15142
(anonymous) @ packed_core.js?8c230bbf:13883
DataUpdater.self._ifInitialized @ packed_core.js?8c230bbf:13907
DataUpdater.self._onPluginMessage @ packed_core.js?8c230bbf:13882
(anonymous) @ packed_client.js?68a8d0b0:390
Pn @ packed_libs.js?ea9fbc9a:11208
(anonymous) @ packed_libs.js?ea9fbc9a:11229
OctoPrintSocketClient.propagateMessage @ packed_client.js?68a8d0b0:389
(anonymous) @ packed_client.js?68a8d0b0:484
(anonymous) @ packed_libs.js?ea9fbc9a:11225
_t @ packed_libs.js?ea9fbc9a:11213
(anonymous) @ packed_libs.js?ea9fbc9a:11224
(anonymous) @ packed_libs.js?ea9fbc9a:11229
onMessage @ packed_client.js?68a8d0b0:483
EventTarget.dispatchEvent @ packed_libs.js?ea9fbc9a:22585
(anonymous) @ packed_libs.js?ea9fbc9a:23298
SockJS._transportMessage @ packed_libs.js?ea9fbc9a:23296
EventEmitter.emit @ packed_libs.js?ea9fbc9a:22501
WebSocketTransport.ws.onmessage @ packed_libs.js?ea9fbc9a:25372
packed_core.js?8c230bbf:11012 Uncaught TypeError: Cannot read property 'ip' of null
    at Object.<anonymous> (packed_plugins.js?40a77695:1061)
    at fire (packed_libs.js?ea9fbc9a:3269)
    at Object.fireWith [as resolveWith] (packed_libs.js?ea9fbc9a:3399)
    at done (packed_libs.js?ea9fbc9a:9306)
    at XMLHttpRequest.<anonymous> (packed_libs.js?ea9fbc9a:9549)
    at XMLHttpRequest.sentryWrapped (packed_core.js?8c230bbf:11012)
(anonymous) @ packed_plugins.js?40a77695:1061
fire @ packed_libs.js?ea9fbc9a:3269
fireWith @ packed_libs.js?ea9fbc9a:3399
done @ packed_libs.js?ea9fbc9a:9306
(anonymous) @ packed_libs.js?ea9fbc9a:9549
sentryWrapped @ packed_core.js?8c230bbf:11012
load (async)
T @ packed_core.js?8c230bbf:11012
(anonymous) @ packed_core.js?8c230bbf:11012
(anonymous) @ packed_core.js?8c230bbf:11012
(anonymous) @ packed_core.js?8c230bbf:11012
send @ packed_libs.js?ea9fbc9a:9601
ajax @ packed_libs.js?ea9fbc9a:9207
self.sendTurnOff @ packed_plugins.js?40a77695:1049
self.turnOff @ packed_plugins.js?40a77695:1044
(anonymous) @ packed_libs.js?ea9fbc9a:11395
dispatch @ packed_libs.js?ea9fbc9a:5184
elemData.handle @ packed_libs.js?ea9fbc9a:4992
sentryWrapped @ packed_core.js?8c230bbf:11012
```

rrauenza, 2019-05-20:
```
self.onDataUpdaterPluginMessage = function(plugin, data) {
        			if (plugin != "tplinksmartplug") {
        				return;
        			}
        			if(data.currentState){
        				//console.log('Websocket message received, checking status of ' + data.ip);
        				self.checkStatus(data.ip);
        			}
        			if(data.updatePlot && window.location.href.indexOf('tplinksmartplug') > 0){
        				self.plotEnergyData();
        			}
        		};
```

and

```
       		self.sendTurnOff = function(data) {
        			$.ajax({
        			url: API_BASEURL + "plugin/tplinksmartplug",
        			type: "POST",
        			dataType: "json",
        			data: JSON.stringify({
        				command: "turnOff",
        				ip: data.ip()
        			}),
        			contentType: "application/json; charset=UTF-8"
        			}).done(function(data){
        					//console.log('Turn off command completed.');
        					//console.log(data);
        					self.processing.remove(data.ip);
        				});
        		}
```
I don't understand -- but this `.done` seems to be invoked twice even though the `.ajax` is invoked once.  The second invocation of `.done()` has a `data` of `null`.

...and even though the 2nd invocation is null, the first appears to have succeeded.  So why isn't there a log entry?  Where does this POST go?  Ok, `on_api_command`.  But we never seem to get the "Turning Off" log.

Adding

```
 def on_api_command(self, command, data):
                self._logger.info("Command: %s, Data: %s", command, data)
                self._tplinksmartplug_logger.info("Command: %s, Data: %s", command, data)
```
That does happen:

> [2019-05-19 21:50:54,050] INFO: Command: turnOff, Data: {u'ip': u'3dprinterswitch/0', u'command': u'turnOff'}

Let's log the response:

```
                self._logger.info("Response: %s", response)
                self._tplinksmartplug_logger.info("Response: %s", response)
                return flask.jsonify(response)
```
/3 has no countdown, /0 does:

```
[2019-05-19 21:54:29,155] INFO: Command: turnOn, Data: {u'ip': u'3dprinterswitch/3', u'command': u'turnOn'}
[2019-05-19 21:54:30,081] INFO: Response: {'ip': '3dprinterswitch/3', 'emeter': {u'get_realtime': {u'total_wh': 0, u'current_ma': 4, u'power_mw': 0, u'voltage_mv': 121056, u'err_code': 0}}, 'currentState': 'on'}
[2019-05-19 21:54:30,136] INFO: Command: turnOn, Data: {u'ip': u'3dprinterswitch/3', u'command': u'turnOn'}
[2019-05-19 21:54:31,386] INFO: Response: {'ip': '3dprinterswitch/3', 'emeter': {u'get_realtime': {u'total_wh': 0, u'current_ma': 4, u'power_mw': 25, u'voltage_mv': 121023, u'err_code': 0}}, 'currentState': 'on'}

[2019-05-19 21:54:52,497] INFO: Command: turnOff, Data: {u'ip': u'3dprinterswitch/3', u'command': u'turnOff'}
[2019-05-19 21:54:55,584] INFO: Response: {'ip': '3dprinterswitch/3', 'emeter': {u'get_realtime': {u'total_wh': 0, u'current_ma': 4, u'power_mw': 0, u'voltage_mv': 120890, u'err_code': 0}}, 'currentState': 'off'}

[2019-05-19 21:55:13,053] INFO: Command: turnOff, Data: {u'ip': u'3dprinterswitch/0', u'command': u'turnOff'}
[2019-05-19 21:55:13,153] INFO: Response: None
```

That seems suspicious.

What the heck, though, because `turn_off` logs first thing.  And we get no log.

Add both loggers as info just in case:

```
def turn_off(self, plugip):
                self._tplinksmartplug_logger.info("Turning off %s." % plugip)
                self._logger.info("Turning off %s." % plugip)
```
Ok, we just had a logging visibility issue:

> [2019-05-19 22:01:45,863] INFO: Turning off 3dprinterswitch/0.

Oh... I think I see the problem and I think this shows it:

>[2019-05-19 22:12:35,922] INFO: Command: {u'count_down': {u'delete_all_rules': None}}, plug=3dprinterswitch/0, plug_num=-1
>[2019-05-19 22:12:36,100] INFO: Out: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
>[2019-05-19 22:12:36,102] INFO: Command: {u'count_down': {u'add_rule': {u'delay': 1, u'enable': 1, u'name': u'turn off', u'act': 0}}}, plug=3dprinterswitch/0, plug_num=-1
>[2019-05-19 22:12:36,108] INFO: Out: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

The /# code isn't handled in the countdown case for both on and off.

```
if plug["useCountdownRules"]:
                        self.sendCommand(json.loads('{"count_down":{"delete_all_rules":null}}'),plug["ip"])
                        chk = self.lookup(self.sendCommand(json.loads('{"count_down":{"add_rule":{"enable":1,"delay":%s,"act":1,"nam
e":"turn on"}}}' % plug["countdownOnDelay"]),plug["ip"]),*["count_down","add_rule","err_code"])
                else:
                        turn_on_cmnd = dict(system=dict(set_relay_state=dict(state=1)))
                        plug_ip = plugip.split("/")
                        if len(plug_ip) == 2:
                                chk = self.lookup(self.sendCommand(turn_on_cmnd,plug_ip[0],plug_ip[1]),*["system","set_relay_state",
"err_code"])
                        else:
                                chk = self.lookup(self.sendCommand(turn_on_cmnd,plug_ip[0]),*["system","set_relay_state","err_code"]
)
```

rrauenza, 2019-05-20:
FYI: ...I think the javascript ajax/done prolem went away once I fixed this.  ...I don't understand why.

jneilliii, 2019-05-22:
Thanks for the pull request, but to achieve what you are trying to do you don't need to use countdown timers at all. You can get away with using the GCODE triggers and there delays.

jneilliii, 2019-05-25:
Test version available with your countdown timer fixes incorporated along with some tweaks to fix sidebar issues for the HS300 and the status will now check status of plug 5 seconds after the countdown timers to give the plug a little bit of time to actually change it's state.  You can test by installing the 0.9.17 version using the url below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.17.zip

Let me know if this resolves your issue with countdown timers.

rrauenza, 2019-05-25:
Works for me -- except for the countdown ON set to 0 -- but that's a different issue.

closed 2019-05-25 by rrauenza