jneilliii/OctoPrint-TPLinkSmartplug #111: Graphing shows no plugs

rrauenza, 2019-05-20:
I can't get the graphing to work.

- I have verified the sqlite database is not empty.
- getEnergyData REST is never called

This appears to be run twice:

```
       		self.energySmartplugs = ko.computed(function(){
        			return ko.utils.arrayFilter(self.arrSmartplugs(), function(item) {
        						return "err_code" in item.emeter.get_realtime;
        					});
        		});
```

The first time, self.arrSmartplugs() returns an empty array.   The 2nd time, after initialization, it has 6 values as expected.  It's as if the HTML select is being initialized with the first result and never the second.

```
Starting dependency resolution...
<I had a breakpoint on the filter from above>
self.arrSmartplugs()
[]
packed_core.js?8c230bbf:15734 ... dependency resolution done
packed_core.js?8c230bbf:16016 Initial application setup done, connecting to server...
(index):7929 Initialized error tracking
packed_core.js?8c230bbf:13728 Connected to the server
packed_core.js?8c230bbf:15994 Finalizing application startup
packed_core.js?8c230bbf:15875 Going to bind 37 view models...
packed_core.js?8c230bbf:15928 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
<I had a breakpoint on the filter from above>
self.arrSmartplugs()
(6) [{…}, {…}, {…}, {…}, {…}, {…}]
```

jneilliii, 2019-05-22:
I don't have a strip to debug this problem.  Other user's from the original HS300 support request have not indicated an issue yet.  If I were to get enough tips to purchase a strip I would be more than happy to debug this further, but it's very difficult to do without it.

ridencww, 2019-05-22:
I can confirm graphing is working for my HS300 plugs and that the plugin controls both my HS300 strip and HS105 plug.  I am also using Themify;s Discorded as my theme (but it also works without Themify).  I haven't refreshed the plugin since I was testing the HS300 code with @jneilliii so in the next evening or two, I can update the plugin to the current release and confirm it continues to work.

rrauenza, 2019-05-22:
@jneilliii  Donation sent.  Also thanks for the bed leveler!

jneilliii, 2019-05-22:
Thanks, I'll Amazon order a strip and see what I can figure out.

jneilliii, 2019-05-24:
Ok guys, I think I figured out something in regards to the graphing after getting my strip.  Try this for me to verify my tests are accurate.  Toggle each of your plugs on/off and then open OctoPrint settings and press the save button.  Once saved, restart OctoPrint and once it reloads the plugs should show up in the pull-down list on the graphing tab. They still don't show on the sidebar, but I can look into that too.

rrauenza, 2019-05-25:
They show up now for me without having to do the settings/save thing.  Thank you!

jneilliii, 2019-05-25:
I don't think this issue is quite solved.  If you add an additional socket to your plugin settings, the plug doesn't automatically get added to the list I don't think. At least that's what I'm still seeing on my end.

closed 2019-06-15 by stale[bot]