jneilliii/OctoPrint-TPLinkSmartplug #112: Can't turn on when Countdown timer off enabled, on set to 0

rrauenza, 2019-05-21:
Again using the HS300 if relevant.

Printer never turns on if I had Countdown Timer enabled and off set to 300, on set to 0.  If I disabled countdown timer, printer came immediately on.  I had GCODE timer on instead.

```
save ...

[2019-05-20 17:33:25,937] DEBUG: Checking status of 3dprinterswitch/0.
[2019-05-20 17:33:25,938] DEBUG: check_status_cmnd: {'system': {'get_sysinfo': {}}}
[2019-05-20 17:33:25,939] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:25,960] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:25,962] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-05-20 17:33:25,963] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:25,979] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:25,980] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 3dprinterswitch
[2019-05-20 17:33:26,077] DEBUG: ^"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"{redacted}5","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-73,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"{redacted}0","state":0,"alias":"Plug 1","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}1","state":0,"alias":"Plug 2","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}2","state":0,"alias":"Plug 3","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}3","state":0,"alias":"Plug 4","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}4","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}5","state":0,"alias":"Plug 6","on_time":0,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-20 17:33:26,090] DEBUG: get_device_id response: {redacted}0
[2019-05-20 17:33:26,091] DEBUG: Sending command {'system': {'get_sysinfo': {}}, 'context': {'child_ids': [u'{redacted}0']}} to 3dprinterswitch
[2019-05-20 17:33:26,215] DEBUG: ^"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"{redacted}5","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-72,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"{redacted}0","state":0,"alias":"Plug 1","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}1","state":0,"alias":"Plug 2","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}2","state":0,"alias":"Plug 3","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}3","state":0,"alias":"Plug 4","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}4","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}5","state":0,"alias":"Plug 6","on_time":0,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-20 17:33:26,224] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:26,231] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:26,234] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-05-20 17:33:26,235] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:26,253] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:26,254] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 3dprinterswitch
[2019-05-20 17:33:26,366] DEBUG: ^"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"{redacted}5","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-72,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"{redacted}0","state":0,"alias":"Plug 1","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}1","state":0,"alias":"Plug 2","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}2","state":0,"alias":"Plug 3","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}3","state":0,"alias":"Plug 4","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}4","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}5","state":0,"alias":"Plug 6","on_time":0,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-20 17:33:26,378] DEBUG: get_device_id response: {redacted}0
[2019-05-20 17:33:26,380] DEBUG: Sending command {'emeter': {'get_realtime': {}}, 'context': {'child_ids': [u'{redacted}0']}} to 3dprinterswitch
[2019-05-20 17:33:26,560] DEBUG: eter":{"get_realtime":{"voltage_mv":123541,"current_ma":4,"power_mw":0,"total_wh":323,"err_code":0}}}
[2019-05-20 17:33:26,562] DEBUG: check_emeter_data: {u'emeter': {u'get_realtime': {u'total_wh': 323, u'current_ma': 4, u'power_mw': 0, u'voltage_mv': 123541, u'err_code': 0}}}
[2019-05-20 17:33:26,623] DEBUG: Checking status of 3dprinterswitch/1.
[2019-05-20 17:33:26,624] DEBUG: check_status_cmnd: {'system': {'get_sysinfo': {}}}
[2019-05-20 17:33:26,626] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:26,653] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:26,656] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-05-20 17:33:26,657] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:26,663] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:26,663] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 3dprinterswitch
[2019-05-20 17:33:26,821] DEBUG: ^"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"{redacted}5","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-73,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"{redacted}0","state":0,"alias":"Plug 1","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}1","state":0,"alias":"Plug 2","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}2","state":0,"alias":"Plug 3","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}3","state":0,"alias":"Plug 4","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}4","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}5","state":0,"alias":"Plug 6","on_time":0,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-20 17:33:26,827] DEBUG: get_device_id response: {redacted}1
[2019-05-20 17:33:26,827] DEBUG: Sending command {'system': {'get_sysinfo': {}}, 'context': {'child_ids': [u'{redacted}1']}} to 3dprinterswitch
[2019-05-20 17:33:26,938] DEBUG: ^"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"{redacted}5","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-72,"longitude_i":0,"latitude_i":0,"alias":"TP-LINK_Power Strip_CE34","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM:ENE","mac":"D8:0D:17:BA:CE:34","updating":0,"led_off":0,"children":[{"id":"{redacted}0","state":0,"alias":"Plug 1","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}1","state":0,"alias":"Plug 2","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}2","state":0,"alias":"Plug 3","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}3","state":0,"alias":"Plug 4","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}4","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}5","state":0,"alias":"Plug 6","on_time":0,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
[2019-05-20 17:33:26,942] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:26,948] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:26,949] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-05-20 17:33:26,949] DEBUG: Invalid ip 3dprinterswitch trying hostname.
[2019-05-20 17:33:26,953] DEBUG: Hostname 3dprinterswitch is valid.
[2019-05-20 17:33:26,953] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 3dprinterswitch
[2019-05-20 17:33:27,089] DEBUG: ^"system":{"get_sysinfo":{"sw_ver":"1.0.6 Build 180627 Rel.081000","hw_ver":"1.0","model":"HS300(US)","deviceId":"{redacted}5","oemId":"5C9E6254BEBAED63B2B6102966D24C17","hwId":"34C41AA028022D0CCEA5E678E8547C54","rssi":-72,"longi [... cut ...] te":0,"alias":"Plug 2","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}2","state":0,"alias":"Plug 3","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}3","state":0,"alias":"Plug 4","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}4","state":0,"alias":"Plug 5","on_time":0,"next_action":{"type":-1}},{"id":"{redacted}5","state":0,"alias":"Plug 6","on_time":0,"next_action":{"type":-1}}],"child_num":6,"err_code":0}}}
```

Could countdown delay of 0 for on be confusing the device?

jneilliii, 2019-05-22:
It very well could be confusing it.  Have you tried with a value of 1?

rrauenza, 2019-05-22:
Oh... I thought maybe I did .. I can try again.  I wasn't always being methodical.  I've switched to GCODE delay instead which I think is what I wanted anyway.  Does the gcode delay wait in octopi or does it delay in the switch itself as well?

jneilliii, 2019-05-22:
It waits in the pi side and then sends the command.

rrauenza, 2019-05-22:
Ah, yes found it -- Timer:

```
                                if plug["gcodeEnabled"]:
					t = threading.Timer(int(plug["gcodeOnDelay"]),self.gcode_turn_on,[plug])
					t.start()
				return
```

I'd rather that anyway (via gcode) because if I manually turn it on/off, I want it off/on now.

btw, my use case for this is I bought a [Thordsen filament dryer ](https://thordsen3d.com/) at Maker Faire this last weekend.  I want to shut it on at starts of prints and then off again when done.   And I want the printer off also after it goes through its cool down.

jneilliii, 2019-05-24:
Your use case then is better served with gcode triggers.  Then you can add `@TPLINKON 192.168.0.2` in your start gcode and `@TPLINKOFF 192.168.0.2` in your end gcode settings of your slicer for the filament dryer. You would probably want to disable the warn while printing option for this socket, as that will block the off command from being sent if the printer is in a printing state.

I actually saw that dryer on 3DPrinting Nerd's last video and thought it was a neat idea.  They are a little more open about what's inside over the FilaDry.

rrauenza, 2019-05-24:
Yes, I did exactly that with tplinkon and tplinkoff, but added /# of course.

Regarding the filament dryer I’m still trying to convince myself the warmed filament doesn’t stretch when pulled... (maybe I’m just using some narrow filament today)

jneilliii, 2019-05-25:
Test version available with your countdown timer fixes incorporated along with some tweaks to fix sidebar issues for the HS300 and the status will now check status of plug 5 seconds after the countdown timers to give the plug a little bit of time to actually change it's state.  You can test by installing the 0.9.17 version using the url below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.17.zip

Let me know if this resolves your issue with countdown timers.

rrauenza, 2019-05-25:
Ok, I've upgraded.  (I was already running my countdown patch already in a locally modified installation).  

I now have populated graphs/select dropdown!   Thanks!!

rrauenza, 2019-05-25:
Actually -- closed wrong bug.  I haven't tested this.

Setting ON countdown to 0 seems to never turn the printer on.  1 does.  So yeah, you may want to send a regular ON when countdown is 0.

jneilliii, 2019-05-25:
Just pushed an update to resolve this to some degree.  Now checks for a delay larger than 0 as well as some UI changes to limit delays to a number larger than 0 and sets new plug defaults to 1. You can still manually override the countdown timer delays to 0, which is now handled in the on command, but still doesn't work in the off command. Hopefully these changes are enough to resolve this issue. Install using the same URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.17.zip

closed 2019-06-15 by stale[bot]