jneilliii/OctoPrint-TPLinkSmartplug #113: wait for cooling of multiple hotends

mentaluproar, 2019-05-23:
I'd like to set my system up to wait for the hotends to cool before cutting power to the printer.  I can use M109 R60 T0 for a single extruder machine, but for my dual extruder, how do I tell it to wait for both hotends to cool before cutting power?

jneilliii, 2019-05-23:
Based on your command it looks like you are using a makerbot?  The T0 I believe indicates which extruder, so M109 R60 T1 would be the second extruder?

https://reprap.org/wiki/G-code#M109:_Set_Extruder_Temperature_and_Wait

mentaluproar, 2019-05-23:
I'm using a heavily modified robo3d R1+.  I understand the command, but the run command before off setting only has one line and I want it to wait for both hotness to cool.

jneilliii, 2019-05-23:
I think you misunderstand the settings.  The run command before off in the plugin is for running system commands on the pi.  This would be used to shut down a pi for example, or run some other script file on the pi to do something else.  You just need to add the following to your end gcode in your slicer or in OctoPrint's after print complete settings.
```
M109 R60 T0
M109 R60 T1
@TPLINKOFF 192.168.0.2
```
Replace the ip with your plug's ip.

jneilliii, 2019-05-23:
And you'd have to enable the GCODE trigger option in the plugin settings too.

mentaluproar, 2019-05-23:
AHHH!  Thank you!

closed 2019-05-24 by jneilliii