jneilliii/OctoPrint-TPLinkSmartplug #115: Problem with refresh button and graph select

devildant, 2019-06-02:
Hello,
for no apparent reason, the refresh button is disabled when loading the page (with the plug activated)
In addition, the chart does not have a listing in this list.
the reason for this malfunction is:
```
"err_code" in item.emeter.get_realtime
```
inside filteredSmartplugs and allPlugsDisabled

devildant, 2019-06-02:
i try to uninstall/install (with clean inside config.yaml and remove db file)
[image]

jneilliii, 2019-07-07:
Can you try this version for me?

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.17.zip

devildant, 2019-07-07:
hello, i tried, but i have an error when il click on plug icon (for turn on or turn off)

```
2019-07-07 10:57:50,714 - octoprint.server.api - ERROR - Error while excuting SimpleApiPlugin tplinksmartplug
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 104, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 283, in on_api_command
    response = self.turn_off("{ip}".format(**data))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 173, in turn_off
    plug_ip, plug_num = plugip.split("/")
ValueError: need more than 1 value to unpack

```

jneilliii, 2019-07-08:
Try the same branch again for me @devildant.  That code was added by someone else and I think I just fixed it. I believe the issue was when you ran into a non energy monitoring plug.

devildant, 2019-07-11:
it's better but the following bugs:
1) turn on printer, refresh octoprint page, the button refresh is disable
[image]

2) the graph select is empty
[image]

sorry for the response time I have problems with internet :(

jneilliii, 2019-07-11:
Weird.  I'm not able to reproduce on my end.  Do you only have the one HS110 plug?  I might be getting different results having multiple plugs configured and can do some additional testing.  Are there any errors in octoprint.log or the browser's console?

devildant, 2019-07-12:
hello, 
yes only one HS110.
no i don't have error inside octoprint.log and browser console

Tronnic, 2019-07-20:
Hello, I have a somewhat related problem. I don't see any power consumption at all (HS110) and the dropdown from the graphs menu is empty as well. 

[image]
[image]

RcTomcat1, 2019-11-18:
I have the same problem.
The power consumption widget is not even visible on my octoprint page, plug to graph is empty and nothing is selectable

schmitzCatz, 2019-12-22:
Hi,
i can confirm this problem:
* HS110(Firmware 1.5.4, Hardware Version 2.0)
* Octoprint Version 1.3.12
* TP-Link Smartplug Version 0.9.17

jneilliii, 2019-12-22:
What about if you enable polling? With debug logging enabled and polling enabled to 1 minute restart octoprint. Force a refresh on your browser by pressing ctrl + f5 (assuming windows) at the same time. Check your browser console for errors. After a couple of minutes to give the plugin plenty of time to poll check octoprint.log and the plugin log for errors. Upload them here for analysis, etc.

schmitzCatz, 2019-12-23:
Hi,

works like a charm. Polling is enabled and after restarting octoprint the plugin is working as intended.

Thank you for your fast replay!

PR #138 (2020-02-09): 0.9.18
#### Added
- cost estimation, may need to re-evaluate, resolves #124 
- list of known plug models the plugin works with, resolves #137 resolves #135 
- strip whitespace when identifying plug, resolves #132 
- automatic power off toggle in sidebar with cancel timer, resolves #139 resolves #140 
- new fontawesome 5.5 icon support, resolves #56 
- updated status update logic and knockout bindings, should resolve #115
closed 2020-02-09 by jneilliii