jneilliii/OctoPrint-TPLinkSmartplug #116: Second Plug Causes Error

Zexxen-Zalos, 2019-07-19:
I have an HS100 for my printer and a HS105 for an AC bed. I've tested the HS105 by itself and it works fine with this plugin. However if I turn the HS105 on or off when the HS100 is on and the raspberry pi is connected to the printer, this error shows up in the terminal:
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2739
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2739)"
Connection closed, closing down monitor

I've also included the octoprint log below.
[octoprint.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/3412472/octoprint.log)

Zexxen-Zalos, 2019-07-19:
Here are my settings and the debug log as well.
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/3412598/plugin_tplinksmartplug_debug.log)
[image]
[image]
[image]

jneilliii, 2019-07-19:
That's odd.  I have a similar setup in my home without any issues, although I don't use it for an AC heater pad but a light.  Does the same thing happen if only this plugin is enabled?

Zexxen-Zalos, 2019-07-19:
I'm not sure. I've been messing around trying to see if a specific plugin I added is conflicting but I cant get it to occur consistently. However, when the error does occur it's always immediately (to the point that its simultaneous) after the HS105 comes on or turns off.

Zexxen-Zalos, 2019-07-20:
So I've tried it with each plugin disabled aside from the ones that are in octoprint by default and I still am able to replicate the error

jneilliii, 2019-07-20:
What about changing the settings of the plugs? Thermal runaway is a fairly new feature that I don't use personally, for example. If you disable all the options for both plugs and restart octoprint dries it still occur?

Zexxen-Zalos, 2019-07-20:
It still occurs even if all settings are disabled.

closed 2019-08-10 by stale[bot]