jneilliii/OctoPrint-TPLinkSmartplug #117: Send Smartplug Data over MQTT

KingisBack, 2019-08-06:
Hi everybody,

is there a way to send the data from the Smartplug over MQTT to my Broker? I have successfully installed the MQTT Plugin. And now I am getting the "normal" Octoprint data onto my Broker. 
Having the energy data too would be a huge advantage.

I am asking because I know that jneilliii also wrote an MQTT Publisher and maybe here is some help.

Thank you in advance.

jneilliii, 2019-08-06:
It's not currently possible, but could potentially be integrated with some effort.

KingisBack, 2019-08-07:
Is there a possibility to provide me with some information, how I could accomplish that? 

I would be very thankful!

Thank you

jneilliii, 2019-08-07:
It's going to take some code programming integrating the checks for mqtt plugin installed and sending the commands to mqtt. It's not something that can be done quickly as it will require a good amount of testing.

jneilliii, 2021-01-27:
I don't really want to expand this plugin to broadcast MQTT data in order to avoid feature creep. I just think it's too minimal of a user base to put in the effort and I'm having to prioritize my work now with so many plugins I'm maintaining. I hope you understand.

closed 2021-01-27 by jneilliii