jneilliii/OctoPrint-TPLinkSmartplug #118: Turn on PSU when print is started

bgnome, 2019-08-17:
I was able to do this via GCode and Octoprint's GCode scripting on my Ender 3, which was great.  However, I upgraded my board to an SKR 1.3 and now it is no longer powered by USB so Octoprint cannot connect to the printer while it is off.  I can still switch it on via GUI but would like to see an option for automatic switching on when a print is sent to Octoprint.

jneilliii, 2019-08-20:
Can I assume you are using Cura and the OctoPrint Connection plugin for Cura?  Just trying to figure out what you mean by "when a print is sent to OctoPrint".  Do you mean when a file is uploaded?

bgnome, 2019-08-20:
Yes, I am using Cura 4.2.1 with an Octoprint plugin.  I also use an
Astroprint plugin in Octoprint which can send sliced GCode as well.
Unfortunately, neither option works as Octoprint is not able to connect to
the printer while the PSU is off.  It worked for me previously as the
printer was powered on through the USB on the stock board, even with the
PSU off.

jneilliii, 2019-08-20:
Perfect.  Just wanted to make sure I understood the use case.  I could see how it would be useful to have your printer on when sending a file via an external source like that.  I could fairly easily add the option to monitor the FileAdded event but that wouldn't help in the instance of when you are just selecting an already existing file. Not sure if it would work due to timing though.  I can't think of any other way to hook into the print started event when you can't start a print since the printer isn't connected.

bgnome, 2019-09-06:
So I set up Klipper and now Octoprint is connected at all times, even when the printer is off.  I can turn on the printer remotely, but I must issue a "restart" command for the printer to actually connect.
I have the TPLink turned on in the GCode script for "before a print job starts" which now works great, but Octoprint will not wait for Klipper to be ready after the restart command.  It will keep issuing subsequent GCode commands and fail the job after those don't go through.
Is there a way to add a delay so that Octoprint doesn't continue to process the script before Klipper is ready?

jneilliii, 2019-09-06:
Interesting, I've never played with klipper so not familiar with it's processes.  But I think you could use the gcode on delay?

bgnome, 2019-09-06:
I believe the GCode On Delay is the delay between the "turn on" command and
actually turning on the relay?  Is there a delay after turning on the relay
to prevent Octoprint from processing more GCode?

jneilliii, 2019-09-06:
I think you're right for the gcode on delay. I wonder how OctoPrint handles the [G4](https://reprap.org/wiki/G-code#G4:_Dwell) command.  I don't know if it will actually wait on the client side or still send the commands to the printer or not.  Might be something to try.

jneilliii, 2020-05-10:
I believe I'll be able to integrate this feature using the FileSelected event and adding additional options in the plugin's settings. This should in theory cover both a print initiated from the web interface or any program that utilizes the API to upload a file with the Auto Start Print option. I really just have to figure out what happens in OctoPrint when there is not an active connection as well as when their is a connection (USB BAck-powered Ender 3s for example) but there is not a usable powered up printer. Not sure how I'll detect that last one, so may focus on the first option and recommend the USB Blocking methods (3D Printed [back-power blocker](https://www.thingiverse.com/thing:3044586) or [Electrical Tape Fix](https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574/2) or other adapters similar to [TH3D](https://www.th3dstudio.com/product/power-blough-r-pi-usb-power-blocker/) or simply clipping the power line in the USB cable).

iceparr0t, 2020-05-11:
Having "played" with running the board off the USB back power with this plugin, I think your intention to ignore that use case for now is wise.   I was unable to find any method to detect that the printer is in the USB back powered state rather than the PSU powered state.  Additionally, while it was possible to toggle the PSU on and off, it was also far too easy to put the printer into an error state which required a hard reset.    One notable example was that if you boot up the firmware with only USB back power, you immediately get a TMC connection error which I was unable to recover without a reset and booting with the PSU on.

If you want help to map out this use case, I'm willing, but for now I've already moved to the Electrical Tape Fix to allow the usage to be far more robust.

Printer: Ender 3 running Marlin 2.0.x on an SKR mini e3

jneilliii, 2020-05-11:
Thanks for this valuable feedback @iceparr0t, I'm going to start developing the first option. This will be a fairly easy change I believe and could potentially have a test version out later tonight to get feedback on.

jneilliii, 2020-06-13:
Thanks to @BryanSmithDev and a couple of additional tweaks for non USB back-powered control boards I think I've got the initial workings for Upload events to auto-power on enabled plugs. It will capture the Upload event and sleep for the auto connect delay time (+1 second) and then select the file and print. 

This seems to be the most requested for users sending files directly from PrusaSlicer, etc. I've also opened a request to be able to better handle this [here](https://github.com/OctoPrint/OctoPrint/issues/3604) in the future, as you may not always want to power on your plug, but only when auto-print option is selected during upload.

Unfortunately, this looks like it might be the only option currently because even the file select event doesn't appear to work without a printer connection in the web interface, but I may be wrong and will require additional testing and possibly another feature request to OctoPrint.

narfotic, 2020-06-27:
I have the checkbox ‘On with upload’ checked but when I send a file from Prusaslicer, nothing happens.
Should I enable something else or install additional plugins? Or is this feature not implemented yet?

Fyi I have a Prusa MK3S plugged into TP-link smart plug and octoprint on pi4b always on.

jneilliii, 2020-06-27:
I've updated the logic to where this feature will work in the next release of OctoPrint (version 1.4.1), which should be soon, because the event hooks I'm monitoring changes slightly.

narfotic, 2020-09-02:
This works now! Now I just have to find a way to 'Start printing after upload' from prusaslicer with a few seconds delay. Because it, well, doesnt :)

jneilliii, 2020-09-02:
Yeah, we're working it out in #207

jneilliii, 2020-09-03:
I believe I have a working implementation now in my current dev branch. If you want to try it out and verify that would be great. You can install by copying/pasting the URL below in plugin manager > get more > ..from URL and clicking install. The vesion will still report 0.9.24 until I make the official release.

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.25.zip
```

iceparr0t, 2020-09-03:
Tested the dev branch linked above and it worked as intended.   Uploading gcode directly to Octoprint from PrusaSlicer with the "Start printing after upload" option enabled turned on the SmartPlug and then started the print after the connection was established.

Printer: Ender 3 running on SKR mini e3 Marlin 2.0.5.3.
Slicer: PrusaSlicer 2.2.0

jneilliii, 2020-09-03:
Perfect, thanks for confirming @iceparr0t

iceparr0t, 2020-09-04:
Of course, after I confirm, I may have found a bug that I originally attributed to a "user error".   I've now had two instances where when I started the printer using this method that OctoPrint hung while heating up the bed (M140 command).    Looking at the logging, it's not obvious that this plugin is to blame, so will try later to isolate it by disabling the other plugins but you can see the setandwait failing at the end of this log snippet, and this only has happened when I turned on the printer and started printing using the PrusaSlicer upload event.

Hopefully another user is able to confirm that it's not a bug in this dev version and it's something in my OctoPrint setup instead.

```
2020-09-04 01:41:00,010 - octoprint.plugins.shutdownprinter - INFO - eventView
2020-09-04 01:41:00,026 - octoprint.server.util.sockjs - INFO - User iceparrot logged in on the socket from client ::ffff:192.168.1.45
2020-09-04 01:41:00,209 - octoprint.plugins.shutdownprinter - INFO - eventView
2020-09-04 01:41:05,266 - octoprint.filemanager.analysis - INFO - Starting analysis of local:3ds_stylus-main_stylus_body-print_0.2mm.gcode
2020-09-04 01:41:05,268 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/3ds_stylus-main_stylus_body-print_0.2mm.gcode
2020-09-04 01:41:05,308 - octoprint.printer.standard - INFO - Cannot load file: printer not connected or currently busy
2020-09-04 01:41:08,106 - octoprint.filemanager.analysis - INFO - Analysis of entry local:3ds_stylus-main_stylus_body-print_0.2mm.gcode finished, needed 2.84s
2020-09-04 01:41:20,443 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2020-09-04 01:41:20,484 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 7 port/baudrate candidates: /dev/ttyACM0@115200, /dev/ttyACM0@250000, /dev/ttyACM0@230400, /dev/ttyACM0@57600, /dev/ttyACM0@38400, /dev/ttyACM0@19200, /dev/ttyACM0@9600
2020-09-04 01:41:20,484 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyACM0, baudrate 115200
2020-09-04 01:41:20,486 - octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 115200
2020-09-04 01:41:20,494 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2020-09-04 01:41:20,496 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-09-04 01:41:22,499 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2020-09-04 01:41:22,510 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-09-04 01:41:24,511 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2020-09-04 01:41:24,521 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-09-04 01:41:26,303 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Operational"
2020-09-04 01:41:26,311 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-09-04 01:41:26,321 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-state' with value Operational (IceEnder3)
2020-09-04 01:41:26,347 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 2.0.5.3 (GitHub)"
2020-09-04 01:41:26,354 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-09-04 01:41:26,762 - octoprint.plugins.printoid - INFO - //// Send alert to Printoid: 'printer-state' with value Operational (IceEnder3)
2020-09-04 01:41:27,022 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: 3ds_stylus-main_stylus_body-print_0.2mm.gcode, owner: None, user: None
2020-09-04 01:41:27,036 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2020-09-04 01:41:27,045 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: 3ds_stylus-main_stylus_body-print_0.2mm.gcode, owner: None, user: None
2020-09-04 01:41:27,068 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-09-04 01:41:27,080 - octoprint.plugins.DeleteAfterPrint - INFO - File selected
2020-09-04 01:41:27,121 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2020-09-04 01:41:27,235 - octoprint.plugins.DeleteAfterPrint - INFO - Printing started. Detailed progress started.{'origin': u'local', 'path': u'3ds_stylus-main_stylus_body-print_0.2mm.gcode', 'name': u'3ds_stylus-main_stylus_body-print_0.2mm.gcode', 'size': 677956L}
2020-09-04 01:41:27,293 - octoprint.util.comm - ERROR - Error while processing hook setandwait for phase sending and command M190 S60:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3435, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_setandwait/__init__.py", line 113, in hook_gcode_sending
    self._gcode_setandwait(cmd)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_setandwait/__init__.py", line 81, in _gcode_setandwait
    actual = heaters['bed']['actual']
KeyError: 'bed'
```

jneilliii, 2020-09-04:
Yeah, that error appears to be in the setandwait plugin. I'm not going to say it's unrelated, but it feels like it might be.

iceparr0t, 2020-09-04:
As expected, disabling SetandWait resolves the issue.   Of course, it *may* still be an interaction between the way SetandWait works and the upload event, but SetandWait only adds the minor functionality to cancel a print while the M140 command is running in Marlin, so not much value in troubleshooting further.

jneilliii, 2020-09-04:
If you enable emergency_parser in marlin, octoprint wil break that heat up on cancel.

closed 2020-11-14 by jneilliii