jneilliii/OctoPrint-TPLinkSmartplug #119: How to connect to Graph?

nexux78, 2019-09-03:
I noted that there is the possibility to have a graph.
How to enable it? I not have the possibility to add ip to the window "connect to graph"

TY
Maurizio

ElMacBgl, 2019-09-03:
I THINK it's only with HS110 model. You have HS100 don't you ? ;)

nexux78, 2019-09-03:
I have HS110(EU) ver.3.1
don't is possible?

jneilliii, 2019-09-04:
Enable the polling option in settings and cycle the power a couple of times and restart octoprint.  I haven't had the time to look into some of these issues, but eventually the interface will recognize you have power data to graph.  You may also have to force refresh the browser by pressing ctrl+f5.

jneilliii, 2019-09-18:
You may need to install my current dev version. Use the url below in plugin manager. 

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.17.zip

closed 2019-10-09 by stale[bot]