jneilliii/OctoPrint-TPLinkSmartplug #120: Countdown issues

ElMacBgl, 2019-09-03:
Hi,

There are some few issues with countdowns, which require to be fixed.
I wonder if you plan to do it, otherwise, I will do so on my own (please tell me ^^):
1. When countdown is set to 0 seconds, it should be considered as disabled and run a simple on/off, instead of creating a countdown.
We use such setting quite often because we need only one of the 2 countdowns to be active...
The problem is, when you have countdown=0 (or even very low value, such 2-3 seconds), the remaining time arrives on plug with negative value (probably because of normal lantencies), so the countdown is never executed...
If I were you, I would disable countdown below 3 seconds, and consider those as simple on/off...
2. State of plug is not properly reported when using countdowns... (green/red state)

IF this is fixed, it will be possible to use existing mods to merge PSU from Raspberry (Octopi) and the one from 3Dprinter, AND use the TPLinkSmartplug plugin:
Today, I would not do that because I don't want to remove power of a raspberry that would be on.
If solved, we can schedule the power off of plug (raspberry + printer) in 30-60s, then, just after the schedule is done, we properly run a command to shutdown the raspberry.
As a result, it will shutdown properly, THEN, power will be off.
With countdown issue, it's kinda annoying... ;)

jneilliii, 2019-09-04:
I'd have to look into number 2 above but wouldn't you want to set the countdown timer to a period further out than the system command that sends the shutdown command to the pi.  I think that would resolve your actual need.  Setting your settings to this for example.

[image]

This would send the shutdown command to your pi after 2.5 minutes and then 2.5 minutes later the power would be turned off on the plug. The timing allows for your printer to cool down a bit as well.

ElMacBgl, 2019-09-04:
Hi,

This is about what I though about the way to do it, and this is when I noticed the issue with countdowns: in your above example, you have "On delay countdown"=0
This make the "On" not to work in such config.
(I traced it with tplink_smartplug.py, "remaining time" arrives with -1 value, and... gets the rule to be disabled before it is actually executed)
Of course, by having Raspberry and printer using same PSU, I should not need this, however, I thought it worth to tell you about the issue (btw, I'm not even fully sure about the way I want to implement it: having only 1 PSU is obviously great in my mind, but it's also annoying if I want to use M80/M81 in GCODE, because I won't be able to 'see' the product after it finished (which is actually kinda great)

Maybe another simple way to fix the countdown issue with low value, would be to have separated "Use countdown timers" checkboxes, one for On, one for off. So you could use one, independently from enabling the other (I guess a warning should still be great to tell users about low values..).

Bests,
Antoine

ElMacBgl, 2019-09-04:
To sumarize, my actual options are:
1. One PSU for both printer + raspberry, so I could consider the pack as a network printer, both working together, and both being turned off together (in case of REAL needs, I could still run both on, using tlink app, but... well you know: it's not funny ;)
2. Printer separated from Raspberry PSU, so I would have OctoPi quite always on, so I could control the printer even when it's off.

Thinking about both options I have, the issue is not a real problem, however, it's worth to fix it : with this working, I'm rpetty sure I could find some other interrestig opotions to try :D

ElMacBgl, 2019-09-05:
I found a fix for 1st issue (at least, a way to achieve it): when countdown value =0, I make the command to be considered as set_relay_state on/off.
It does not solve the issue with other low values (latency can still create the issue with such low value), but now, there is a way to use countdown for on, OR for off, without being forced to use it for both.

In __init__.py:
*Go into turn_on function, and change
`if plug["useCountdownRules"]:`
with
 `if plug["useCountdownRules"] and plug["countdownOnDelay"] != "0":`

*Go into turn_off function, and change
`if plug["useCountdownRules"]:`
with
`if plug["useCountdownRules"] and plug["countdownOffDelay"] != "0":`
and
`if not plug["useCountdownRules"]:`
with
`if not plug["useCountdownRules"] or plug["countdownOffDelay"] == "0":`

[__init__.zip](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/3580702/__init__.zip)

Remains the 2nd issue with status, when using countdowns.

Bests,
Antoine Lévy
@lias IT

PR #149 (2020-03-15): 0.9.19
Hide tab if there are no energy monitoring capable plugs configured, resolves #145 
Fix countdown related issues, resolves #120 
Fix gcode processing related issues, resolves #148 
Abort power off now clears countdown timers, resolves #144 
Polling moved to server side code, resolves #141, resolves #151 
Added German translations, thanks to @TAOGde
Changed automatic power off option to an idle timeout, resolves #100, resolves #152
Added cooldown option, resolves #146 
Added calculated cost per print job, **highly experimental, and possibly inaccurate**, resolves #150
closed 2020-03-15 by jneilliii