jneilliii/OctoPrint-TPLinkSmartplug #123: Data not showing on side bar

skibum-za, 2019-09-17:
Just got myself a TP-Link HS110 and installed the plugin. The plugin graph works fine. Also it is reporting lots of good stuff in both octoprint.log and plugin_tplinksmartplug_debug.log

However, the sidebar just doesn't show anything. I've disabled all of my other plugins and rebooted the pi. 

When the page first loads, it does display the field names on the side bar, but once its finished it does then hide them.

I've tried checking and unchecking most of the options and tried firefox, chrome and edge. All the same.

`pi@octopi:~/.octoprint/logs $ tail plugin_tplinksmartplug_debug.log
[2019-09-17 13:55:04,163] DEBUG: Sending command {'emeter': {'get_realtime': {}}} to 192.168.1.95
[2019-09-17 13:55:04,630] DEBUG: ▒o▒"emeter":{"get_realtime":{"voltage_mv":242018,"current_ma":133,"power_mw":8633,"total_wh":1069,"err_code":0}}}
[2019-09-17 14:20:16,037] DEBUG: Checking status of 192.168.1.95.
[2019-09-17 14:20:16,040] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-09-17 14:20:16,041] DEBUG: IP 192.168.1.95 is valid.
[2019-09-17 14:20:16,041] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.95
[2019-09-17 14:20:16,367] DEBUG: ▒l▒"system":{"get_sysinfo":{"sw_ver":"1.5.7 Build 180806 Rel.135437","hw_ver":"2.1","type":"IOT.SMARTPLUGSWITCH","model":"HS110(UK)","mac":"D8:0D:17:6C:88:CA","dev_name":"Smart Wi-Fi Plug With Energy Monitoring","alias":"Prusa Plug","relay_state":1,"on_time":13115,"active_mode":"none","feature":"TIM:ENE","updating":0,"icon_hash":"","rssi":-54,"led_off":0,"longitude_i":0,"latitude_i":0,"hwId":"0750E2C15BB77902833ABF45366B8E9A","fwId":"00000000000000000000000000000000","deviceId":"8006C70E07DD88478233C73019C02A571B0B3E9C","oemId":"AB8C79FE7869756511CDC455BDFE41EA","next_action":{"type":-1},"ntc_state":0,"err_code":0}}}
[2019-09-17 14:20:16,369] DEBUG: IP 192.168.1.95 is valid.
[2019-09-17 14:20:16,369] DEBUG: Sending command {'emeter': {'get_realtime': {}}} to 192.168.1.95
[2019-09-17 14:20:16,553] DEBUG: ▒o▒"emeter":{"get_realtime":{"voltage_mv":242362,"current_ma":131,"power_mw":8101,"total_wh":1072,"err_code":0}}}
`

[image]

jneilliii, 2019-09-17:
Sometimes a restart is necessary, but you may get better results with my current development version.  It has some fixes in it for things like this. You can install using the url below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.17.zip

skibum-za, 2019-09-17:
Thanks. that version fixed it instantly. Previously I'd restarted both pi and printer quite a few times and nothing had happened. 

Close this one off. Thanks again.

closed 2019-09-17 by skibum-za