jneilliii/OctoPrint-TPLinkSmartplug #124: Cost calulation

Deapper, 2019-09-22:
Hi thanks for the great plugin.

Not trying to blow your project out as I'm sure you have RL to deal with. But it would be great if you could enter your cost for KWH and have a readout for it... Just a thought, its great as it is. All the best lol

kylane, 2019-12-11:
Agree with this and would be happy to donate to get it to happen!

jneilliii, 2019-12-11:
How would you guys propose it to be displayed, in the sidebar or as part of the tab, possibly in the graph (this could be  difficult to implement).

kylane, 2019-12-17:
> How would you guys propose it to be displayed, in the sidebar or as part of the tab, possibly in the graph (this could be difficult to implement).

I think the sidebar tab would be fine - maybe even a historical list of prints with their associated costs?

jneilliii, 2019-12-17:
Eventually some of this is going to get integrated into [OctoPrint-PrintJobHistory](https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1) plugin but I'll definitely try to update the sidebar at least with some simple math that multiplies a number by the current `Power` value.

kylane, 2019-12-17:
Eager to see it implemented @jneilliii - Ill have to polish up on my python to offer any pull requests.

jneilliii, 2019-12-17:
This actually can be achieved fairly easily on the javascript knockout observable side using a computed observable and a reference to it in the jinja template.

kylane, 2019-12-19:
Great news

jneilliii, 2019-12-24:
So I'm not sure how helpful this is going to be. Since there is no snapshot of start and end time for the print job in the current implementation.  You could reset the plug's stored data and start from scratch for each print, then at the end take note of the final cost value.  Or the value at start of print and at end of print and take the difference.  I think what you're really looking for is the [PrintJobHistory](https://github.com/OllisGit/OctoPrint-PrintJobHistory) plugin and once it integrates power consumption it would give you that data for billing that you are looking for.

[image]

[image]

jneilliii, 2020-02-02:
So my latest dev branch has integrated this stuff now.  You can install using the URL below in plugin manager.  If you don't mind testing and give some feedback that would be great.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.18.zip

kylane, 2020-02-03:
> So my latest dev branch has integrated this stuff now. You can install using the URL below in plugin manager. If you don't mind testing and give some feedback that would be great.
> 
> https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.18.zip

Awesome - will test tonight.

closed 2020-02-09 by jneilliii

kylane, 2020-02-10:
Apologies havent had a chance to test - had issues with my printer - just waiting on new metal hotend and dual Z kit.

PR #138 (2020-02-09): 0.9.18
#### Added
- cost estimation, may need to re-evaluate, resolves #124 
- list of known plug models the plugin works with, resolves #137 resolves #135 
- strip whitespace when identifying plug, resolves #132 
- automatic power off toggle in sidebar with cancel timer, resolves #139 resolves #140 
- new fontawesome 5.5 icon support, resolves #56 
- updated status update logic and knockout bindings, should resolve #115