jneilliii/OctoPrint-TPLinkSmartplug #126: Won't function as intended. No power on/off

krykiett, 2019-10-25:
Thank you for the amazing plugin, first and foremost. And thank you for taking the time to help us out! :)

When I first installed, earlier this week, before the update, everything was working as intended. I am not sure if the problem arose as a result of the update or not, because I just noticed I have lost control of the power function. 

I can add the plug to my config, the icon is there; it is clickable and it spins, but never changes color or activates the plug. My log says it's checking status when I click it, but it doesn't say anything beyond that, from what I can see. But I also don't really know how to read this log.

I have: Uninstalled the plugin, Uninstalled it and removed the information from the config.yaml, Removed any plugins I am willing to get rid of, Enabled debugging in TP-Link plugin, but am unsure of how to access that. 

The plug is accessible via the app, pushing the button on the plug works as intended, everything else seems to be in order. I even have the plug set as a reserved IP.

I have rebooted Octoprint after every step, I have rebooted the Pi several times as well. The log attached is showing from the last boot until I came here to post for help. I have only included that portion because I was also checking from remote connection, through reverse proxy and I would rather not attempt to remove all the instances of my IP.

The final error in the log is as a result of getting reverse proxy to work correctly with additional password security. It required me to change some things in my webcam, which makes the cam visible through the remote address, but makes it inaccessible on the local end. I believe though, the plugin was working after I made those changes.

I'm really at a loss of what to do at this point, because my knowledge base is limited and I've hit the end of it. Again, thank you in advance for all the hard work.

[log.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/3773198/log.txt)

jneilliii, 2019-10-25:
Did your plug's ip address change?  The log I need to see is the plugin_tplinksmartplug_debug.log file.  You can get it from the logging section of octoprint's settings.

krykiett, 2019-10-25:
The IP is reserved and being read as active in my router.

The log is saying valid IP but can't connect. I might have to try removing the plug and reinstalling it to the network.

- [plugin_tplinksmartplug_debug.log.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/3773799/plugin_tplinksmartplug_debug.log.txt)

jneilliii, 2019-10-26:
That's weird.  It is seeing the ip address as valid but the command is not getting a response when it sends it.  Have you tried unplugging your tplink from the wall and plugging it back in, basically rebooting the system.

krykiett, 2019-10-26:
My print finally finished, I power cycled the plug and it works fine.

I'm really unsure why I didn't start with that option and I feel foolish for wasting your time!

Thank you again though for all your help :)

closed 2019-10-26 by krykiett