jneilliii/OctoPrint-TPLinkSmartplug #127: Status not updating when turning on with app

biniecki, 2019-12-07:
I have a TP-Link Smart plug connected to my Ender 3 pro and octopi running on a raspberry pi 3b. I am able to turn on the printer from the server web interface and the printer automatically connect to octopi. My problem occurs when using the Kasa app to turn on the printer. When I turn the printer on using the app, octopi doesnt know the printer is on and wont connect. I've tried enabling status polling and that doesnt seem to help. Any help you can provide in rectifying this issue is greatly appreciated.

jneilliii, 2019-12-07:
That's why I put status polling into the settings. Have you restarted octoprint since enabling that option? I think it only initiates on startup.

biniecki, 2019-12-07:
Yes. I initially just restarted the octopi server and no change. I rebooted the entire pi and nothing has changed.

biniecki, 2019-12-07:
[2019-12-07 18:23:03,401] DEBUG: Checking status of 192.168.1.25.
[2019-12-07 18:23:03,402] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-12-07 18:23:03,403] DEBUG: IP 192.168.1.25 is valid.
[2019-12-07 18:23:03,404] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.25
[2019-12-07 18:23:03,686] DEBUG: « N"system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 181225 Rel.101038","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"68:FF:7B:2D:29:C1","dev_name":"Smart Wi-Fi Plug","alias":"3d printer","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-62,"led_off":0,"longitude_i":-832530,"latitude_i":420750,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800666A3D1D3A322F593C1A8571A31F31B344DF0","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2019-12-07 18:23:03,753] DEBUG: TIM
[2019-12-07 18:23:09,333] DEBUG: Checking status of 192.168.1.25.
[2019-12-07 18:23:09,334] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-12-07 18:23:09,335] DEBUG: IP 192.168.1.25 is valid.
[2019-12-07 18:23:09,335] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.25
[2019-12-07 18:23:09,446] DEBUG: « N"system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 181225 Rel.101038","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"68:FF:7B:2D:29:C1","dev_name":"Smart Wi-Fi Plug","alias":"3d printer","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-62,"led_off":0,"longitude_i":-832530,"latitude_i":420750,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800666A3D1D3A322F593C1A8571A31F31B344DF0","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2019-12-07 18:23:09,535] DEBUG: TIM
[2019-12-07 18:24:10,038] DEBUG: Checking status of 192.168.1.25.
[2019-12-07 18:24:10,039] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-12-07 18:24:10,040] DEBUG: IP 192.168.1.25 is valid.
[2019-12-07 18:24:10,041] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.25
[2019-12-07 18:24:10,297] DEBUG: « N"system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 181225 Rel.101038","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"68:FF:7B:2D:29:C1","dev_name":"Smart Wi-Fi Plug","alias":"3d printer","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-62,"led_off":0,"longitude_i":-832530,"latitude_i":420750,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800666A3D1D3A322F593C1A8571A31F31B344DF0","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2019-12-07 18:24:10,364] DEBUG: TIM
[2019-12-07 18:25:11,066] DEBUG: Checking status of 192.168.1.25.
[2019-12-07 18:25:11,068] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-12-07 18:25:11,069] DEBUG: IP 192.168.1.25 is valid.
[2019-12-07 18:25:11,070] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.25
[2019-12-07 18:25:11,271] DEBUG: « N"system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 181225 Rel.101038","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"68:FF:7B:2D:29:C1","dev_name":"Smart Wi-Fi Plug","alias":"3d printer","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-62,"led_off":0,"longitude_i":-832530,"latitude_i":420750,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800666A3D1D3A322F593C1A8571A31F31B344DF0","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2019-12-07 18:25:11,337] DEBUG: TIM
[2019-12-07 18:26:11,997] DEBUG: Checking status of 192.168.1.25.
[2019-12-07 18:26:11,999] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-12-07 18:26:12,000] DEBUG: IP 192.168.1.25 is valid.
[2019-12-07 18:26:12,001] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.25
[2019-12-07 18:26:12,374] DEBUG: « N"system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 181225 Rel.101038","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"68:FF:7B:2D:29:C1","dev_name":"Smart Wi-Fi Plug","alias":"3d printer","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-62,"led_off":0,"longitude_i":-832530,"latitude_i":420750,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800666A3D1D3A322F593C1A8571A31F31B344DF0","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2019-12-07 18:26:12,443] DEBUG: TIM
[2019-12-07 18:27:12,985] DEBUG: Checking status of 192.168.1.25.
[2019-12-07 18:27:12,989] DEBUG: {'system': {'get_sysinfo': {}}}
[2019-12-07 18:27:12,990] DEBUG: IP 192.168.1.25 is valid.
[2019-12-07 18:27:12,991] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.25
[2019-12-07 18:27:13,199] DEBUG: « N"system":{"get_sysinfo":{"sw_ver":"1.5.4 Build 181225 Rel.101038","hw_ver":"2.0","type":"IOT.SMARTPLUGSWITCH","model":"HS100(US)","mac":"68:FF:7B:2D:29:C1","dev_name":"Smart Wi-Fi Plug","alias":"3d printer","relay_state":0,"on_time":0,"active_mode":"none","feature":"TIM","updating":0,"icon_hash":"","rssi":-63,"led_off":0,"longitude_i":-832530,"latitude_i":420750,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"0000000000 [... cut ...] e","feature":"TIM","updating":0,"icon_hash":"","rssi":-64,"led_off":0,"longitude_i":-832530,"latitude_i":420750,"hwId":"0DC28CDD0B7E6C55F52AD35B8B68277E","fwId":"00000000000000000000000000000000","deviceId":"800666A3D1D3A322F593C1A8571A31F31B344DF0","oemId":"80CB8C7ED4CAD7DA4570979BFF731BA1","next_action":{"type":-1},"err_code":0}}}
[2019-12-07 19:17:44,898] DEBUG: TIM

Looks like its detecting the relay state change but octopi but icon doesnt change and octopi doesnt auto connect to the printer

jneilliii, 2019-12-08:
Ok, so there is a little logic change here that I may have to program for.  It looks like you are expecting the server to poll for the plug to turn on and when it is detected as on autoconnect to the printer?  

That's not how the plugin is programmed now and after further review appears that the polling is managed by the JavaScript client side, so if the browser isn't opened to the octoprint page the polling doesn't occur at all. 

I just tested it. If octoprint is loaded and polling is enabled and I turn the plug on in the Kasa app it will eventually pick up that change with the page opened on the next polling interval.

biniecki, 2019-12-09:
Yes. I would like the server to know that the plug is on so that it can attempt to auto connect to the printer. If this could be added to a future update it would be greatly appreciated.

jneilliii, 2020-09-03:
So I'm getting back into the backlog of issues and thought I would recommend you install the portlister plugin, linked below. It will resolve your issue I think with auto connecting on power on of the printer.

https://plugins.octoprint.org/plugins/portlister/

closed 2020-09-27 by stale[bot]