jneilliii/OctoPrint-TPLinkSmartplug #130: Turn on/off plug when print is started/stopped

Jackjan4, 2019-12-17:
Hi,

would it be possible to add a feature that automatically turns on/off the plug when a print has started/ended? I know that there is the feature to listen on M80/M81 codes but in this case for example (as I have a little print farm with nearly a dozen of printers) I cannot create custom g-code for every printer because they share a common gcode-repo. Would probably be interesting for many more people here :)

jneilliii, 2019-12-17:
You can add those gcode options in octoprint's gcode scripts settings and it doesn't have to reside in the file. I still haven't figured out a good solution for print start because you can't start a print without a printer connection.

Jackjan4, 2019-12-17:
Thank you for that great tip! I will try that out. 
Actually my printers are connected to octoprint (even when the plug is turned off) because they are powered by the USB cable. When I click on print without turning on the plug, they of course throw a power error. This is the reason it would be great if this plugin would turn it on (or maybe warn the user with a popup that he forgot to turn on the plug?)

jneilliii, 2019-12-17:
You might be able to use the gcode command then in the `Before print job starts` section of gcode scripts since your printer is powered/connected via USB.  Not all printers work that way.

closed 2019-12-30 by Jackjan4