jneilliii/OctoPrint-TPLinkSmartplug #131: Compatibility with P100 SmartPlug

mistic100, 2019-12-20:
Has anyone tested this plugin with the P100 SmartPlug ? 
https://www.tp-link.com/en/home-networking/smart-plug/tapo-p100/

I cannot find any info online if someone has reverse engineered this model too and if it uses the same protocol.

jneilliii, 2019-12-20:
No one's mentioned it that I know of.

mistic100, 2019-12-21:
I found this https://community.home-assistant.io/t/tp-link-tapo-p100/147792  
It seems this model uses a totally different API, apparently more secure than the HS100

jneilliii, 2019-12-21:
Good find. If the communication isn't possible internally then there's no way the plugin will be able to be adapted for it.

ippo1000, 2020-11-11:
There are some updates?

jneilliii, 2020-11-11:
It does appear someone has created a module recently. It's different enough to the other kasa line that I believe warrants a different plugin. I don't have the time or interest to create that however. If someone wanted to create this as a fork, the relevant info appears to be in this github. 

https://github.com/fishbigger/TapoP100

JamesSimpson, 2020-11-12:
Looks like there's a fair bit of breakthrough here

https://community.home-assistant.io/t/tp-link-tapo-p100/147792/

jneilliii, 2020-11-12:
Yeah, that's the same guy that did the fixes above. We're actually testing this module against specific UK kasa models that have broke communication with a firmware update.

Tyfy, 2020-12-29:
I have put something together using this project and the PyP100 library here [https://github.com/Tyfy/OctoPrint-TapoSmartplug](https://github.com/Tyfy/OctoPrint-TapoSmartplug)

This is my first look at OctoPrint plugins and I am not a Python developer so please treat it as a proof of concept at this point.

jneilliii, 2020-12-29:
Cool @Tyfy good work.

closed 2020-12-29 by jneilliii

JamesSimpson, 2021-01-29:
> I have put something together using this project and the PyP100 library here https://github.com/Tyfy/OctoPrint-TapoSmartplug
> 
> This is my first look at OctoPrint plugins and I am not a Python developer so please treat it as a proof of concept at this point.

Tried installing the fork, fails on install - maybe back to the drawing board?

Downloading https://files.pythonhosted.org/packages/a1/33/510a209f3f0dcf6dbce309b667a9e3ebde0d5e8e4697cf16392b0cd42c1f/PyP100-0.0.11.tar.gz
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-wj3v4i/PyP100/setup.py'"'"'; __file__='"'"'/tmp/pip-install-wj3v4i/PyP100/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-wj3v4i/PyP100/pip-egg-info
cwd: /tmp/pip-install-wj3v4i/PyP100/
Complete output (6 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-wj3v4i/PyP100/setup.py", line 31
setup(**setup_args, install_requires=install_requires)
^
SyntaxError: invalid syntax
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

StevilKnevil, 2021-01-29:
> Tried installing the fork, fails on install - maybe back to the drawing board?

I also had this problem due to Octoprint using Python 2 rather than Python 3. See:
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

Ideally, it would be possible to log these issues on https://github.com/Tyfy/OctoPrint-TapoSmartplug, however I think that requires @Tyfy to enable that functionality in that repo too.

Tyfy, 2021-01-30:
> Ideally, it would be possible to log these issues on https://github.com/Tyfy/OctoPrint-TapoSmartplug, however I think that requires @Tyfy to enable that functionality in that repo too.

Issues have now been enabled.

I have it working on my Octoprint but I can't remember what I had to do but I will have a look. 

I think I just downloaded the test2 release and it worked.

JamesSimpson, 2021-01-31:
Looks like the upgrade to Python 3 worked for me :) - all set up and working  - cheers all

jneilliii, 2025-08-17:
Version 2.0.0rc1 is out and is working with my Kasa based HS300, Kasa based HS103 and Tapo based P115. You can change the release channel in OctoPrint's Software Update settings to Release Candidate and update when prompted to get this version. 

Things I've noted so far, the P115 doesn't provide as much energy data as the kasa devices do. Tapo devices will require entering your username and password registered in the Tapo app, Kasa devices don't require those settings to be entered.