jneilliii/OctoPrint-TPLinkSmartplug #132: How to test?

tablatronix, 2019-12-22:
How do you test this, If I type M80/81 codes in terminal , nothing happens.

I tried using some plugins also that are supposed to be able to emit this, and they didnt work either..

I have GCODE Trigger checked

jneilliii, 2019-12-23:
You have to include the IP address of your plug, for example `M81 192.168.0.2`.

tablatronix, 2019-12-23:
Yeah I know that, nothing happens

jneilliii, 2019-12-23:
Enable debug logging and check the logs after trying. We'll s3 if there are any errors.

tablatronix, 2019-12-23:
```
2019-12-23 01:47:28,276 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {u'printer_baudrate': 115200, u'printer_port': None, 'firmware_name': 'Marlin Creality 3D'}
2019-12-23 01:47:32,189 - octoprint.util.comm - ERROR - Error while processing hook tplinksmartplug for phase queuing and command M81:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 500, in processGCODE
    if plug["gcodeEnabled"]:
TypeError: 'NoneType' object has no attribute '__getitem__'
2019-12-23 01:47:32,455 - octoprint.plugins.portlister - INFO - do_auto_connect
2019-12-23 01:47:32,459 - octoprint.plugins.portlister - INFO - realpath match
2019-12-23 01:47:32,546 - octoprint.plugins.portlister - INFO - Not autoconnecting; printer already connected
2019-12-23 01:47:32,910 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-12-23 01:47:37,738 - octoprint.util.comm - ERROR - Error while processing hook tplinksmartplug for phase queuing and command M80 192.168.1.2:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 491, in processGCODE
    if plug["gcodeEnabled"]:
TypeError: 'NoneType' object has no attribute '__getitem__'
2019-12-23 01:48:02,972 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-12-23 01:48:04,255 - octoprint.util.comm - ERROR - Error while processing hook tplinksmartplug for phase queuing and command M81 192.168.1.245:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 500, in processGCODE
    if plug["gcodeEnabled"]:
TypeError: 'NoneType' object has no attribute '__getitem__'
2019-12-23 01:48:06,012 - octoprint.util.comm - ERROR - Error while processing hook tplinksmartplug for phase queuing and command M81 192.168.1.245:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 500, in processGCODE
    if plug["gcodeEnabled"]:
TypeError: 'NoneType' object has no attribute '__getitem__'
2019-12-23 01:48:06,815 - octoprint.util.comm - ERROR - Error while processing hook tplinksmartplug for phase queuing and command M81 192.168.1.245:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 500, in processGCODE
    if plug["gcodeEnabled"]:
TypeError: 'NoneType' object has no attribute '__getitem__'
2019-12-23 01:48:09,798 - octoprint.util.comm - ERROR - Error while processing hook tplinksmartplug for phase queuing and command M80 192.168.1.245:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3292, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 491, in processGCODE
    if plug["gcodeEnabled"]:
TypeError: 'NoneType' object has no attribute '__getitem__'
```

jneilliii, 2019-12-23:
Thanks. Which version of octoprint?

tablatronix, 2019-12-23:
Should be latest, let me check

tablatronix, 2019-12-23:
'OctoPrint 1.3.12 running on OctoPi 0.15.1'

jneilliii, 2019-12-24:
Ok, after looking a little more, can you please upload your plugin_tplinksmartplug_debug.log file please.  The error shown above is indicating that the plugin is not able to find your plug in settings with ip `192.168.1.245`.

jneilliii, 2019-12-24:
Would also be helpful to see your plug's settings.

tablatronix, 2019-12-24:
I figured it out, I had recognized spaces in the ip field..

I had empty whitespace ->`  xxx.xxx.xxx.xxx`

The plugin worked fine, the gcode did not, so I did not notice it.

Might want to add a input pattern regex, or just filter whitespace on that..

tablatronix, 2019-12-24:
[image]

tablatronix, 2019-12-24:
I updated this issue in case you want to track it as a bug

jneilliii, 2019-12-24:
Thanks for confirmation.  There's not a whole lot of error checking in any of my plugins for bad input. Will consider adding this in on the next release I'm working on now.

PR #138 (2020-02-09): 0.9.18
#### Added
- cost estimation, may need to re-evaluate, resolves #124 
- list of known plug models the plugin works with, resolves #137 resolves #135 
- strip whitespace when identifying plug, resolves #132 
- automatic power off toggle in sidebar with cancel timer, resolves #139 resolves #140 
- new fontawesome 5.5 icon support, resolves #56 
- updated status update logic and knockout bindings, should resolve #115
closed 2020-02-09 by jneilliii