jneilliii/OctoPrint-TPLinkSmartplug #134: Verifying communication between tempfailsafe and TPSmartPlug

ThePaleHorseman, 2020-01-08:
I followed your write up here:
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/59

Everything appears to be good, I am able to power TP link off via the nav bar, how could I test the feature for tempfailsafe to properly execute the script? I am all new to this, and I am trying different executions in the terminal but I can't get the script to run and kill the power as if it were an actual over temp scenario.

I an ssh into my pi via putty.  Thanks for your time.

jneilliii, 2020-01-08:
I've never actually used the tempfailsafe plugin so I doubt really know how to test it.

ThePaleHorseman, 2020-01-08:
If I wanted to just use the tp link plugin. How do I verify I added that kill power script you wrote in the link I posted?

jneilliii, 2020-01-08:
That post and associated script was specifically for the tempfailsafe plugin. The tplink plugin has a temperature runaway option, but it's definitely not fail safe, as it only monitors for temperatures above a threshold. In the case where there is a thermistor failure for example, the temperature goes negative, which doesn't trigger my plugin currently.

ThePaleHorseman, 2020-01-08:
Yes I have both I was just curious if there was a way to test the setup I followed in the link to make sure I did it right.

jneilliii, 2020-01-08:
Reviewing that other plugin's code it seems the easiest approach would be to set the high threshold for the nozzle to something low and then set the nozzle's temperature to something above that threshold.

closed 2020-01-30 by stale[bot]