jneilliii/OctoPrint-TPLinkSmartplug #135: Plugin unable to connect to my Tapo Smart Plug

bellini13, 2020-01-19:
I purchased this smart plug, since it says it is TP-Link.  
https://www.amazon.com/gp/product/B07ZQYVXH8

I was able to setup and turn on/off the smart plug from the Tapo application.  I confirmed the IP address of the Tapo smart plug using my router and the Tapo app, which is 192.168.1.152

I also can ping 192.168.1.152 from my computer.

I configured the plugin to use the IP address, however it appears the TPLinkSmartplug is unable to communicate with the smart plug.  Below is the log file

[2020-01-19 17:17:27,760] DEBUG: OPERATIONAL
[2020-01-19 17:18:07,545] DEBUG: Checking status of 192.168.1.152.
[2020-01-19 17:18:07,546] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-01-19 17:18:07,548] DEBUG: IP 192.168.1.152 is valid.
[2020-01-19 17:18:07,548] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.152
[2020-01-19 17:18:07,652] DEBUG: Could not connect to 192.168.1.152.
[2020-01-19 17:18:07,653] DEBUG: 
[2020-01-19 17:18:07,653] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

Anything I am missing?

bellini13, 2020-01-19:
also to add, i connected to octopi using SSH, and confirmed that I can ping 192.168.1.152

running this command, returns "Cound not connect to host 192.168.1.152:9999"
./tplink_smartplug.py -t 192.168.1.152 -c info

Telnet connection refused...
telnet 192.168.1.152 9999
Trying 192.168.1.152...
telnet: Unable to connect to remote host: Connection refused

The octopi and smartplug are connected to the same wifi, and I do not have any

jneilliii, 2020-01-19:
Previously discussed, the tapo devices aren't compatible and originally never will be. 

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/131

closed 2020-01-19 by jneilliii

bellini13, 2020-01-19:
I see.  So how do i determine what are compatible devices?

jneilliii, 2020-01-19:
That's hard to say for sure. I know that the HS100, HS107, HS110, HS300 all work. My assumption is that any kasa smart plug device is compatible.

bellini13, 2020-01-20:
I see what I did wrong.  I first tried to order this product, the HS100
https://www.amazon.com/gp/product/B0178IC734/

Saw there was a cheaper new mini size option, which was the Tapo but did not realize it was not an HS100

I ordered the HS100 to give that a try, thansk for your assistance!

PR #138 (2020-02-09): 0.9.18
#### Added
- cost estimation, may need to re-evaluate, resolves #124 
- list of known plug models the plugin works with, resolves #137 resolves #135 
- strip whitespace when identifying plug, resolves #132 
- automatic power off toggle in sidebar with cancel timer, resolves #139 resolves #140 
- new fontawesome 5.5 icon support, resolves #56 
- updated status update logic and knockout bindings, should resolve #115