jneilliii/OctoPrint-TPLinkSmartplug #136: TPLink stopped working

santanig5, 2020-01-20:


santanig5, 2020-01-20:
Greetings,
As of this morning TPLink is grey and does not control the plug. I have verified the reserved IP address of the plug is the same, can control it with the Kasa app and google home, but Octoprint can no longer turn off the printer at the end of a print.

jneilliii, 2020-01-20:
Any errors in log?

santanig5, 2020-01-20:
Please tell me how to check

Gary J. Santaniello

jneilliii, 2020-01-21:
In the plugin's settings, enable the debug logging option and then restart OctoPrint.  Then once the web interface comes back up, try to check the status of the tplink and then in OctoPrint settings under the logging section download the plugin_tplinksmartplug_debug.log file.

santanig5, 2020-01-21:
Per your request

santanig5, 2020-01-21:
Hello again,
The issue was totally on my side, even though I could manage the plug via the app, I could not see it on my network.
I had to remove the plug from the app, remove the reservation and re-add the plug and reservation and it is now pingable and controllable via octoprint.
It did take an update so I don’t know if that had any affect on the network settings.
Thanks for your time, sorry to trouble you .

closed 2020-01-21 by jneilliii