jneilliii/OctoPrint-TPLinkSmartplug #139: [FR] Toggle for automatic stop after a printout

ShagoY, 2020-01-26:
Hi :)

First of all thank you for this plugin!

I would like to know if there is a possibility to integrate a similar feature to what the "[OctoPrint-ShutdownPrinte](https://github.com/devildant/OctoPrint-ShutdownPrinter)r" plugin does?

I like to have the choice to automatically shutdown or not, my printer after a print.

nteuscher, 2020-01-27:
+1 on this

I use this to control a room heater, so being able to start/shutdown the heater before/after the print would be helpful. Right now I am using the GCODE method and adding a bit to Octoprint before/after the print. I am using the PSU control plugin to control the power to the printer.

jneilliii, 2020-02-02:
> I would like to know if there is a possibility to integrate a similar feature to what the "OctoPrint-ShutdownPrinter" plugin does?

That plugin actually was written for that exact use case, originally for use with this plugin.  Why not just use it in conjunction with this one? The documentation on that repository's readme shows config examples at the bottom [here](https://github.com/devildant/OctoPrint-ShutdownPrinter#compatibility).

> I am using the PSU control plugin to control the power to the printer.

Don't think I understand quite what you mean here. My plugin controls power to tplink, so why use that one in combination?

nteuscher, 2020-02-02:
I set up power control through the PSU control plugin first and now have added a room heater for the winter months. I figured out how to auto shutdown the heater after a print using the @TPLINK command.

jneilliii, 2020-02-02:
Oh, I see.  You're controlling the heater with tplink and the printer with PSU Control.  That makes sense.

ShagoY, 2020-02-03:
Hi jneilliii :)

I think it's a "shame" to have to use another plugin which is entirely dependent on yours. Just to have a toggle button in octoprint ui (the "Shutdown Printer after print finishes" one).

I don't have anything against this creator but I usually try to avoid having several plugins for the same thing. For update reasons etc...

Besides, I use a lot of your plugins (even the ones that aren't released), I trust you much more ^^.

jneilliii, 2020-02-03:
> I think it's a "shame" to have to use another plugin which is entirely dependent on yours. Just to have a toggle button in octoprint ui (the "Shutdown Printer after print finishes" one).

I'll try to squeeze the changes into the next release. I noticed the auto shutdown plugin @nteuscher is using has a better code implementation in my opinion for the checkbox stuff, so I'll look into what it would take to do something similar.

PEstes, 2020-02-04:
If this is implemented, would it be possible to also have a configurable delay before shutting it down? Just to give the hotend time to cool down before its fan loses power.

jneilliii, 2020-02-04:
@PEstes, with the current gcode implementation it already does that, so will take that into consideration when implementing.

jneilliii, 2020-02-08:
Added **Automatic Power Off** toggle in sidebar with cancel timer in my latest 0.9.18 branch. Adds settings to configure timer, remember toggle state, and `Off on Print Complete` within each plugs individual settings. If you start another print with the toggle enabled the power off will be cancelled automatically. If you want to test prior to release you can use the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.18.zip

Let me know your thoughts.

ShagoY, 2020-02-08:
Hi jneilliii :)

After testing, it works correctly by adding the option "Off on Print Complete" in the outlet options (To manage several outlets differently, I suppose).

On the other hand, do you think you can integrate the "Nozzle temperature target" option ? In my opinion, it's the last thing your plugin is missing :)

Thanks again !

Ps: Now that I'm going to use your plugin to turn off my printer when I'm printing, I realize that the sidebar takes up some vertical space. Maybe create a sidebar especially for this feature or allow to hide/show/modify the layout ^^

jneilliii, 2020-02-08:
> sidebar takes up some vertical space. Maybe create a sidebar especially for this feature or allow to hide/show/modify the layout ^^

Please elaborate or share a mock-up of what you mean. I've given you exactly what you asked for and now you don't want it visible?

ShagoY, 2020-02-08:
Many thanks for the functionality!
I hadn't thought about how you were going to integrate the toggle.
Now that we have it, I realize that I had placed your sidebar at the very bottom of octoprint because I didn't find the display relevant (Current power etc..) unlike the "status/Files/etc.." sidebar.
Now that I placed it higher, I realized that on my small screen, I don't see the "files" sidebar anymore. That's why I was suggesting that you dissociate the toggle from the rest of your sidebar, or allow you to reduce the amount of information visible from it. That's the concept :) 
But again, it was just a suggestion because I'm very happy with what you did ;-)

jneilliii, 2020-02-08:
Click the title TPLink-Smartplug and the panel will collapse, but the toggle still be available. Maybe that will help. Then if you use my other plugin [Stateful Sidebar](https://plugins.octoprint.org/plugins/statefulsidebar/) the collapse will be persistent between reloads of the page.

ShagoY, 2020-02-08:
Oh my God, I didn't even think about the collapse... Good point for OctoPrint-StatefulSidebar, I didn't care that it saves the state of collapse, I thought it was a duplicate of 
 OctoPrint-DragonOrder, but I had read the description too quickly :)

Thanks a lot.

ShagoY, 2020-02-08:
Silly little question... Will I have to do a manual update when all this is on the master tag?

jneilliii, 2020-02-08:
> Will I have to do a manual update when all this is on the master tag?

Unfortunately because I already increased the version number in that branch any other changes I make moving forward with this release will not automatically get detected on your end when I officially release to master.  That being said, I think there are enough changes now in this version to go ahead and release version 0.9.18 officially so what you have installed now will more than likely be up to date. If there are any other changes (going through current issues now) I'll post back here in case you want to manually re-install.

jneilliii, 2020-02-09:
So there were some bugs in the version you installed that won't allow adding new plugs.  I just made a major overhaul to some of the bindings as well.  If you don't mind testing/verifying for me that would be helpful. I also made the sidebar collapsed by default.

ShagoY, 2020-02-09:
After a quick test with no change in settings, my plug switched off properly.

closed 2020-02-09 by jneilliii

ShagoY, 2020-02-10:
Hello jneilliii,

I ran into a problem I didn't think would happen...

In the absence of a "Nozzle temperature target" feature, I put a timeout/timer of 1800sec (30min) and I launched a very long print of about 9h and thus activated the automatic stop.
Finally at the end of this impression that I thought would last longer, I launched a second impression during the night and unfortunately I noticed that the timer of the first impression was still in line... So my printer shut down in the middle of work.

I had never encountered this kind of problem with OctoPrint-ShutdownPrinter because I was using the "Nozzle temperature target" which (I think) delayed the switch off at each print restart.

So I guess it might be interesting to add either :
- A kill queue on retrying the print...
- the nozzle temperature target.
- Display the "cancel auto stop" at the end of a printout regardless of the timer.

what do you think?

Translated with www.DeepL.com/Translator (free version)

jneilliii, 2020-02-10:
The automatic shutdown is separate from all other automatic options in the plug's settings. Can you share a screenshot of your plug's settings please?

ShagoY, 2020-02-10:
[image]
[image]

jneilliii, 2020-02-10:
Do you still have the gcode commands to power off in your files or octoprint's gcode scripts section? I think that's what triggered this issue. The abort hasn't been integrated to stop any of the previous methods, so it was either that or the countdown timer option that did it.

ShagoY, 2020-02-10:
No gcode trigger in my files/octoprint. I was using it with OctoPrint-ShutdownPrinter only.

Yes, it's my opinion the "off delay countdown" caused this. As it's 30min away, I didn't have the "cancel auto stop" prompt and when I restarted another print before the end of the current timer, obviously your plugin turned off my printer.

As indicated, I didn't have this kind of issue with OctoPrint-ShutdownPrinter because the shutdown of my printer was linked to the nozzle temperature (110°c).
So I could restart as many prints as I wanted the "switch off" would only be activated once it reached the temperature, which of course would mean that there is no printing in progress.

PR #138 (2020-02-09): 0.9.18
#### Added
- cost estimation, may need to re-evaluate, resolves #124 
- list of known plug models the plugin works with, resolves #137 resolves #135 
- strip whitespace when identifying plug, resolves #132 
- automatic power off toggle in sidebar with cancel timer, resolves #139 resolves #140 
- new fontawesome 5.5 icon support, resolves #56 
- updated status update logic and knockout bindings, should resolve #115