jneilliii/OctoPrint-TPLinkSmartplug #142: [Request] PowerStrip Support

stantond, 2020-02-09:
Cool plugin. Have you seen https://www.tp-link.com/uk/home-networking/smart-plug/kp303/ ? Thinking of getting one for always-on power to the Pi via USB, and printer itself and other accessories powered by the individually controlled 3 power sockets on the strip. Would be handy to control them in OctoPrint, and even have it turn the printer on/off at the start/end of the print.

jneilliii, 2020-02-09:
That plug should work with the plugin already.  Strip support was added a while back.  I use the 6 outlet HS300 model.  In the ip field you add `/0` after the ip address to tell the plugin to control the first outlet on the strip.

closed 2020-02-09 by jneilliii

stantond, 2020-02-09:
Oh nice! Might be worth adding the details to the plugin description in case anyone else asks

jneilliii, 2020-02-09:
I have on the repository readme on the last update, just haven't updated the plugin repository details.

ssombra, 2021-12-13:
> Oh nice! Might be worth adding the details to the plugin description in case anyone else asks

@stantond Hi. I know this is an old post but I wonder if you tried the USB connectors for your Pi. Is the current enough to power it (2.4A according to website)? I am using a Pi 4.

jneilliii, 2021-12-13:
It won't be enough for the Pi 4 I don't think without causing undervoltage issues. Pi 4 needs 3.1 IIRC. I do use them to power my pi zero 2 w though and haven't noticed any ill effects, although that device isn't connected to a printer.

ssombra, 2021-12-14:
That's what I thought. I will get it anyway, connecting the Pi with its power source to one of the outlets. thanks