jneilliii/OctoPrint-TPLinkSmartplug #143: [Bug] 0.9.18 can no longer control plug

Mesmer88, 2020-02-09:
Just upgraded to version 0.9.18
I notice that the power on/ power off icon in octoprint no longer works on my HS110 plug.

jneilliii, 2020-02-09:
Please check your octoprint.log and in your browser's developer console (typically press f12) for any errors.

jneilliii, 2020-02-10:
So i just tested by uninstalling and deleting all the plugin's settings in config.yaml, installing 0.9.17 and setting up an HS110.  Was able to pull status and toggle power on/off with the button in the navbar.  Used plugin manager to perform an upgrade and the process restarted octoprint followed by a request to reload the web interface. After doing that the plug's current status was identified on load and I was able to toggle power on/off and the graph updated, etc.  From what I'm seeing on my end everything is working as expected in the new version.

Mesmer88, 2020-02-10:
Here's the error i am getting:

```
2020-02-10 18:45:20,129 - octoprint.server.api - ERROR - Error while excuting SimpleApiPlugin tplinksmartplug
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 104, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 343, in on_api_command
    response = self.turn_off("{ip}".format(**data))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tplinksmartplug/__init__.py", line 229, in turn_off
    if plug["useCountdownRules"] and int(plug["countdownOffDelay"]) > 0:
TypeError: 'NoneType' object has no attribute '__getitem__'
2020-02-10 18:45:20,141 - tornado.access - ERROR - 500 POST /api/plugin/tplinksmartplug (::ffff:192.168.0.12) 32.83ms

```

Also tried deleting and adding the plug again. Now i am no longer getting the + icon to add a new plug :(

Also tried reinstalling the plugin and having the same issue (can no longer add a new plug)

[image]

jneilliii, 2020-02-10:
Thanks.  That's what I needed.  I'll have to look into the countdown rule error, but the button missing issue I just fixed.  Reinstall using this URL, force refresh and you should be able to add the plug back.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/master.zip

Mesmer88, 2020-02-10:
Thanks for the patch.
The add button is back and also the smart plug works now!

One odd thing i saw in the log is that this line:
`500 POST /api/plugin/tplinksmartplug (::ffff:192.168.0.12) 32.83ms`

The IP is not actually the smart plug IP

closed 2020-02-10 by Mesmer88