jneilliii/OctoPrint-TPLinkSmartplug #146: Add temperature threshold so the extruder cool down before turning off the plug

editter, 2020-02-10:
Can you add a "Temperature Target" check similar to the [ShutdownPrinter](https://plugins.octoprint.org/plugins/shutdownprinter/) plugin to prevent the printer from turning off while the extruder is still hot?

jneilliii, 2020-02-10:
Will review.

markp89, 2020-02-13:
You can just use the ShutdownPrinter plugin with this plugin. It works perfectly for me.

on ShutdownPrinter plugin

tick the activate this mode for the ' Mode GCODE'

GCODE - M81
Parameter - IP of TPLink Smartplug

At the bottom of the plugin settings 'Nozzle Temperature Target' set to your required value.

editter, 2020-02-13:
Thanks @markp89 That's how I have it set up but with the latest release this plugin added the "turn off printer" option as well so now my UI shows two boxes on the left that do the same thing, just with different capabilities. 

Another option I was thinking of was to hide the toggle switch if the setting isn't enabled for the plug. It makes the window a little silly because it would just show the refresh button for me because I don't have a power monitoring plug.

jneilliii, 2020-03-04:
I've just pushed changes to my current development branch that incorporates the cooldown option.  You can install and test using the URL below in plugin manager.  Currently, the cooldown option will only work with the automatic power off, so you must enable that option as well as enable the option `Off on Print Complete` you want to automatically power off on print complete. It would be helpful to get some additional testing prior to me releasing an official update.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.19.zip

PR #149 (2020-03-15): 0.9.19
Hide tab if there are no energy monitoring capable plugs configured, resolves #145 
Fix countdown related issues, resolves #120 
Fix gcode processing related issues, resolves #148 
Abort power off now clears countdown timers, resolves #144 
Polling moved to server side code, resolves #141, resolves #151 
Added German translations, thanks to @TAOGde
Changed automatic power off option to an idle timeout, resolves #100, resolves #152
Added cooldown option, resolves #146 
Added calculated cost per print job, **highly experimental, and possibly inaccurate**, resolves #150
closed 2020-03-15 by jneilliii