jneilliii/OctoPrint-TPLinkSmartplug #148: Icon not updating correctly anymore

Jackjan4, 2020-02-15:
Hi,

since the new update the topbar icon is not updating correctly anymore. When a print is done and the plug is shut off by GCODE the icons stays green even if the plug was turned off, and vice versa.
Before the update it always worked flawlessly.

jneilliii, 2020-02-15:
Quick question. If you press the refresh icon in the sidebar does thre status correct itself?

Jackjan4, 2020-02-15:
Yes, the status icon corrects itself after clicking the refresh button.

jneilliii, 2020-02-15:
Perfect, thanks. I'll check into this.

jneilliii, 2020-02-16:
I've staged the next release and would appreciate if you could test it before me prior to maing it public.  Install using the URL below in plugin manager to upgrade.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.19.zip

Jackjan4, 2020-02-16:
I tested it now and it works again. Thanks very much :)

HuguesDug, 2020-03-08:
Hello
Problem was also present on my configuration a year ago or so.
Then, via the update, it got corrected.
Recently, it came back with some updates distributed:
- TPlinkSmartplug 0.9.18
- Octoprint 1.4

My configuration uses as well the "power off via GCODE". 

All works like a charm when I click on the icon (green or red).

Are checked:
- warning prompt
- auto connect (5)
- auto disconnect (5)

Gcode Trigger
- Gcode on delay (0)
- Gcode off delay (300) --> time for printer to cool down the nozzle.

jneilliii, 2020-03-08:
I'm currently in development of the next release, could you try that version to see if you still have issues? Install using the url below in plugin manager. 

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.19.zip

HuguesDug, 2020-03-08:
Hello,

Install done. It seems it works OK with 0.9.19.

I will use the new parameters that you setup power off on "idle" instead of the current GCODE controled power off. I will post how it works.

PR #149 (2020-03-15): 0.9.19
Hide tab if there are no energy monitoring capable plugs configured, resolves #145 
Fix countdown related issues, resolves #120 
Fix gcode processing related issues, resolves #148 
Abort power off now clears countdown timers, resolves #144 
Polling moved to server side code, resolves #141, resolves #151 
Added German translations, thanks to @TAOGde
Changed automatic power off option to an idle timeout, resolves #100, resolves #152
Added cooldown option, resolves #146 
Added calculated cost per print job, **highly experimental, and possibly inaccurate**, resolves #150
closed 2020-03-15 by jneilliii