jneilliii/OctoPrint-TPLinkSmartplug #150: Request: Cost overview for started print

TAOGde, 2020-02-21:
Hello,

is it possible to see in the sidebar the cost ONLY for the actual startet print and the summary of the cost in the graph?
Is it also possible to save the cost of an running and finished print in the summary of the file overview?
[image]
[image]

jneilliii, 2020-02-21:
I'm sure it's possible, I'm just not really that interested in implementing it. I was hesitant to add it at all. You may want to look into print job history plugin, I think Ollis was considering integrating power data. 

https://github.com/OllisGit/OctoPrint-PrintJobHistory

jneilliii, 2020-02-21:
If a donation was received to implement said request, then that's a different story...

TAOGde, 2020-02-21:
> If a donation was received to implement said request, then that's a different story...

Donation is on the way ;)

jneilliii, 2020-02-21:
I appreciate that.  Quick question, how would you go about calculating the cost of a print if you were to manually do it using the Kasa app? I'm not really sure what I could do to push the final result into the file metadata to display in the file list, but I can research that aspect of it once we get the calculation side figured out.

TAOGde, 2020-02-21:
> I appreciate that. Quick question, how would you go about calculating the cost of a print if you were to manually do it using the Kasa app? I'm not really sure what I could do to push the final [...]

Well your plugin shows the cost with the user added cost per kWh. KASA does not have this entry but this would be my idea to see it in KASA.
I mean only temporary shown Cost per Print in the left sideboard. the values are already set in your settings.
I thing capture the start command. set the global used kWh in your calculation to zero. at the end of the print, capture finished command, u have a difference to the start kWh and that can be the cost for the print.

jneilliii, 2020-02-21:
> set the global used kWh in your calculation to zero.

That was my assumption too, doing a difference between the total kWh from when print starts to when print ends, accounting for cancelled prints, etc.  Just wanted to make sure my logic was sound.  The current implementation of cost estimate is very rudimentary, so I'm not even sure if it's very accurate, haven't really paid much attention to it.

jneilliii, 2020-02-22:
We have success, started a print last night with the changes incorporated into my latest dev version and voila! This .0032 was based on a kWh cost setting of 0.119. The trick was really getting the information injected into the file list.  

[image]

While doing my programming though I was wondering, have you seen the kWh/Cost data over the course of a print that took more than a day?  I think the total kWh being returned from the call to the kasa device is based on "last day". If that's true, then the total kWh used for the baseline "zeroing" number might actually be higher than the final total kWh reported when print completes, causing inaccurate calculations.

If you want to give it a try you can install using the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.19.zip

TAOGde, 2020-02-22:
> We have success, started a print last night with the changes incorporated into my latest dev version and voila! This .0032 was based on a kWh cost setting of 0.119. The trick was really getting the [...]

Hello from Germany,

i test your latest work. it works and the calculation is right.
I know what u mean with the "last day" etc. in Kasa. I will have a look what is do when I run a longer print over night.

Translation in German I suggest from "Last power cost:" to "Energiekosten"

jneilliii, 2020-02-22:
Glad it's working.  Let me know the results of a longer print, I haven't printed something that took an extended period in a while.  I'll have to look into how to make the plugin translation compatible.  I think the way I implemented it should work, just need to figure out how to include/bundle the translation.

TAOGde, 2020-02-22:
> While doing my programming though I was wondering, have you seen the kWh/Cost data over the course of a print that took more than a day? I think the total kWh being returned from the call to the [...]

When im use this graph it looks like the summary is global and not for Day ore Week?
 
[image]

jneilliii, 2020-02-22:
Well, that's a good sign.  I guess it would only get reset if the data was cleared/reset from the kasa app. I was looking into the translations and I think I'm going to need a bunch of these.  If you don't mind putting in the work to fill in the information below, I'll incorporate it into the next release.

```
# German translations for OctoPrint-TPLinkSmartplug.
# Copyright (C) 2020 The OctoPrint Project
# This file is distributed under the same license as the
# OctoPrint-TPLinkSmartplug project.
# FIRST AUTHOR <EMAIL@ADDRESS>, 2020.
#
msgid ""
msgstr ""
"Project-Id-Version: OctoPrint-TPLinkSmartplug 0.9.19\n"
"Report-Msgid-Bugs-To: i18n@octoprint.org\n"
"POT-Creation-Date: 2020-02-22 17:16-0500\n"
"PO-Revision-Date: 2020-02-22 17:17-0500\n"
"Last-Translator: FULL NAME <EMAIL@ADDRESS>\n"
"Language: de\n"
"Language-Team: de <LL@li.org>\n"
"Plural-Forms: nplurals=2; plural=(n != 1)\n"
"MIME-Version: 1.0\n"
"Content-Type: text/plain; charset=utf-8\n"
"Content-Transfer-Encoding: 8bit\n"
"Generated-By: Babel 2.6.0\n"

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:20
msgid "Model size"
msgstr ""

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:26
#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:30
msgid "Filament"
msgstr ""

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:30
msgid "Tool"
msgstr ""

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:35
msgid "Estimated print time"
msgstr ""

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:38
msgid "Last printed"
msgstr ""

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:40
msgid "Last print time"
msgstr ""

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:44
msgid "Last power cost"
msgstr "Energiekosten"

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:86
msgid "Powering off in "
msgstr ""

#: octoprint_tplinksmartplug/static/js/tplinksmartplug.js:88
msgid "Automatic Power Off"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_navbar.jinja2:15
msgid "Are you sure you want to proceed?"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_navbar.jinja2:19
msgid "Cancel"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_navbar.jinja2:20
msgid "Proceed"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:5
msgid "Plug"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:6
msgid "Options"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:43
msgid "Max Bed Temp"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:47
msgid "Max Extruder Temp"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:84
msgid "Minutes between checks"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:90
msgid "Cost per kWh"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:119
msgid "IP"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:120
msgid "Icon"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:121
msgid "Label"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:143
msgid "On Delay Countdown"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:144
msgid "Off Delay Countdown"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:153
msgid "GCODE On Delay"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:154
msgid "GCODE Off Delay"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:159
#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:163
msgid "Delay"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_settings.jinja2:168
msgid "Close"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_tab.jinja2:9
msgid "Plug to Graph"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_tab.jinja2:15
msgid "Records"
msgstr ""

#: octoprint_tplinksmartplug/templates/tplinksmartplug_tab.jinja2:21
msgid "Offset"
msgstr ""
```

TAOGde, 2020-02-23:
So we could try this first version.
[german_translation.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4240742/german_translation.txt)

jneilliii, 2020-02-23:
Thanks, I've updated the 0.9.19 branch and installed and tested.  After doing so I realized there are lots of hard coded English elements still.  I'll see about getting those templates updated and reach back out for additional translations.

jneilliii, 2020-02-23:
Ok, I think I've got most of them from google translate, but if you don't mind verifying accuracy, that would be awesome.

[messages.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4242394/messages.txt)

TAOGde, 2020-02-25:
> Ok, I think I've got most of them from google translate, but if you don't mind verifying accuracy, that would be awesome.
> 
> [messages.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4242394/messages.txt)

I will at weekend do the correction :)

on this way; I got this strange cost for a print.
startet a print overnight. it looks that the start cost variable is not the global one. it looks like this 1 day cost.
I put the two log files here to check.

[image]
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4252379/plugin_tplinksmartplug_debug.log)

[plugin_tplinksmartplug_debug.2020-02-23.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4252380/plugin_tplinksmartplug_debug.2020-02-23.log)

jneilliii, 2020-02-25:
Looks like only one log file made it through, but this is what I was concerned about.  When the print start event is triggered it subtracts the "current total kWh" from all configured plugs, creating a negative number/baseline.  Then when the print finished event is triggered it adds the "current total kWh" from all the plugs to that base line, giving you the calculated difference of the two.  If that "current total kWh" value resets to 0 every 24 hours then the estimated cost for the print is going to be wrong obviously and end up being a negative value. I think this would only be a problem for extremely long prints.  Of course, that's all assumptions because I haven't looked at your logs yet.

jneilliii, 2020-02-25:
You know what, I might not be resetting that global value to 0 on print complete.  I need to review the code again. That might be the issue here.

jneilliii, 2020-02-25:
Nope, that's not it.  I do reset the global value back to 0 [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/569e44f2e45a0870dcfd7e68acf37738aa7e0462/octoprint_tplinksmartplug/__init__.py#L483) in code.  I also reset it on print cancel [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/569e44f2e45a0870dcfd7e68acf37738aa7e0462/octoprint_tplinksmartplug/__init__.py#L452).

TAOGde, 2020-03-01:
i tried another print and had to switch on all my utilities like TP-Link, Raspberry etc..

as you can see in the screenshot the global cost graph starts with zero.
after the print it is 0.101 cost and 0.381 kWh.
but in your entry in the file list the cost is 0.2815

the print ins 3:14h and not overnight.

question: where you get the global cost graph? this graph only resets when the tp-link is powerless.
I know that the kasa app does not shown that value but gives kasa this value back in the api?

jneilliii, 2020-03-01:
> as you can see in the screenshot

You didn't include the screenshot of your given example.

> where you get the global cost graph

The data displayed in the graph is what's being stored in the database as a response to sheck_status [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/569e44f2e45a0870dcfd7e68acf37738aa7e0462/octoprint_tplinksmartplug/__init__.py#L331). What's shown in the sidebar is as described before, a calculation of the difference of the total kWh between the [print start](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/569e44f2e45a0870dcfd7e68acf37738aa7e0462/octoprint_tplinksmartplug/__init__.py#L458) and [print end](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/569e44f2e45a0870dcfd7e68acf37738aa7e0462/octoprint_tplinksmartplug/__init__.py#L467).

If you have debug logging enabled I and upload your plugin_tplinksmartplug_debug.log file it would be helpful in diagnosing.

TAOGde, 2020-03-01:
> You didn't include the screenshot of your given example.

update: images included

jneilliii, 2020-03-01:
The global graph is a real-time calculation of the 0.381 kWh value * the Cost per kWh from settings at that given point in time. The value displayed in the sidebar for the print is the calculation of ((kWh at End of Print - kWh at Start of Print) * Cost per kWh).

What I don't understand is why the value goes to 0 at exactly 4 o'clock, unless the timezone of your pi or plug doesn't match that of real-world time maybe?

PR #149 (2020-03-15): 0.9.19
Hide tab if there are no energy monitoring capable plugs configured, resolves #145 
Fix countdown related issues, resolves #120 
Fix gcode processing related issues, resolves #148 
Abort power off now clears countdown timers, resolves #144 
Polling moved to server side code, resolves #141, resolves #151 
Added German translations, thanks to @TAOGde
Changed automatic power off option to an idle timeout, resolves #100, resolves #152
Added cooldown option, resolves #146 
Added calculated cost per print job, **highly experimental, and possibly inaccurate**, resolves #150
closed 2020-03-15 by jneilliii