jneilliii/OctoPrint-TPLinkSmartplug #151: Automatic Power Off toggle doesn't turn off the printer.

jkkato, 2020-02-27:
Previously I have been using the gcode to turn off my printer. Lately, I have been trying to get the Automatic Power Off (the toggle at the bottom of the left side panel) to turn off the printer, as I like the idea of canceling the power off if I am at the computer.  It is not working. The countdown starts at the end of the print. When the countdown ends nothing happens, everything stays on. I did remove the gcode for the TP Link from my end code in cura.

Also, it is not very clear when the Automatic Power Off is enabled or disabled.
I do have two smart plugs configured in the TP Link plug in. One is for some LED lights, which I do not want to be turned off automatically.

jneilliii, 2020-02-27:
There are a couple of places it needs to be enabled.  I'm rethinking this, as I haven't officially released that feature yet, so I assume you're running my current dev version 0.9.19. The toggle on the sidebar allows you to enable/disable the process on the fly.  If it's enabled you'll get the pop-up message.  If that part is working, then you just need to make sure that within the plugin settings that the plugs you want to power off have the option "Off on Print Complete" enabled as well.

jkkato, 2020-02-27:
OK. I am not sure what version I am running (not at the computer). When the pop up showed on Octoprint that a new version was available, I clicked it. I was not aware that it was a developer version. 

If this feature is not officially released, then I will go back to the gcode method. Thanks for the quick reply.

PS is there another way to donate? I look at the PayPal link, but there is a $3.50 service charge on my end, in addition to the donation...

jneilliii, 2020-02-27:
Oh, I may have released that in the last version. Just can't remember right now. I don't have any other methods to receive donations, but don't sweat it. That's a steep charge.

jneilliii, 2020-03-04:
You were right, that option was added to my last release version.  The next version will have some improvements to the workflow logic and simplify the settings a little.  For it to work though (even in the version you have installed), both the automatic power off option must be enabled and the option `Off on Print Complete` must be enabled within the plug's settings as well. When the print completes it will loop through all your configured plugs and power off the ones that have that option enabled.

jneilliii, 2020-03-07:
I just updated my current development version with an idle timeout instead of on print complete option for automatically powering off plugs.  You can install using the URL below if you want to test it out.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.19.zip

PR #149 (2020-03-15): 0.9.19
Hide tab if there are no energy monitoring capable plugs configured, resolves #145 
Fix countdown related issues, resolves #120 
Fix gcode processing related issues, resolves #148 
Abort power off now clears countdown timers, resolves #144 
Polling moved to server side code, resolves #141, resolves #151 
Added German translations, thanks to @TAOGde
Changed automatic power off option to an idle timeout, resolves #100, resolves #152
Added cooldown option, resolves #146 
Added calculated cost per print job, **highly experimental, and possibly inaccurate**, resolves #150
closed 2020-03-15 by jneilliii