jneilliii/OctoPrint-TPLinkSmartplug #152: [Feature Request]: Power Off on Idle

jneilliii, 2020-03-05:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Would be nice to have the option to power off on idle, similar to PSUControl plugin.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

closed 2020-03-15 by jneilliii

Pointer-de, 2020-04-28:
Hi, realy great. Will this feature be available in your Tasmota-Plugin too?

jneilliii, 2020-04-28:
@Pointer-de, pretty sure it already is.

Pointer-de, 2020-04-28:
Am i blind (yes, it could really be so! ;-) ) or is there another Version of Tasmota plugin out there than 0.8.11?
[image]

jneilliii, 2020-04-28:
It's worded differently. It's on the left hand side and is named `Enable Automatic Power Off..."

Pointer-de, 2020-04-28:
I'm really blind... I allways looked for these three idle-options - sorry. Thanks

PR #149 (2020-03-15): 0.9.19
Hide tab if there are no energy monitoring capable plugs configured, resolves #145 
Fix countdown related issues, resolves #120 
Fix gcode processing related issues, resolves #148 
Abort power off now clears countdown timers, resolves #144 
Polling moved to server side code, resolves #141, resolves #151 
Added German translations, thanks to @TAOGde
Changed automatic power off option to an idle timeout, resolves #100, resolves #152
Added cooldown option, resolves #146 
Added calculated cost per print job, **highly experimental, and possibly inaccurate**, resolves #150