jneilliii/OctoPrint-TPLinkSmartplug #153: [BUG]: Octoprint Temperature graph doesn't work when plugin enabled.

bluemanuk, 2020-03-06:
**Describe the bug**
Hi, I just upgraded to Octoprint 1.4.0 and the main octprint graph does not display, it worked in safe mode so i switch off this plugin and the graph displayed as normal. 

**To Reproduce**
Steps to reproduce the behavior:
Installed octoprint 1.4.0 and connected printer
No graph was displayed
Ran terminal command and it worked fine
Prints also worked fine

**Expected behavior**
Graph should display as normal

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Browser : Safari
 - OctoPrint Version 1.4.0
 - Plugin Version 0.9.18

**Additional context**
Add any other context about the problem here.

jneilliii, 2020-03-06:
This is odd. I'm running 1.4.0 on several devices and don't have this issue. If tou look into your browser's developer console are there any errors shown on initial page load?

bluemanuk, 2020-03-06:
> Reply to: jneilliii/OctoPrint-TPLinkSmartplug <reply@reply.github.com>

Oh man, I just re-enabled it to check for browser errors and its working. Sorry dude!

closed 2020-03-06 by jneilliii