jneilliii/OctoPrint-TPLinkSmartplug #154: [Feature Request]: Execute GCODE at power-off or power-on

ahernsean, 2020-03-10:
**Is your feature request related to a problem? Please describe.**
When I turn on the machine and connect OctoPrint, the print head and bed move to the center from its home position. But sometimes the print head isn't anywhere near the home. If this is the case, the bed can sometimes hit the maximum Y and try to continue, jumping the belt loudly.

**Describe the solution you'd like**
I would like the ability to have GCODE execute when I power on the switch. This would mean that, upon first connect, my specific GCODE would be sent. In my case, I would send G1 X0 Y0 (or something similar).

**Describe alternatives you've considered**
There may be an existing OctoPrint plugin that could send GCODE upon certain events (like "connect"), but I'm not finding anything at the moment.

jneilliii, 2020-03-10:
Have you considered using octoprint's core gcode scripts settings? `After connection to printer is established` I assume would work for your use case.

ahernsean, 2020-03-10:
There we go. "GCODE Scripts". I think what I want can likely be done with this functionality.
[image]
I'll close out this issue.

closed 2020-03-10 by ahernsean