jneilliii/OctoPrint-TPLinkSmartplug #155: [BUG]: After update to 0.9.19 GCODE shutdown no longer works.

Mesmer88, 2020-03-23:
**Describe the bug**
After update to 0.9.19 GCODE shutdown no longer works.

**To Reproduce**
Steps to reproduce the behavior:

Using regular gcode to shutdown printer after job is finished:
@TPLINKOFF  192.168.0.24

Printer does not power off

Here is what i found in the octoprint.log:

```
2020-03-23 12:47:54,798 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-03-23 13:02:38,897 - octoprint.util.comm - INFO - Finished in 11026.883 s.
2020-03-23 13:02:38,899 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2020-03-23 13:02:38,913 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: CE5_mascarilla_v_mod.gcode, owner: Mesmer
2020-03-23 13:02:38,933 - octoprint.plugins.detailedprogress - INFO - Printing stopped. Detailed progress stopped.
2020-03-23 13:02:38,986 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2020-03-23 13:02:54,799 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-03-23 13:17:54,800 - octoprint.server.heartbeat - INFO - Server heartbeat <3
```

jneilliii, 2020-03-23:
If warn while printing is enabled you need to make sure the gcode off delay is set for something a little longer to give time for the print to finish.

Mesmer88, 2020-03-23:
It's disabled for me, the gcode timeout is 180.

jneilliii, 2020-03-23:
Can you please enable debug logging, restart octoprint, and test. If it still fails upload your plugin_tplinksmartplug_debug.log file.

jneilliii, 2020-03-23:
It would also be good to know if the same thing happens using M81 versus @TPLINKOFF.

Mesmer88, 2020-03-23:
Just tried the debug log, i see no set command for shut down.
Only the queries to get the power usage. Will also try reconfiguring the plug.

jneilliii, 2020-03-23:
Thanks @Mesmer88, there was a small change that switched processing GCODE from the queuing stage to the sent stage and that could have broke this. That's why it would be good to know if the M81 command does the same thing or not.

Mesmer88, 2020-03-24:
Just checked, it works with M81.

jneilliii, 2020-03-24:
Thanks @Mesmer88, I have an idea of how to fix this for the other command and will try to get that fixed up for the next release.

jneilliii, 2020-03-24:
I just pushed an update to the 0.9.20 branch that should resolve this issue for you @Mesmer88. If you'd like to try it out to confirm that would be great. Use the URL below in plugin manager to install this patched version.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.20.zip

fredondem, 2020-03-24:
Hello, i confirm the problem with 0.9.19, (but solved using M81 Gcode command instead @TPLINKOFF)
When my print will finish i try 0.9.20. Thank's

Mesmer88, 2020-03-24:
Confirmed, bug is fixed with 0.9.20.

PR #156 (2020-05-02): 0.9.20
Fixed @TPLINKON and @TPLINKOFF commands that were broken in last update, resolves #155
closed 2020-03-24 by Mesmer88