jneilliii/OctoPrint-TPLinkSmartplug #157: [BUG]: TPLINKOFF/M81 doesn't work with HS100

Petrda8, 2020-03-26:
**Describe the bug**
The GCODE commands to turn off the plug (HS100) stopped working. Cannot say when exactly but it's not that long (2-4 weeks maybe?). I have been using mainly the TPLINKOFF command, now tried also M81 and no change. 

I checked the IP address, all correct there. I'm also using a delay of 10min to let the hotend cool down. No settings were changed that could be messing this up (that I know of). 

Many thanks! 

**Desktop (please complete the following information):**
 - OS: RaspberryPi 4
 - OctoPrint Version: 1.4.0
 - Plugin Version: 0.9.19

jneilliii, 2020-03-26:
This is already queued up to be fixed, see [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/155).

closed 2020-03-26 by jneilliii

Petrda8, 2020-03-26:
Ah, sorry, didn't see that one, many thanks!