jneilliii/OctoPrint-TPLinkSmartplug #158: [Feature Request]: Turn off TPlink using a shell script

Petrda8, 2020-03-26:
**Is your feature request related to a problem? Please describe.**
I have built an enclosure and often printing overnight at home. In order to boost fire safety, I'm introducing a "second temperature control loop" that is not dependent on printer's sensors using a DS18B20 temp sensor and the [Enclosure](https://github.com/vitormhenrique/OctoPrint-Enclosure) Plugin. 

**Describe the solution you'd like**
I'd like to turn off my HS100 plug when the temperature hits a certain threshold (measured by the DS sensor) executing a printer-independent script (Gcode is only possible to send if the printer is connected and responsive if I'm not mistaken, that's why I think a shell script is needed). 

I'm already successfully measuring temperature from the sensor but the Enclosure plugin can only trigger IO pins based on the temp data. Looking at the source code here it is clear what kind of network request the plug expects, but unfortunately, I'm not a developer myself to put it all reliably together. Since it is a safety feature, I'd very much appreciate help from a professional. 

Thanks for any help and keep up the great work!

jneilliii, 2020-03-26:
See this post for command line options to send the API request to power on/off a plug. You will have to create a user API key (or application key) in OctoPrint's user manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/147

Petrda8, 2020-03-27:
Awesome, many thanks! Now I'll just need to figure out the triggering mechanism in the Enclosure plugin :D 
Btw the API call should also bypass the delay set up in the TPlink settings, right?

jneilliii, 2020-03-28:
All the same delays will still be followed, so if you are using countdown timers, system commands, and auto disconnect those delays will still be respected. This is so that if you are using a system command to safely shutdown your pi it has enough time to complete before powering off.

Petrda8, 2020-03-30:
I see, thanks! Is there any way to force a different delay (or no delay) in the API call? 
I run the Rpi on a different power source so turning off the TPlink won't do any damage and by default, I have a 10min delay to allow the hotend to cool after a print. In my scenario, I need to power it off immediately as it will only be triggered if there is a potential fire happening (in such case it cannot wait 10min). Many thanks for considering it

jneilliii, 2020-03-30:
Currently no, but will consider your use case for future update.

renorr3d, 2020-04-12:
> 
> 
> Awesome, many thanks! Now I'll just need to figure out the triggering mechanism in the Enclosure plugin :D
> Btw the API call should also bypass the delay set up in the TPlink settings, right?

Hi,

did you found a way to trigger the gcode when the temp is reached.
I would like to do the same to turn on and off the heater of my enclosure to keep it at the same temp during the print.

Many thanks

jneilliii, 2020-05-10:
I've added I added a wiki page with API examples.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/wiki/API

closed 2020-05-10 by jneilliii