jneilliii/OctoPrint-TPLinkSmartplug #159: [Feature Request]: Auto turn ON

KissmanCZ, 2020-04-01:
Hi, now we have auto turn off and this feature is absolutelly oustanding, but what about auto turn on?

Example: printer and lights are on two separate HS110, both turned off. I want to slice file in my PrusaSlicer and with this slicer, I can use auto upload to the OctoPrint and start print automaticaly after slicing. Is there any chance, how to catch this event and start printer before? Otherwise auto start print will fail bcs of offline printer.

Applepi, 2020-04-12:
This would be a very nice feature to have.

jneilliii, 2020-04-12:
I'll have to think of the logic here. I could trigger using the file added event, but then that would trigger power on with every file upload. I use Cura and the octoprint connect plugin in cura now supports powering on automatically prior to upload. Maybe if the file's name being uploaded starts with a specific string that could resolve the concern of every file upload. Does PrusaSlicer allow you to set the file upload name using tokens/parameters like it does for saving or does it use the same pattern as the file output option?

iceparr0t, 2020-05-05:
PrusaSlicer uses the same pattern as the file output option as it does for the file upload.   It can be changed (manually) before uploading.   It also does have a check box for "start print automatically", so I assume this is a separate event or Octoprint API command it is sending to start a print vs. just uploading it.    

I understand your reluctance just to have the plugin react to the file upload event, but PrusaSlicer users likely are not uploading gcode manually (e.g. through the web interface) very often, if ever, so it may be an acceptable solution, at least if it's an option in the plugin.   I certainly would use it in that manner, as it would cover >99% of my use cases.

blalor, 2020-05-08:
I would also like to see this. Why not trigger when the PrintStarted event fires, or when the printer state changes?  I'm assuming these are generally available, as the MQTT plugin emits them.

jneilliii, 2020-05-08:
Not all users have the option for print started event because their usb connection to the printer doesn't back power the controller board. When I do this feature, it will have to work for everybody, not just a select few.

jneilliii, 2020-05-10:
Closing as duplicate of request #118

closed 2020-05-10 by jneilliii