jneilliii/OctoPrint-TPLinkSmartplug #161: [BUG]: SmartPlug shuts off as soon as it attempt to connect

Abushawish, 2020-04-10:
OctoPlug in here means "OctoPrint-TPLinkSmartplug"

**Describe the bug**
Often times when I turn on the OctoPlug (to turn on my Ender 3), the OctoPlug suddenly turns off (thus cutting the power to the printer). This seems to happen when it's attempting to auto connect after I turn it on. This does not always happen but often. 

Video of the issue: https://www.youtube.com/watch?v=i9nV76ClQRg

My workaround:

1) This gets resolved by rebooting OctoPrint.
2) Disabling the auto connect option in the OctoPlug settings.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the OctoPlug in the OctoPrint web app
2. Wait a few seconds
3. The error: Notice the OctoPlug, and thus the Ender 3, turns off.

**Expected behavior**
The OctoPlug should turn on and remain on. 

**Screenshots**
Video of the issue: https://www.youtube.com/watch?v=i9nV76ClQRg

Screenshots of settings and installed plugins:

[image]
 
[image]

[image]

**Desktop (please complete the following information):**
 - OS: macOS 10.15.4
 - Browser: Chrome 80.0.3987.163
 - OctoPrint Version: 1.4.0
 - Plugin Version: 0.9.19

jneilliii, 2020-04-10:
I don't know if I've thought of this use case or not, but if you disabled the Disconnect Event Monitoring option and restart OctoPrint does the issue still happen? 

I think what might be happening based on your description is that the Ender 3 is being back-powered by the USB connection from the Pi and therefore is already connected when the plug gets powered on. When the AutoConnect happens it triggers a disconnect of the previous connection state and then that disconnect event triggers the plugin to power off the plug. I'd have to go back and verify though.

Abushawish, 2020-04-10:
Thanks for the prompt reply, I will experiment with the "Disconnect Event Monitoring" tomorrow after my current print is complete.

However, I don't think the Ender 3 is being back-powered by the USB connection since I taped off the 5V on the USB cable so the Ender 3 doesn't get back-powered. 

Also, not sure if these are helpful, but here are my logs: 

```
2020-04-09 20:31:57,983 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2020-04-09 20:31:58,006 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Error: Failed to autodetect serial port, please set it manually."
2020-04-09 20:31:59,183 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.140) 19.64ms
2020-04-09 20:32:10,587 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.140) 18.89ms
2020-04-09 20:32:13,335 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=Error
2020-04-09 20:32:13,825 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_autodetect_port, payload: {'commerror_text': u'Failed to autodetect serial port, please set it manually.'}
2020-04-09 20:32:22,779 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.140) 21.62ms
2020-04-09 20:32:34,893 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.140) 18.75ms
2020-04-09 20:32:46,682 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.140) 20.98ms
```

[image]

Thanks again!

jneilliii, 2020-04-10:
Can you upload the plugin_tplinksmartplug_debug.log file please? You can just drag the file into a comment here.

Abushawish, 2020-04-10:
So I was able to disable "Disconnect Event Monitoring" as you mentioned and so far I have not been able to reproduce the issue :D! There's a chance it _may_ still happen because the issues was happening often but not guaranteed every time, however so far so good:

[image]

Looking through the plugin_tplinksmartplug_debug.log I can actually see `powering off due to Disconnected event` log the last time I saw the issue happen, see:

```
[2020-04-09 20:31:51,785] DEBUG: TIM
[2020-04-09 20:31:51,796] DEBUG: powering off due to Disconnected event.
[2020-04-09 20:31:51,798] DEBUG: powering off 192.168.0.147 due to Disconnected event.
[2020-04-09 20:31:51,798] DEBUG: Turning off 192.168.0.147.
```

I've also attached the full log:

[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4462200/plugin_tplinksmartplug_debug.1.log)

So it seems you're right, something is causing a disconnect event however I can definitely say (1) OctoPrint is not connected when I turn on the OctoPlug, (2) OctoPrint actually showed State: Offline, and (3) the 5V power in the USB is taped off. 

I wonder if there's an OctoPrint setting or another plugin doing something funky?

acraigl, 2020-04-29:
I'm seeing this same issue, with an HS103. I've disabled the plugin to keep the printer alive during a long print (in progress), but once complete I'll see if the Disconnect Event Monitoring setting is in play here as well.

acraigl, 2020-04-30:
Update: I confirmed that Disconnect Event Monitoring was already disabled, yet I kept getting the immediate shutdowns. After some trial and error I determined it's the "Thermal Runaway" setting in the TP-Link Smartplug Editor that is causing the issue. Once I disabled that the shutdowns no longer occurred.

And for what it's worth, I was definitely NOT exceeding the temps. Shutdowns happened after cold start pretty regularly. And I never got to full temps (hotend at 205) prior to any disconnect. I also observed the same behavior with the HS100 as well as the HS103.

If you need more info, happy to help.

jneilliii, 2020-04-30:
Could you enable debug logging in the plugin's settings for me and restart octoprint completely and let's see if we can get some data to analyze on this please?

acraigl, 2020-04-30:
> Could you enable debug logging in the plugin's settings for me and restart octoprint completely and let's see if we can get some data to analyze on this please?

I will! I have another day of printing in front of me. But will get you logs ASAP.

acraigl, 2020-05-01:
OK, so semi-embarrassed. For some reason, in the TP Smartplug Settings, the max temps were both zero'd out. I have no idea why, as this was working fine for months. But it totally explains why the thermal runaway option was shutting the printer off immediately on power up. Since entering some temps, and re-enabling thermal runaway, the plug continues to operate normally.

I'm running a test print now with everything set, so hopefully this will be a non-event. Curious as to why the temp values got overridden/removed though.

closed 2020-05-31 by stale[bot]