jneilliii/OctoPrint-TPLinkSmartplug #162: [Feature Request]: On/Off with dht22 sensor data

renorr3d, 2020-04-12:
Hi,
First thanks for your plugin ! I use it evryday and it work flawless.
I do have a request though, I don't know if its totally impossible or an easy fix so i'll try to describe the thing.

I use your plugin to plug the heater I use to warm my enclosure.
I also use enclosure plugin to gather data with dht22 sensor and to turn on/off fans
I dont really want to connect my 1500w heater on a mosfet to the Pi like I have done for the fans.
Is it possible to add a feature like the thermal runaway monitoring but with the data gathered by the dht22 through enclosure plugin ? This will turn off and on the heater to maintain a constant temperature in the enclosure.
It would be perfect if I could set a maximum temp auto shutdown in case I forgot to shut it down.
And an On/off range like 45-50 to automatically turn the heater on and off to stay in the range.

Don't know if I explained the thing very well..
Thanks for your return.

jneilliii, 2020-04-12:
Couldn't you use the enclosure plugin to trigger the power on/off of the plug? I'm not familiar enough with its options if what it can do and how.

renorr3d, 2020-04-12:
I haven’t found any way to do it.
I don’t know how you tell to the plug to shut down or to turn of with your plugin !

But I have ask the same question to the developper maybe he can help. If I find anything I will tell it here. 
But it will be a safe way to control high voltage trough octoprint without any wire to the pi

jneilliii, 2020-04-12:
I'll look at the Enclosure Plugin and see what it's settings/capabilities are. I have a plan to do a similar thing for a heater box, but outside of octoprint all together using a sonoff relay and tasmota firmware and its rules..

jneilliii, 2020-04-12:
Doing a quick check it appears the enclosure plugin does have an option to run gcode and system commands as `output` devices, do I would think you could trigger either of those to power on/off the tplink. I'm not sure how the thresholds work in Enclosure, but I assume it's the same as you have with the fans.  If you want to go the system command route, then take a look at the examples I provided [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/147#issuecomment-586740610) in another similar issue/request.

renorr3d, 2020-04-12:
Thanks, saw a post about gcode commands but was not sure if it was possible. 
I will check if I find a way with your advices.
Many thanks !

jneilliii, 2021-01-27:
Cleaning up a little and closing this issue as I think it's possible with the above information previously linked. I don't want to get into programming GPIO for my plugins so hopefully you've been able to get what you needed.

closed 2021-01-27 by jneilliii