jneilliii/OctoPrint-TPLinkSmartplug #163: [BUG]: New Firmware for HS103 Removes support

Yeraze, 2020-04-18:
(Tracking this on python-kasa/python-kasa#42 also)

**Describe the bug**
A new firmware update for the HS103 removes all local network access. Everything works fine on a fresh stock unit if you bypass the Firmware Update during configuration.

jneilliii, 2020-04-18:
Not much I'll be able to do if tplink blocks the direct communication, but thanks for the heads up.

eschulma, 2020-05-27:
HS300 is not working either. I did upgrade the firmware and don't have a "fresh" one to test.

jneilliii, 2020-05-27:
That blows really. Hopefully someone smarter than me is able to reverse engineer it again. Otherwise my plugin is dead in the water. Thanks for the heads up, as that is the unit I'm using. Cam you tell me what the firmware version is?

Yeraze, 2020-05-27:
So I'm very confused now.. I setup the upgraded unit this weekend on a separate appliance (box fan), and now it works.  the `kasa` commandline tool finds both units.  Both units report Hardware 2.1 and Firmware 1.1.2.

eschulma, 2020-05-27:
It may only be partial. I can see the energy usage. But auto shutoff etc. does not work.

jneilliii, 2020-05-27:
@eschulma do mean using countdown timers? Can you enable debug logging in the plugin's settings and restart OctoPrint and attempt to do whatever is not working, and then download the `plugin_tplinksmartplug_debug` file from the logging section of OctoPrint's settings. That might help see what's going on.

eschulma, 2020-05-27:
Ok I will.

eschulma, 2020-05-29:
Here is the log. As you see at 2020-05-29 13:45:52,982 it says that it is powering off the plug -- but it doesn't happen. The plug remains on and the printer is still running.
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4703453/plugin_tplinksmartplug_debug.log)

jneilliii, 2020-06-05:
So I just had the time to really look at your log @eschulma and it appears you are using a power strip HS300. Did you enable the option within the plugin's settings to automatically power off?

[image]

@Yeraze can you confirm the plugin is working for you now or not?

Yeraze, 2020-06-05:
It's working beautifully now.

eschulma, 2020-06-05:
That checkbox was off. I will see what happens tonight.

eschulma, 2020-06-06:
I'm sorry to say that I still cannot control the plug even after ensuring the "Off on idle" is set.

jneilliii, 2020-06-06:
Not sure if it would make a difference, but did you restart octoprint after making the change? Can you please share screenshots of all your settings and provide log again after restarting octoprint and letting the idle time be reached?

eschulma, 2020-06-06:
Good news, it is now working. 

I had restarted OctoPrint, but it still didn't work. But I realized that my changes to the individual plugs were not sticking. I had to change them, press close, then on the main screen choose save even if none of the main screen options changed.

To avoid confusion it might be nice to have a save button directly on the plug option page. But this was my error and I thank you for your help.

jneilliii, 2020-06-06:
Oh right, didn't think of just pressing close on the main settings modal.

mikaelnelson, 2020-07-17:
I just bought a HS103 for my 3D printer and ran into this issue where it couldnt be controlled locally. I did see this on the TPLink site and was able to get the plug set to local only and working with this plugin: https://www.tp-link.com/us/support/faq/2707/

My HS103 is on firmware 1.1.2, not sure if this is the version people are having issues with or not. The app says my device is on the most recent software version.

jneilliii, 2020-07-17:
Thank you for sharing that faq article.

jneilliii, 2020-07-17:
I've added a link to the FAQ to the readme and it seems that everyone here effected by this issue is no longer having the problem. I'm going to go ahead and close this ticket now. Don't hesitate to open a new one if this comes back.

closed 2020-07-17 by jneilliii

whooooosh1, 2021-02-24:
I have the tplink p100 smart plug (other series of tplink) but I get the same debug file error's as the rest. Now can i not set my plug to local only because there is not an option for in the app. Does anyone still have this problem and know exactly now how to fix it? Or else, any suggestions on what I should do or buy to control my printer?

jneilliii, 2021-02-24:
@whooooosh1 the p100 is not a kasa based device, it's a tapo based device. You need to look into this fork to control your plug, this one won't.

https://github.com/Tyfy/OctoPrint-TapoSmartplug

whooooosh1, 2021-02-24:
@jneilliii Ahh this would be helpful. I taught that the tp link plugin would work. Thanks for that, but now i have a second problem that it keeps getting stuck on "checking status of (the ip for the smart plug)". The thunderbolt icon keeps spinning but would not connect.

Then I have the question if by the ip address you need to add /0 on the end when only one is connected, or if you only have to put the ip in and it would work.
Also, for the user-name and password I assume that I need to fill in my user name (not email) and password of my tapo account?

Thanks for helping me!

jneilliii, 2021-02-24:
I have no idea how that other plugin works. Best to seek help in it's issue tracker.

https://github.com/Tyfy/OctoPrint-TapoSmartplug/issues

whooooosh1, 2021-02-25:
Oke thanks, I'll ask if he can help me further.