jneilliii/OctoPrint-TPLinkSmartplug #164: [BUG]: Bad print quality

theresetmaster, 2020-04-22:
HI. After installing the plugin I start seeing blobs in the print - mostly on curves. Looking closer on the print, it is noticeable that the printer have several short stops that looks like the printer is not getting information from Octoprint. It is during these stops that the blobs are forming. 

When I disable the TPLinkSmartplug plugin the problem is gone. It seems that the plugin is causing the entire system to hang for a short moment.

My setup is: 
- Raspberry PI 2
- Ethernet connection
- Octoprint: 1.4.0
- OctoPi 0.17.0

(I have several printers and the problem is also on other versions of OctoPi)

[plugin_tplinksmartplug_debug (1).log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4513689/plugin_tplinksmartplug_debug.1.log)

jneilliii, 2020-04-22:
This may very well be related to the print progress status checking of the plug since it appears you have an energy monitoring enabled device. The older generation Pi you're using could also contribute to the issue because I haven't personally seen this type of problem using a Pi 3B or Pi 4. I'll look into making the status polling an option that can be disabled, but at the cost of not polling the energy data while printing.

theresetmaster, 2020-04-22:
Hi. Sounds very much in line with my thought about this problem. 

Making it an option to pull data would be great. Maybe also an option to set how often the data is pulled from plug ?

jneilliii, 2020-05-03:
I made some tweaks in my latest development version that I would appreciate some testing on to see if it resolves the issue. You can install using the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.21.zip

jneilliii, 2020-05-11:
So after 8 days I haven't heard back? I re-reviewed your posted log and it seems the print progress on these prints are happening very quickly, so I'm assuming these are small.short prints. In some cases the status check of the plug are separated by milliseconds, so I could see where this definitely could be problematic. Does the same issue occur for you with slower/larger prints that take longer than 20 minutes to complete?

theresetmaster, 2020-05-13:
Hi

I'm sorry that I did not get back to you, but I'm having some troubles with OctoPrint at the moment that needs to be fixed first. It seems that Seems that the printers I have running Repetier firmware is having some problems with transferring data from the Raspberry PI in a steady stream, and the funny thing is that I do not see the same problem on my printer that runs mendel. The same g-code file simply prints different in quality. 

I know this does not have to do directly with you, however, just to explain why I have not been giving too much feedback on your update. 

I did however install it, but did not see any difference.

On your question on the duration of the prints, you may see prints that are stopped, but normally what I print is prints that take 2-3 hours, while some initial testing prints might be 5-15 minutes.

Best
Jørgen Larsen

Denne e-mail blev kontrolleret for virusser af Avast antivirussoftware.
https://www.avast.com/antivirus

ybjlus, 2020-05-20:
@jneilliii I also have the same exact problem with a raspberry pi 2B. I did a clean octopi install and no problems what so ever. I then downloaded your 0.9.21 version and the stuttering started again...

jneilliii, 2020-05-20:
Thanks @ybjlus I assume you also have an energy monitoring plug? Having that information will help isolate where this might be happening and come to a solution quicker. Unfortunately, I've been busy with some other plugin updates and haven't had a chance to really dig into this problem.

ybjlus, 2020-05-20:
> Thanks @ybjlus I assume you also have an energy monitoring plug? Having that information will help isolate where this might be happening and come to a solution quicker. Unfortunately, I've been busy with some other plugin updates and haven't had a chance to really dig into this problem.

Hi I have a tp-link hs100 standard version. But this plug is always connected to the printer also when I tested it without the plugin installed

cbarth3, 2020-05-30:
My setup is:

    Raspberry PI 3 B
    Wifi
    Octoprint: 1.3.12
    OctoPi 0.17.0 Fresh install
    Plugin v0.9.20
    HS100 plug and I am seeing stops while making circles. I printed a cylindrical piece and it had lots of bumps. When I print a calibration file that has small circles at the leveling knobs I can see it stop. I disabled the plugin and printed from the SD card and do not have this problem.

Going to try the v0.9.21 that you posted earlier.

jneilliii, 2020-05-30:
@cbarth3 printing from SD probably resolved your problem, because it sounds like your issue is relative to the firmware buffer getting full. There's a large discussion about this in the community forum. If you disable the plugin and print from OctoPrint (not SD printing) the same stuttering may also happen, let me know.

cbarth3, 2020-05-30:
@jneilliii sorry I should have been more descriptive on this. If the plug-in is disabled I do not see this issue. I have completely removed the plug-in and started using the wemo version and do not see this issue with that one either. 

The only time the issue occurs is when the tplink plug-in is enabled.

jneilliii, 2020-05-30:
Ok, thanks. Your description before indicated printing from SD, and that threw me off.

jneilliii, 2020-05-31:
@cbarth3 I just made some changes to that 0.9.21 version. If you don't mind risking bad print quality I would appreciate it if you could try with this updated version. Use the URL below in plugin manager and let me know if you are seeing the stutters like before.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.21.zip

cbarth3, 2020-06-01:
Sounds good. I have a print finishing up late tonight so I will try it out in the morning.

cbarth3, 2020-06-02:
Tried to make some videos to review.
https://photos.app.goo.gl/xe5Mjh1rkTBynfX89

jneilliii, 2020-06-02:
Thanks, and I understand printing the same file with tplink disabled that doesn't happen. Can you please share your gcode file, your tplink settings (including the plug settings), and key me know what firmware/ version your printer is running? Any reason you've not upgraded octoprint to the latest?

cbarth3, 2020-06-02:
Originally that was a fresh install of octopi. I have since updated to the latest. The videos I uploaded it was on the latest of octoprint. I don't have the plugin enabled right now I will have to get that over a little later. Printer is Ender 3 pro v1.1.6.2

`;FLAVOR:Marlin
;TIME:123
;Filament used: 0.181969m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.4.1
M190 S50
M104 S210
M109 S210
M82 ;absolute extrusion mode
;Sliced at Sat 11-08-2018 11:43:25
;Basic settings:
;Layer height: 0.2
;Walls: 0.8
;Top/Bottom: 0.6
;Fill: 10
;Retraction Distance: 7
;Retraction Speed: 60
G28 ;Home

G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
G92 E0
G1 F3600 E-7
;LAYER_COUNT:2
;LAYER:0
M107
M204 S5000
M205 X30 Y30
G0 F3600 X26.401 Y26.181 Z0.2
M204 S500
M205 X20 Y20
;TYPE:SKIRT
G1 F3600 E0
G1 F1800 X26.992 Y25.631 E0.02115
G1 X28.036 Y24.784 E0.05636
G1 X28.663 Y24.34 E0.07648
G1 X29.772 Y23.659 E0.11057
G1 X30.478 Y23.288 E0.13146
G1 X31.754 Y22.726 E0.16798
G1 X32.692 Y22.405 E0.19394
G1 X34.617 Y21.922 E0.24593
G1 X35.29 Y21.794 E0.26387
G1 X35.787 Y21.751 E0.27694
G1 X37.864 Y21.659 E0.33139
G1 X38.549 Y21.668 E0.34934
G1 X38.982 Y21.716 E0.36075
G1 X40.588 Y21.953 E0.40327
G1 X41.43 Y22.141 E0.42586
G1 X42.72 Y22.531 E0.46116
G1 X43.447 Y22.805 E0.48151
G1 X44.644 Y23.348 E0.51594
G1 X45.342 Y23.723 E0.53669
G1 X46.498 Y24.449 E0.57245
G1 X47.128 Y24.906 E0.59283
G1 X48.128 Y25.737 E0.62689
G1 X48.525 Y26.099 E0.64096
G1 X52.833 Y30.403 E0.80046
G1 X54.729 Y32.146 E0.86792
G1 X58.828 Y35.609 E1.00847
G1 X63.079 Y38.805 E1.14777
G1 X67.517 Y41.766 E1.28751
G1 X72.097 Y44.459 E1.42667
G1 X76.838 Y46.898 E1.56632
G1 X81.744 Y49.08 E1.70695
G1 X84.18 Y50.043 E1.77556
[... 5979 lines cut ...]
;End of Gcode
;SETTING_3 {"extruder_quality": ["[general]\\nversion = 4\\nname = Ender3 0.2\\n
;SETTING_3 definition = fdmprinter\\n\\n[metadata]\\nsetting_version = 4\\ntype 
;SETTING_3 = quality_changes\\nposition = 0\\nquality_type = normal\\n\\n[values
;SETTING_3 ]\\nbottom_thickness = =top_bottom_thickness\\nbrim_line_count = 5\\n
;SETTING_3 infill_pattern = triangles\\ninfill_sparse_density = 10\\nline_width 
;SETTING_3 = 0.35\\nmaterial_diameter = 1.75\\nmaterial_flow = 90\\nmaterial_pri
;SETTING_3 nt_temperature = =default_material_print_temperature\\noptimize_wall_
;SETTING_3 printing_order = True\\nraft_margin = 5\\nretraction_amount = 7\\nret
;SETTING_3 raction_speed = 60\\nskirt_line_count = 3\\nspeed_infill = =speed_pri
;SETTING_3 nt\\nspeed_layer_0 = =speed_print * 30 / 60\\nspeed_travel = =speed_p
;SETTING_3 rint if magic_spiralize else 120\\nspeed_wall_0 = =speed_wall\\nsuppo
;SETTING_3 rt_angle = 70\\nsupport_interface_enable = True\\nsupport_z_distance 
;SETTING_3 = 0.2\\ntop_bottom_pattern = lines\\ntop_thickness = =top_bottom_thic
;SETTING_3 kness\\ntravel_avoid_supports = True\\nwall_line_count = 3\\nz_seam_t
;SETTING_3 ype = back\\n\\n"], "global_quality": "[general]\\nversion = 4\\nname
;SETTING_3  = Ender3 0.2\\ndefinition = fdmprinter\\n\\n[metadata]\\nsetting_ver
;SETTING_3 sion = 4\\ntype = quality_changes\\nquality_type = normal\\n\\n[value
;SETTING_3 s]\\nadaptive_layer_height_variation = 0.3\\nadaptive_layer_height_va
;SETTING_3 riation_step = 0.08\\nadhesion_type = skirt\\ndefault_material_bed_te
;SETTING_3 mperature = 45\\nlayer_height = 0.2\\nmaterial_bed_temperature = =def
;SETTING_3 ault_material_bed_temperature\\nprime_tower_size = 10.0\\nretraction_
;SETTING_3 combing = noskin\\nsupport_enable = False\\nsupport_type = everywhere
;SETTING_3 \\n\\n"}
`

cbarth3, 2020-06-02:
Here are the settings screens for the plugin
[image]
[image]

debren27, 2020-07-27:
Hi @jneilliii, thanks for pointing me toward this. I'm definitely seeing the stuttering issue and have narrowed it down to the TPLink plugin.

Prints are smooth from SD card of course, but also with Octoprint in Safe Mode, Octoprint in normal mode with no plugins enabled, and even with some plugins enabled. Granted, it's a little slower under each of these successive combinations, but it's only when I enable OctoPrint-TPLinkSmartplug that I see true stuttering -- actual stopping -- and therefore blobs and zits from ooze. The difference between no plugins enabled, and only TPLink enabled, is stark.

Besides the stuttering, I also noticed that the octoprint process used 100% of the cpu when only TPLink was enabled, versus 50% when no plugins were enabled.

I'm happy to provide whatever info I can, and to do testing. I don't have videos yet but I can make some if needed.

My setup:
OctoPrint 1.4.0 running on OctoPi 0.15.1
OctoPrint-TPLinkSmartplug 0.9.23
CR-10S S5
TH3D EZBoard on latest TH3D Unified (Marlin) firmware
Kasa Smart Plug 6-Outlet Power Strip HS300 http://amzn.com/B07G95FFN3
Kasa Smart Plug by TP-Link HS100 http://amzn.com/B0178IC734

The Kasa/TP-Link stripe does appear to have energy monitoring, although I don't use it. The other plug does not have energy monitoring. I have 3 of the plugs on the stripe, plus the other plug for the bed heater, configured in the plugin. (I plug the strip into a UPS but can't run the 1300W Keenovo through the UPS.)

My testing all weekend was on a 5-minute print (grid of 200mm-around 2-layer rings with segment sizes of .1mm to 2mm in increments of .1), which I now know exacerbates the issue based on this thread.

Let me know next steps.

Here's my plugin config:
[image]

Here's my gcode:
[curve stutter test v2 4x5 ringCircumf120 fs.1-2 60mmps horizExpNo.gcode.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4980100/curve.stutter.test.v2.4x5.ringCircumf120.fs.1-2.60mmps.horizExpNo.gcode.txt)

Here's the debug log:
[plugin_tplinksmartplug_debug.log.gz](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4980144/plugin_tplinksmartplug_debug.log.gz)

jneilliii, 2020-07-27:
@debren27 to confirm my suspicions could you remove all plugs except the HS100 and try the quick print again? I think you are right, the small print would exaggerate the issue. Also, what type of device are you running OctoPrint on and do you happen to know what your firmware's buffer size is configured at?

debren27, 2020-07-27:
> @debren27 to confirm my suspicions could you remove all plugs except the HS100 and try the quick print again? I think you are right, the small print would exaggerate the issue. Also, what type of [...]

I should be able to do this tonight. I have an 8-hour print running right now but after that.

Just to confirm, I'll remove the 3 HS300 plugs (or maybe just all of them) from the plugin configuration, but leave the plugin enabled. (I'll try to find a config file to back up so I don't have to totally reconfig afterward. :) ) Leaving the printer actually plugged into the smart outlets shouldn't matter and should be fine.

I have a Pi 3B+.

jneilliii, 2020-07-27:
Yeah, possibly having it with no plugs, and then one with just the HS100 setup. Specifically interested in the CPU utilization. I know @cp2004 mentioned a potential PR he was going to put together to fix a potential issue with the plugin's threading and that might be at play here as well.

debren27, 2020-07-28:
OK I did 4 different tests, all with Octoprint in normal (not Safe) mode and no other 3rd party plugins enabled:
no plugins enabled, not even TPLink:
printing: https://www.youtube.com/watch?v=3ALnoGthY0E
top: https://www.youtube.com/watch?v=cYO3y05t1YM
only TPLink plugin with all 4 plugs:
printing: https://www.youtube.com/watch?v=qucx70YYv8A
top: https://www.youtube.com/watch?v=xHmCgTXXvK8
only TPLink plugin with only HS100 plug:
printing: https://youtu.be/TFYE5wmMWPg
top:
only TPLink plugin with no plugs:
printing: https://youtu.be/-SDd1uzki_s
top: https://youtu.be/34F0xDmCQA0

All the videos of top (CPU usage) were taken right after the video of printing.

As we pretty much expected, the biggest difference was the HS300 plugs. Again CPU usage went from 50% to 100% with the plugs configured.

Here's the debug log, obviously only for 3 of the 4 prints (in the order above): [plugin_tplinksmartplug_debug.log.2020-07-27.gz](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4986291/plugin_tplinksmartplug_debug.log.2020-07-27.gz)

And here's a picture of the resultant blobbing in the test with all 4 plugs still configured:

[image]

jneilliii, 2020-07-28:
Thank you so much @debren27 I will be looking into this issue more in depth this weekend. I assume I'm not effected by this issue like you guys because I'm running a Pi4 4GB model. I can see it would definitely be exaggerated on circular round things if you're not using arc support in your firmware and the Arc Welder plugin. It's all the little moves that exaggerate the issue.

jneilliii, 2020-07-28:
@debren27 all your videos are marked private, so I can't open them...

debren27, 2020-07-28:
> @debren27 all your videos are marked private, so I can't open them...

Should be fixed.

I just ordered an RPi 4B 8GB to help with this (before I knew about this issue) so I may get a chance to test that. Somehow I didn’t know about Arc Welder so I’ll have to look into that.

jneilliii, 2020-07-28:
If your firmware supports arcs it's a miracle plugin.

jneilliii, 2020-07-28:
Videos are very helpful in seeing the problem. It does appear that the HS100 plug alone is unaffected by the issue, which does lead me to believe it's the print progress polling for energy data. I will be checking into this over the weekend to see if I can find a solution.

cp2004, 2020-07-28:
@debren27 Just a warning, the 8GB pi 4 is not currently *officially* supported but you can head to the forum to find a beta 64bit image, or I believe (Might be wrong) the 0.18 nightly builds support it.

Been following your issue, will look a little closer into it at some point.

debren27, 2020-07-28:
> @debren27 Just a warning, the 8GB pi 4 is not currently _officially_ supported but you can head to the forum to find a beta 64bit image, or I believe (Might be wrong) the 0.18 nightly builds support it.

Oh thanks. Last I knew the 4 wasn't supported at all but then I read that it is now. Didn't realize the 8GB was different. But now that I know my stuttering is an issue with the plugin and my plugs, I have less reason to upgrade, so maybe I'll wait.

> Been following your issue, will look a little closer into it at some point.

I really appreciate it!

jneilliii, 2020-07-28:
So far at this point we don't think there is anything that would be computationally intense from the plugin side. From what I'm seeing though, the performance without the energy monitoring running seems to be "normal". Can't say for sure, because it looks like the top video for just the one plug isn't linked.

In any case I think as a test I'll remove the print progress processing from the equation by making energy monitoring an option in settings to avoid the double hit to the plug status check, etc. I'm fairly confident in taking that out it will help your situation, at the loss of not having the power graphing during a print. 

I wonder if it might have something to do with the logging to sqlite3 during the process maybe @cp2004?

cp2004, 2020-07-28:
That did seem to be the only thing I spotted that could cause a 'stutter' in the print, might have to investigate. @debren27 Do you know what class/speed your SD card is? Could be a source of slowness, potentially?
The 3B+ should have no problem with this, I used to use a 3B - until it died :( (separate story) but just so I'm clear you have 3 plugs that have energy monitoring (Even if you don't use it, I think the plugin will log it) and one without?

@jneilliii Good call on the sqlite db - I have used sqlalchemy with sqlite db, but left it connected the whole way through. The query itself shouldn't be hard, however it may be the connect/query/disconnect (x3) cycle that slows things down. Best tactic I think here is to isolate it and test how long that takes.

debren27, 2020-07-28:
> because it looks like the top video for just the one plug isn't linked.

Argh, sorry, fixed above in my original comment.

debren27, 2020-07-28:
Also, I really like the idea of making energy monitoring optional, at least as a stopgap. For me the auto-shutoff is the killer app of this plugin. I love energy monitoring but I can live without it, to have both print quality and auto-shutoff.

> Do you know what class/speed your SD card is? Could be a source of slowness, potentially?

I believe it's this Samsung EVO Select 128GB Class 10 U3: http://amzn.com/B06XWZWYVP

> The 3B+ should have no problem with this, I used to use a 3B - until it died :( (separate story) but just so I'm clear you have 3 plugs that have energy monitoring (Even if you don't use it, I think the plugin will log it) and one without?

That's correct about the plugs. And I agree about the 3B+, I thought all indications were that it was more than qualified for Octoprint.

> it may be the connect/query/disconnect (x3) cycle that slows things down.

Based on my 30 years of sysadmin experience, that sounds very plausible! :)

jneilliii, 2020-07-29:
>> it may be the connect/query/disconnect (x3) cycle that slows things down.

> Based on my 30 years of sysadmin experience, that sounds very plausible! :)

I kind of agree now that I look at the code again. I should just create the connection on startup and leave it open until shutdown maybe? That could definitely take some cycles out of the equation. Just not sure how to handle a potential disconnected state for some reason.

jneilliii, 2020-07-29:
@debren27 I've added a new branch to begin working on this issue and have the optional print progress polling in the settings now. If you want to give it a try just install using the URL below in Plugin Manager > Get More > ...from URL and click Install button. Not need to uninstall the previous version or anything. The new option is disabled by default so after installing it should be ready to give it a test.

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.24.zip

jneilliii, 2020-07-29:
@cp2004 what's a good method of timing during debugging in Pycharm?

debren27, 2020-07-29:
Something's strange because I install from the 0.9.24 URL and it looks like it's doing it but then it says it installed 0.9.23. Not sure if that matters. I'll include the output below.

Is "Enable polling during print progress" the new option? I do see that.

I restored arrSmartplugs from a backup of config.yaml. I hope that's ok.

Here's the test print:
printing: https://youtu.be/q-ISD7W_Bls
top: https://youtu.be/bRWw4bBceuQ

Looks great! This makes me very happy, because I can have the on/off/autooff functionality AND the print quality. All I have to sacrifice is energy monitoring which I don't use so it's like giving up broccoli for Lent. And hopefully you'll fix that for those that want it (and maybe I'll want it later).

I checked the debug log and there's absolutely nothing from when it was printing. The print started at 22:49 and the log jumps from 22:39 to 22:55, which is when I canceled the print.

Here's my installer window output:

`Installing plugin from https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.24.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.24.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.24.zip
Downloading https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.24.zip (2.4 MB)
Requirement already satisfied (use --upgrade to upgrade): OctoPrint-TPLinkSmartplug==0.9.23 from https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.24.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-TPLinkSmartplug==0.9.23) (1.4.0)
Requirement already satisfied: flask<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.12.5)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (2.8.1)
Requirement already satisfied: tornado==4.5.3 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (4.5.3)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (2018.2.21)
Requirement already satisfied: Flask-Login<0.5,>=0.4.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.4.1)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.12.2)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.12)
Requirement already satisfied: blinker<1.5,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (1.4)
Requirement already satisfied: werkzeug<0.17,>=0.16 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.16.1)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.1)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (5.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (3.1.1)
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (3.4)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.7.19)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.9.0)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.1.5.post0)
Requirement already satisfied: netifaces<0.11,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (0.10.9)
Requirement already satisfied: pylru<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (1.2.0)
Requirement already satisfied: rsa<5,>=4.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (4.0)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (1.5.0.1)
Requirement already satisfied: requests<3,>=2.22.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (2.23.0)
Requirement already satisfied: semantic_version<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (2.8.4)
Requirement already satisfied: psutil<5.7,>=5.6.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TPLinkSmartplug==0.9.23) (5.6.7)
[... cut ...]
Building wheel for OctoPrint-TPLinkSmartplug (setup.py): finished with status 'done'
Created wheel for OctoPrint-TPLinkSmartplug: filename=OctoPrint_TPLinkSmartplug-0.9.23-py2-none-any.whl size=2221485 sha256=f1e6d5eaf8e03d5d080ee650c0ad64bd2f68feacf23b1e0f8f1cbe3fef883ec2
Stored in directory: /tmp/pip-ephem-wheel-cache-RhZexO/wheels/09/32/b0/ca7140508d9f961fb63bfe26520cf47b851646ba7fa4e498cf
Successfully built OctoPrint-TPLinkSmartplug
Installing collected packages: OctoPrint-TPLinkSmartplug
Successfully installed OctoPrint-TPLinkSmartplug-0.9.23
Done!`

cp2004, 2020-07-29:
> @cp2004 what's a good method of timing during debugging in Pycharm?

Usually I just add some points where it saves the time, then calculate. Obviously if you are going down to the wire, printing times takes slightly longer (But incredibly incredibly small). This is what I did last night https://gist.github.com/cp2004/8ff1d64254b05e67b929728d5f7a66ba

One more thing to bear in mind, PyCharm's debugger will run a lot slower than in real life, but would still be an ok representation.

On my Windows machine, it took 0.06 secs but I have a fast SSD etc, haven't had a chance to put it on a Pi and record the output. Based on it taking 0.06, it could take in the region of 0.2secs or more on the Pi, which could be enough to stutter OctoPrint.

debren27, 2020-08-01:
I kinda doubt it's related, but now I'm noticing that the plugin often turns off my HS100 for no apparent reason. I may have seen this once or twice before I upgraded tho. If I can narrow down why and it's not a boneheaded setting of mine, I'll post here or open a new issue.

jneilliii, 2020-08-01:
I have seen cases of back-powered printers from USB connection to pi cause this if you have disconnect monitoring enabled.

jneilliii, 2020-08-10:
@debren27 just merged in changes from @cp2004 on the timers and was hoping you'd be willing to test to see if with the print progress polling enabled if this new update still does the same thing or not? You can install using the URL below in Plugin Manager > Get More > ...fromURL

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.24.zip
```

debren27, 2020-08-10:
Sorry, I wish I had better news, but I still get a lot of stuttering, and 100+% CPU. I'll post the videos after work.

I did upgrade Octoprint to 1.4.2 in the meantime, but I wouldn't imagine that would make much difference. I first did a confirmation test by enabling polling in the old version of your plugin (and left all my other plugins still enabled) and got really bad stuttering. Probably other plugins made it a little worse.

Then I upgraded by copy and pasting your URL (although the installer output still said 0.9.23), disabled all other plugins, and rebooted. I'm printing the exact same gcode file as before.

jneilliii, 2020-08-10:
That's unfortunate news. I was hoping those changes would have cleared up some of that load. No need for videos, I'll have to dig in more on this. In the meantime I'm going to go ahead and release this new version that resolves another issue.

debren27, 2020-08-10:
Here's the plugin manager log for the install, and the debug log for the print.
[plugin_tplinksmartplug_debug.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5053002/plugin_tplinksmartplug_debug.log)
[plugin_pluginmanager_console.log](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/5053003/plugin_pluginmanager_console.log)

cp2004, 2020-08-10:
@debren27 I'm going to take a further look into this, but just want a couple of things:-
* Is it *constant* 100% CPU usage that you are seeing, and is it with or without polling enabled?
* Does it only happen with more than one plug, or is it solved with only a single plug.

Thanks for your consistent help with this!

debren27, 2020-08-10:
> * Is it _constant_ 100% CPU usage that you are seeing, and is it with or without polling enabled?

With polling enabled, watching `top` for half a minute, it bounces between 90% and 110%. So it's not quite pegged at a constant 100%, but it averages around that. But I believe it stays that high for the entire print, especially since it's a small (10-min) print consisting entirely of small circles with lots of segments.

Without polling enabled, while I didn't test the new version, the previous version bounced between 45% and 60%. That's this video: https://youtu.be/bRWw4bBceuQ. I can test the new version with polling disabled if you'd like.

> * Does it only happen with more than one plug, or is it solved with only a single plug set up.

Do you mean only a single plug with energy monitoring, or only the single HS100 (which doesn't have energy monitoring) that I left enabled for some of the previous tests? I've only tested the new plugin with all 4 plugs (3 with EM) enabled. I can pare that down to one plug with EM if you'd like.

> Thanks for your consistent help with this!

My pleasure. I appreciate the fast responses from you both. And it's kinda cool to be involved!

Let me know what other tests you'd like me to run.

cp2004, 2020-08-10:
Might have to hook it up to a debugger and run some hard prints. I've found a bearing cover that actually broke the virtual printer because it had too many segments, I reckon I can re-create something like what you're seeing.

>Do you mean only a single plug with energy monitoring, or only the single HS100 (which doesn't have energy monitoring) that I left enabled for some of the previous tests? I've only tested the new plugin with all 4 plugs (3 with EM) enabled. I can pare that down to one plug with EM if you'd like.

I've only got one EM plug here, so that's what I was wondering, as I don't have this issue - but then I also have a Pi 4, where they increased the speed of the SD card interface over the 3. Might see if I can get something to run on a Pi Zero, to really restrict the resources...

Nothing more for now, I think I've got a good picture of the issue, and (hopefully) an idea to investigate

jneilliii, 2020-08-10:
You both are great. I really appreciate the help @cp2004 and great support interaction with @debren27. It's always good to get someone else to look at this with fresh eyes. I too am running a pi4 4gb model with no issues, but do have multiple plugs with energy monitoring and have never seen this.

debren27, 2020-08-10:
I bought a Pi 4B (actually two; first the 8GB then you gave me the heads up about compatibility, so I got a 4GB) and I finally imaged the SD card over the weekend but "luckily" for you both I haven't deployed it yet so the 3B+ is still driving my printer. I have plenty of other projects and turning off polling is a great workaround for me so I won't rush to deploy the upgrade.

jneilliii, 2020-08-10:
The other fix is enough for a release. Getting the option out there to not poll on print progress is also good to have in the wild for people like you that don't really care about it. I just release the 0.9.24 version, which is where we are at now. @cp2004 I created a new branch 0.9.25 for work on possible CPU utilization fixes.

PR #178 (2020-06-19): 0.9.21
* use threading.timer for print progress status checks and switched to queuing gcode hook and other gcode processing improvements, attempting to resolve #164 
* added error classes for ip address entry errors, resolves #177
closed 2026-10-18 by jneilliii