jneilliii/OctoPrint-TPLinkSmartplug #165: [BUG]: Cannot have two OctoPrint servers running

jarocht, 2020-04-29:
**Describe the bug**
With two OctoPrint servers running, both with the plugin installed and configured, only one of the servers is able to access and control the plugs.

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy two octoprint servers
2. Configure both servers with the plugin
3. Configure one plug for each plugin/server
4. When both servers are running, only one server will be able to connect to its plug

The server that functions as expected is able to access both plugs if the second plug is added to the plugin.

**Expected behavior**
Each server/plugin should be able to access its own plug

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: N/A - same bevahior on Mac/Windows
 - Browser N/A - same bevahior on Firefox/Chrome
 - OctoPrint Version 1.4.0
 - Plugin -- TP-Link Smartplug (0.9.19) 

**Additional context**
HS100 plugs

jneilliii, 2020-04-29:
So I have multiple octoprint servers running on different pis and windows, all with their own plug/config. Do you mean multiple instances running in the same server/pi?

jarocht, 2020-04-29:
These are entirely separate Pis, separate IP addresses and hostnames.

jneilliii, 2020-04-29:
Then I will need additional logs, because this is not expected. Enable debug logging on the non-functional server and restart OctoPrint. Once restarted force a refresh of your browser (shift + F5 in chrome, or clear your browsing history/cache) and load the webpage. Check the developer tools (F12) for errors on the console tab. Then try to control the plug. If it's a no go then grab octoprint.log and plugin_tplinksmartplug_debug.log from the logging section of octoprint, hash out any personal identificable information in them and drag them into a comment here to upload so I can evaluate.

jarocht, 2020-04-29:
>plugin_tplinksmartplug_debug.log 
[2020-04-29 17:42:21,786] DEBUG: 
[2020-04-29 17:42:21,787] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-04-29 17:42:22,055] DEBUG: ImmutableMultiDict([('checkStatus', u'172.16.2.201')])
[2020-04-29 17:42:22,056] DEBUG: Checking status of 172.16.2.201.
[2020-04-29 17:42:22,056] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-04-29 17:42:22,056] DEBUG: IP 172.16.2.201 is valid.
[2020-04-29 17:42:22,057] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 172.16.2.201
[2020-04-29 17:42:32,067] DEBUG: Could not connect to 172.16.2.201.
[2020-04-29 17:42:32,069] DEBUG: 
[2020-04-29 17:42:32,069] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

> octoprint log
no errors of note

jarocht, 2020-04-29:
firefox kicks out this error on the browser:
>Firefox can’t establish a connection to the server at ws://e3.local/sockjs/719/0jka5i5c/websocket.

but it does it for both servers, the working one and the non-working one.

In my testing I have not been able to flip the 'non-working' to the other server. so I'm focusing on that machine specifically now.

jarocht, 2020-04-29:
This was a networking issue. The pi's were switching between IP addresses for making the request (wifi/etho) and my strict firewall rules were blocking the requests. Disabling the wifi radio entirely resolved the issue.

I applaud your responsiveness! Be well!

closed 2020-04-29 by jarocht