jneilliii/OctoPrint-TPLinkSmartplug #166: [Feature Request]: Control USB port power

jangrewe, 2020-04-30:
**Is your feature request related to a problem? Please describe.**
I'm using your plugin with two HS110 on two of my printers. It works perfectly fine on a Prusa, where everything is powered off correctly, but on my Artillery Sidewinder X1 the TFT still gets power from the USB port, so the board and the display stay on even when the printer has no power.

**Describe the solution you'd like**
It would be great if you could either integrate https://github.com/OutsourcedGuru/OctoPrint-USBControl, or provide identical functionality in your plugin, e.g. a checkbox "disable power on USB port after disconnect" and "enable power on USB port before connect".

The plugin also provides an API, so that's maybe the easiest route to go.

jneilliii, 2020-04-30:
There are a couple of ways to handle this, one of which is using a power blocking usb cable. The other is using system commands to shutdown your pi completely and enabling the timer option to power off the printer. The later obviously requires that both devices use or share a smartplug. Add you mentioned, the usb control plugin exists as well and you could use the system command option to trigger its api commands.

jangrewe, 2020-04-30:
duh, that data-only USB cable was **way** too simple to think of right away... found myself some electrical tape and scissors, now it's behaving as expected.

Thanks! 😄

closed 2020-04-30 by jangrewe

jneilliii, 2020-04-30:
Yeah, there was a discussion about it on the community forum over [here](https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574) that's why I knew about it. I believe there is also a thin print you can use to in effect do the same thing, but sounds like you got the tape to work.

jangrewe, 2020-04-30:
Thanks for that link! I was about to buy a new power supply to get rid of the undervoltage warning, but after reading the post i'm hoping that tape-fix kills two birds with one stone :-)