jneilliii/OctoPrint-TPLinkSmartplug #169: [BUG]: Freeze values Current Power (main page)

technoo10201, 2020-05-09:
Hi ! First of all thanks for your nice work.

I would like to report a freeze "bug" in the main page of octoprint in the div TP-Link smartplug.

When i open the web page of octoprint my values like "**Current power**" & "**Total Consumption**" are not refreshed even if i put Polling Interval = 0 in the settings page...

**To Reproduce**
1. Put Polling Interval = 0 in the settings
2. Refresh your web browser on octoprint page

[image]

3. Then open the div with all info, you will see values are freezed

[image]

When i click on the refresh button, it refresh only once the value and then it is freezed again.

To unfreeze the values : 

I have to put in the settings Polling Interval = 1
Then Save settings
And go again in settings & put Polling Interval = 0
Then Save settings.

Values are not freeze anymore.
I have to do the same steps each time i refresh my octoprint page..

Any tips, bug or solution ?
Thanks in advance

jneilliii, 2020-05-10:
Thanks, I'll update the UI to block that by setting the min to 1. And then handle on save options to filter out those as well.

[image]

jneilliii, 2020-05-10:
It's basically a 0 minute delay in an infinite loop, so setting the minimum to 1 would at least give you a minute to fix any changes necessary.

technoo10201, 2020-05-10:
Actually, i would like to keep the "1 second" refresh to get the reel consumption at real time.
Is it possible ?

Thanks for your work.

jneilliii, 2020-05-10:
I'm not sure I'd recommend 1 sec polling. 1 minute polling might be ok, but based on #164 1 second might be bad.

jneilliii, 2020-05-11:
Also, to be clear. This additional polling will only effect the graphing, the final metadata added to the file for power consumption/cost is based on the `[total kWh at print complete] - [total kWh at print start]` where total kWh is being provided by the plug directly. So if you look at your Kasa app's energy tab for the plug when you start the print you will see Total Consumption, and then at the end of the print check again. Hopefully that makes sense. I still haven't received concrete evidence this is accurate in any way, as I don't know what happens for multi-day prints for example.

closed 2021-01-27 by jneilliii