jneilliii/OctoPrint-TPLinkSmartplug #170: [Feature Request]: Support FIRMWARE_RESTART of Klipper and other terminal commands

spammer23, 2020-05-15:
**Is your feature request related to a problem? Please describe.**
I'm using Klipper and my raspberry stays turned on the whole time.
Before i can start printing i need to connect the printer (auto connect from Plugin) and issue a "FIRMWARE_RESTART" command in the terminal.
First i thougt i can use the "Führen Sie den Systembefehl nach Ein aus" for this but i won't work

**Describe the solution you'd like**
Make a field for terminal commands like for the Systemcalls.

jneilliii, 2020-05-15:
So something like a "gcode after on command"?

spammer23, 2020-05-15:
Im not really sure if it is a gcode in my case, but yes for naming it should be okay.

```Recv: ok
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Recv: ok
[...]
Send: FIRMWARE_RESTART
Recv: ok
Recv: // Klipper state: Disconnect
[...]
Recv: // Klipper state: Ready```

Thats how it looks in terminal

jneilliii, 2020-05-15:
Yeah thanks. Klipper is weird in that aspect, technically from an octoprint perspective that is considered gcode.

zpatten, 2020-08-06:
Would love to see this feature.  Here is a workaround I'm using.

1. Set connect delay to something like 15 seconds.
2. Set a system command after power on to something like 10 seconds:
    `sudo service klipper restart`

Whatever delays you use just make sure the connect is a few seconds after the system command so Klipper has enough time to cycle.

Generally speaking you shouldn't need to issue the firmware restart to the MCU as it is effectively reset when it's powered on; however Klipper needs to be restarted.  For me this clears any "funky" state Klipper will go into after it has lost connection to the MCU from the previous power down.

This seems to work like a charm:
```
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0xac6e9750, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.8.0-626-gcd8f250e FIRMWARE_NAME:Klipper
```

That being said I think what you really need to do is just issue the `RESTART` gcode to Klipper after power on as this has the same effect of the `sudo service klipper restart` system command.

spammer23, 2020-08-18:
Thanks, your workaround works fine 👍 
never figured out which part of klipper is stupid at startup :D

jneilliii, 2021-02-28:
Added with the above commit, you can test by changing the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted.

PR #264 (2021-04-14): 1.0.0
* clear _autostart_file on print done or canceled events to prevent from auto starting a file on next connect.
* fix missing poll_status in init
* resolve idle timeout issues related to plugs powering off with gcode commands, resolves #254
* change gcode trigger logic for off commands to account for printer buffer, resolves #258
* change index values for strip devices to match labels and the Kasa app, resolves #243
* add sending gcode commands before off and after on, resolves #160, resolves #170
* turn on only if we are closed or errored state with upload event, thanks to @ayufan
* adde [... cut]
closed 2021-02-28 by jneilliii