jneilliii/OctoPrint-TPLinkSmartplug #171: [BUG]: system powers down when Pi is off for longer than idle timer.

lokimonster82, 2020-05-22:
Firstly, great work. I love this plugin, but i do have a minor issue.

I have it set up so that if the printer is idle for 15mins, the Pi will be shut down and then the plug will turn off. The next time i turn my plug on, the Pi boots up and straight away triggers the idle behaviour, shutting down the Pi and turning off the plug. If I then turn the plug back on, all works ok. Its just a little annoying that i have to turn off the plug, wait to it to start then shut down before starting it again to use it.

**Log:**

This log shows the shutdown from the previous day and first start up following day.

[2020-05-21 18:07:57,702] DEBUG: Idle timeout reached after 15 minute(s). Turning heaters off prior to powering off plugs.
[2020-05-21 18:07:57,704] DEBUG: Heater bed already off.
[2020-05-21 18:07:57,705] DEBUG: Heater tool0 already off.
[2020-05-21 18:07:57,706] DEBUG: Heater tool0 = 36.0C
[2020-05-21 18:07:57,706] DEBUG: Heaters below temperature.
[2020-05-21 18:07:57,707] DEBUG: Checking timelapse status before shutting off power...
[2020-05-21 18:07:57,708] DEBUG: Starting abort power off timer.
[2020-05-21 18:08:27,718] DEBUG: Automatically powering off enabled plugs.
[2020-05-21 18:08:27,722] DEBUG: Turning off 192.168.1.223.
[2020-05-21 18:08:27,725] DEBUG: {'autoConnect': False, 'ip': '192.168.1.223', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': True, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Printer', 'warnPrinting': False, 'thermal_runaway': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': '60', 'sysRunCmdOff': 'sudo shutdown -h now', 'sysRunCmdOn': '', 'autoDisconnect': False, 'automaticShutdownEnabled': True, 'useCountdownRules': True, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': '15', 'event_on_disconnect': False, 'currentState': 'unknown', 'sysCmdOn': False, 'sysCmdOffDelay': 0}
[2020-05-21 18:08:27,726] DEBUG: IP 192.168.1.223 is valid.
[2020-05-21 18:08:27,727] DEBUG: Sending command {u'count_down': {u'delete_all_rules': None}} to 192.168.1.223
[2020-05-21 18:08:27,963] DEBUG: «  2â"count_down":{"delete_all_rules":{"err_code":0}}}
[2020-05-21 18:08:27,978] DEBUG: IP 192.168.1.223 is valid.
[2020-05-21 18:08:27,979] DEBUG: Sending command {u'count_down': {u'add_rule': {u'delay': 60, u'enable': 1, u'name': u'turn off', u'act': 0}}} to 192.168.1.223
[2020-05-21 18:08:28,188] DEBUG: «  R"count_down":{"add_rule":{"id":"598659FAE5C9CB95A87C24BEDDF954F5","err_code":0}}}
[2020-05-21 18:08:28,200] DEBUG: 0
[2020-05-21 18:08:28,206] DEBUG: Checking status of 192.168.1.223.
[2020-05-21 18:08:28,208] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-05-21 18:08:28,209] DEBUG: IP 192.168.1.223 is valid.
[2020-05-21 18:08:28,210] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.223
[2020-05-21 18:08:28,307] DEBUG: « ^"system":{"get_sysinfo":{"sw_ver":"1.0.3 Build 191111 Rel.144915","hw_ver":"4.1","model":"HS100(UK)","deviceId":"80063388A8DC52F555E9D739B329601E1CD5D420","oemId":"BDE8C5BFBD2235EFFDD4DC1257333B68","hwId":"00105B47562710D8AA66EF95D3FDCE49","rssi":-67,"longitude_i":-18649,"latitude_i":521683,"alias":"3DPrinter","status":"new","mic_type":"IOT.SMARTPLUGSWITCH","feature":"TIM","mac":"B0:95:75:5F:00:01","updating":0,"led_off":0,"relay_state":1,"on_time":34602,"active_mode":"count_down","icon_hash":"","dev_name":"Smart Wi-Fi Plug","next_action":{"type":2,"id":"598659FAE5C9CB95A87C24BEDDF954F5","schd_sec":65367,"action":0},"ntc_state":0,"err_code":0}}}
[2020-05-21 18:08:28,495] DEBUG: TIM
[2020-05-21 18:08:48,212] DEBUG: abortTimeout: 30
[2020-05-21 18:08:48,214] DEBUG: powerOffWhenIdle: True
[2020-05-21 18:08:48,215] DEBUG: idleTimeout: 15
[2020-05-21 18:08:48,216] DEBUG: idleIgnoreCommands: M105
[2020-05-21 18:08:48,217] DEBUG: idleTimeoutWaitTemp: 50
[2020-05-22 10:12:39,917] DEBUG: Checking status of 192.168.1.223.
[2020-05-22 10:12:39,927] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-05-22 10:12:39,961] DEBUG: IP 192.168.1.223 is valid.
[2020-05-22 10:12:39,960] DEBUG: Idle timeout reached after 15 minute(s). Turning heaters off prior to powering off plugs.

[image]

[image]

Thank you

jneilliii, 2020-05-22:
@lokimonster82 do you get the abort pop-up when this happens?

lokimonster82, 2020-05-22:
No, if I connect straight to the web interface, its hardly had time to even load before shutting down. no abort shutdown alert pops up.

jneilliii, 2020-05-22:
Thanks. I plan on looking into this over the weekend. There happened to be another user on discord mention the exact same problem.

snecklifter, 2020-06-10:
@jneilliii thanks for awesome plugin! Looks like I'm seeing the same? I powered on this morning and it stayed on for around a couple of minutes then powered off.

My Pi is powered from my printer, no idea if that makes a difference but it wont have any power going to it once the printer is off, although Pi has no RTC so...?

It looks like its immediately triggering the idle timeout on power on?

I've pasted some debug here (sysinfo lines removed for privacy):

```
[2020-06-09 19:51:03,375] DEBUG: Could not connect to 192.168.0.30.
[2020-06-09 19:51:03,398] DEBUG: None
[2020-06-09 19:51:03,399] DEBUG: powering off due to Disconnected event.
[2020-06-09 19:34:30,822] DEBUG: abortTimeout: 30
[2020-06-09 19:34:30,823] DEBUG: powerOffWhenIdle: True
[2020-06-09 19:34:30,823] DEBUG: idleTimeout: 15
[2020-06-09 19:34:30,824] DEBUG: idleIgnoreCommands: M105
[2020-06-09 19:34:30,824] DEBUG: idleTimeoutWaitTemp: 50
[2020-06-10 09:41:49,906] DEBUG: Idle timeout reached after 15 minute(s). Turning heaters off prior to powering off plugs.
[2020-06-10 09:41:49,906] DEBUG: Heater bed already off.
[2020-06-10 09:41:49,907] DEBUG: Heater tool0 already off.
[2020-06-10 09:41:49,907] DEBUG: Heater tool0 = 19.37C
[2020-06-10 09:41:49,908] DEBUG: Heaters below temperature.
[2020-06-10 09:41:49,908] DEBUG: Checking timelapse status before shutting off power...
[2020-06-10 09:41:49,918] DEBUG: Starting abort power off timer.
[2020-06-10 09:41:49,926] DEBUG: Checking status of 192.168.0.30.
[2020-06-10 09:41:49,932] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-10 09:41:49,934] DEBUG: IP 192.168.0.30 is valid.
[2020-06-10 09:41:49,935] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.0.30
[2020-06-10 09:41:50,530] DEBUG: TIM
[2020-06-10 09:42:19,978] DEBUG: Automatically powering off enabled plugs.
[2020-06-10 09:42:19,979] DEBUG: Turning off 192.168.0.30.
[2020-06-10 09:42:19,980] DEBUG: {'autoConnect': True, 'ip': '192.168.0.30', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': True, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Power', 'warnPrinting': True, 'thermal_runaway': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': True, 'countdownOffDelay': '30', 'sysRunCmdOff': 'sudo shutdown', 'sysRunCmdOn': '', 'autoDisconnect': True, 'automaticShutdownEnabled': True, 'useCountdownRules': True, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': 10, 'event_on_disconnect': False, 'currentState': 'on', 'sysCmdOn': False, 'sysCmdOffDelay': '5'}
[2020-06-10 09:42:19,980] DEBUG: IP 192.168.0.30 is valid.
[2020-06-10 09:42:19,981] DEBUG: Sending command {u'count_down': {u'delete_all_rules': None}} to 192.168.0.30
[2020-06-10 09:42:20,160] DEBUG: «^@^@2â"count_down":{"delete_all_rules":{"err_code":0}}}
[2020-06-10 09:42:20,168] DEBUG: IP 192.168.0.30 is valid.
[2020-06-10 09:42:20,168] DEBUG: Sending command {u'count_down': {u'add_rule': {u'delay': 30, u'enable': 1, u'name': u'turn off', u'act': 0}}} to 192.168.0.30
[2020-06-10 09:42:20,353] DEBUG: «^@^@R<U+0082>"count_down":{"add_rule":{"id":"D7B7FF370CD8FAD93298C5959FFF04AF","err_code":0}}}
[2020-06-10 09:42:20,375] DEBUG: 0
[2020-06-10 09:42:20,375] DEBUG: Checking status of 192.168.0.30.
[2020-06-10 09:42:20,376] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-10 09:42:20,377] DEBUG: IP 192.168.0.30 is valid.
[2020-06-10 09:42:20,378] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.0.30
[2020-06-10 09:42:20,380] DEBUG: powering off due to Disconnected event.
[2020-06-10 09:42:20,473] DEBUG: TIM
[2020-06-10 09:42:29,129] DEBUG: abortTimeout: 30
[2020-06-10 09:42:29,131] DEBUG: powerOffWhenIdle: True
[2020-06-10 09:42:29,131] DEBUG: idleTimeout: 15
[2020-06-10 09:42:29,132] DEBUG: idleIgnoreCommands: M105
[2020-06-10 09:42:29,132] DEBUG: idleTimeoutWaitTemp: 50
```

snecklifter, 2020-06-11:
@jneilliii sorry, spoke too soon. Even with disconnect monitoring disabled it seems the idle timeout is immediately triggered:

```
[2020-06-11 09:12:56,626] DEBUG: Checking status of 192.168.0.30.
[2020-06-11 09:12:56,628] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-11 09:12:56,629] DEBUG: IP 192.168.0.30 is valid.
[2020-06-11 09:12:56,631] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.0.30
[2020-06-11 09:12:56,635] DEBUG: Idle timeout reached after 15 minute(s). Turning heaters off prior to powering off plugs.
[2020-06-11 09:12:56,640] DEBUG: Heater bed already off.
[2020-06-11 09:12:56,640] DEBUG: Heater tool0 already off.
[2020-06-11 09:12:56,642] DEBUG: Heater tool0 = 18.12C
[2020-06-11 09:12:56,643] DEBUG: Heaters below temperature.
[2020-06-11 09:12:56,643] DEBUG: Checking timelapse status before shutting off power...
[2020-06-11 09:12:56,645] DEBUG: Starting abort power off timer.

```

snecklifter, 2020-06-11:
Is it possible that the idle timeout is still running when the system gets booted up?
Or does the code need a timeout reset when starting up?

jneilliii, 2020-06-11:
If I remember correctly the timeout is initialized on start up.

snecklifter, 2020-06-11:
If I turn it on again after it turns off it stays on which makes me suspect an ntp thing.

lokimonster82, 2020-06-11:
I've done some checking this afternoon. I don't have 'Disconnect Event monitoring' enabled, so its not the the cause of my issue.

If i turn my printer on (after powering off due to idle), once the Pi has powered on (connected to the plug, not the printer) it almost immediately shuts down and then a minute later the plug turns off. If i power the plug on straight away the Pi boots on and stays on for the idle timeout (15m).

If manually shut down the Pi and plug, and turn it back on (within 15m) then the Pi stays on.

jneilliii, 2020-06-11:
Thanks guys, I'll try to do some extensive testing this weekend, especially since a sponsor is involved in the discussion....thanks @snecklifter.

snecklifter, 2020-06-11:
> Thanks guys, I'll try to do some extensive testing this weekend, especially since a sponsor is involved in the discussion....thanks @snecklifter.

Not at all, thank you! Please @ me if you need any patches testing. Am afraid python is not one of my strengths.

jneilliii, 2020-06-11:
lol, not really mine either obviously...I didn't start Python programming until I started writing OctoPrint plugins and it snow-balled.

jneilliii, 2020-06-11:
So I've moved some things around to hopefully help out with this situation by resetting the idle timer on client connection. This means when you load the web interface it should restart the timer again. If you don't mind testing you can install using the URL below in plugin manager. Let me know if that resolves the idle. The one thing I'm not sure of yet @lokimonster82 is the countdown timers. I know you're using those to shutdown your pi cleanly prior to powering off the plug, so I'm curious if those countdown timers might be causing the larger problem. I'm curious if you could check the Kasa app during this 15 minute idle period and let me know if that is possibly causing the problem?

https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.21.zip

snecklifter, 2020-06-18:
@jneilliii sorry for delay, was on holiday, just installed and will test and update

lokimonster82, 2020-06-18:
@jneilliii i've installed your update and it looks to have fixed the issue. great work.

lokimonster82, 2020-06-25:
@jneilliii sorry, i spoke to soon. I've not had chance to do much printing the past few weeks but can confirm the same issue does still exist. I've changed my 'abort power off timeout out' to 300 seconds, so when the system starts up i get around 270 seconds to abort the timeout.

snecklifter, 2020-06-30:
@jneilliii to confirm what @lokimonster82 has said, I'm also seeing this still with 0.9.23:

```[2020-06-29 20:36:14,787] DEBUG: «^@^B<U+008E>\"system":{"get_sysinfo":{"sw_ver":"1.0.3 Build 191111 Rel.144915","hw_ver":"4.1","model":"HS100(UK)","deviceId":"8006F26897FC[2020-06-29 20:32:58,209] DEBUG: abortTimeout: 30
[2020-06-29 20:32:58,209] DEBUG: powerOffWhenIdle: True
[2020-06-29 20:32:58,210] DEBUG: idleTimeout: 15
[2020-06-29 20:32:58,210] DEBUG: idleIgnoreCommands: M105
[2020-06-29 20:32:58,211] DEBUG: idleTimeoutWaitTemp: 50
[2020-06-30 14:54:26,416] DEBUG: Idle timeout reached after 15 minute(s). Turning heaters off prior to powering off plugs.
[2020-06-30 14:54:26,417] DEBUG: Heater bed already off.
[2020-06-30 14:54:26,417] DEBUG: Heater tool0 already off.
[2020-06-30 14:54:26,418] DEBUG: Checking status of 192.168.0.30.
[2020-06-30 14:54:26,419] DEBUG: Heater tool0 = 21.87C
[2020-06-30 14:54:26,420] DEBUG: Heaters below temperature.
[2020-06-30 14:54:26,421] DEBUG: Checking timelapse status before shutting off power...
[2020-06-30 14:54:26,420] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-06-30 14:54:26,422] DEBUG: IP 192.168.0.30 is valid.
[2020-06-30 14:54:26,422] DEBUG: Starting abort power off timer.
```

snecklifter, 2020-06-30:
@jneilliii I have no idea what I'm talking about but is it worth putting in a sleep function on _start_idle_timer, e.g.

```time.sleep(self.idleTimeout)```

or something?

jneilliii, 2020-06-30:
Yeah @snecklifter that is what is built into the RepeatableTimer function I'm utilizing [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/ca8bcc6ab3cddac7f5010ef41e94c9659985c442/octoprint_tplinksmartplug/__init__.py#L634).

THX2112, 2020-07-19:
I'm having this problem as well.

jneilliii, 2020-07-19:
Just to confirm in all your cases you are shutting the pi down? That's the one thing that I never do because the OctoPrint Connection plugin in Cura I don't believe works if the pi is powered off and I use it to manage my power on automatically on upload and auto-print. Then the idle timeout occurs later and powers off the printer.

I could see where this could be problematic because if you don't immediately connect and start a print (or get the temp above threshold) then the idle is getting triggered, which is then shutting down your pi again?

THX2112, 2020-07-20:
No. Octoprint shuts itself down soon after the Pi boots up, unless I cancel the countdown. Exactly the same as https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/189

That issue is marked as closed and points to here.

I tried all of the things mentioned in this thread but the behaviour is the same.

jneilliii, 2020-07-20:
@THX2112 can you please share screenshots of your settings too? Is your Pi powered from your printer's PSU?

lokimonster82, 2020-07-20:
@jneilliii I use the idle timer cmd function to shutdown down the Pi, wait 1 min then turn off the printer. When i power on the printer again, it comes up with the abort shut down prompt. I've increased this timer to 5mins to give me plenty of time to abort the shut down when powering on.

LarryVand, 2020-08-13:
Thanks for pointing me to this thread. Until now, I thought the on idle pop-up was getting triggered only the first time I turned on the TP-Link plug in the morning. However after reading this thread, I can see that the issue is triggered if you let the TP-Link plug stay OFF longer than the idle timeout setting (in my case 30min). So if everything is turned off for longer than 30min, the next time everything is powered up, the idle timeout will be triggered and the pop-up will appear. I currently circumvent this by putting a longer time for the "abort power off timeout" setting (like 75sec), which gives enough time for everything to boot up and for me to see the pop-up on the screen and abort it.

THX2112, 2020-08-13:
> 
> 
> @THX2112 can you please share screenshots of your settings too? Is your Pi powered from your printer's PSU?

The Pi is powered separately.

I don't have access to the machine now for a snapshot, but I tried everything suggested in this thread. The only thing I can do is increase the timeout so I have a chance of catching it before it powers down.

snecklifter, 2020-08-17:
@jneilliii any chance this can be revisited? Happy to send you logs or whatever is required?

jneilliii, 2020-08-17:
I will get to it eventually.

Tarzanelly, 2020-09-08:
I opened a new thread on the same issue and I'm sorry for this. I did everything in this thread but i still have the same issue. My Ender 3 turns off few seconds after the extruder starts its pre-heating process. Are you working on some solutions? Did I miss something? I tried also putting 300 mins of timeout on powering off on idle but it's still the same

jneilliii, 2020-09-08:
Your issue I think is slightly different, I commented over there.

jneilliii, 2020-11-14:
I think I have this issue resolved after a little logic assistance from @snecklifter by making sure the uptime is not shorter than the idle time configured in settings. This has the small caveat of a potential 1 minute additional length in time to your idle time but figured that's not a huge deal for most users. If you want to test it out you can copy/paste the URL below into plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/archive/0.9.26rc2.zip
```

lokimonster82, 2020-11-17:
jneillii, with the above release I find that after the timer completes the plug is not turned off.

snecklifter, 2020-11-17:
@lokimonster82 we are hitting some plug firmware upgrades so not sure if that's related. There's also some additional debug output in the latest code so if you post the output of that it would help.

snecklifter, 2020-11-17:
@lokimonster82 if you want to work around this in hardware (and I'm assuming its the same problem as I was having) you could consider fitting an RTC which should keep time correctly and prevent the jump that triggers the immediate power down. Alternatively (not tested) you could try hooking the Pi up to wired ethernet rather than wireless (not sure if you are doing this anyway) which might be faster and therefore time sync occurs before Octoprint starts.

lokimonster82, 2020-11-20:
@snecklifter below is a snipit from the log, it appears the plugin can no longer connect to the plug. The log is full of "Could not connect to 192.168.1.223".

For reference the plug is a HS100, firmware V1.1.0.

[2020-11-20 05:42:25,445] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-11-20 05:57:25,448] DEBUG: Checking status of 192.168.1.223.
[2020-11-20 05:57:25,450] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-11-20 05:57:25,450] DEBUG: IP 192.168.1.223 is valid.
[2020-11-20 05:57:25,451] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.223
[2020-11-20 05:57:25,526] DEBUG: Could not connect to 192.168.1.223.
[2020-11-20 05:57:25,527] DEBUG: 
[2020-11-20 05:57:25,528] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-11-20 06:12:25,532] DEBUG: Checking status of 192.168.1.223.
[2020-11-20 06:12:25,534] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-11-20 06:12:25,534] DEBUG: IP 192.168.1.223 is valid.
[2020-11-20 06:12:25,535] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.223
[2020-11-20 06:12:25,631] DEBUG: Could not connect to 192.168.1.223.
[2020-11-20 06:12:25,632] DEBUG: 
[2020-11-20 06:12:25,634] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-11-20 06:27:25,638] DEBUG: Checking status of 192.168.1.223.
[2020-11-20 06:27:25,639] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-11-20 06:27:25,640] DEBUG: IP 192.168.1.223 is valid.
[2020-11-20 06:27:25,641] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.223
[2020-11-20 06:27:25,769] DEBUG: Could not connect to 192.168.1.223.
[2020-11-20 06:27:25,770] DEBUG: 
[2020-11-20 06:27:25,771] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-11-20 06:39:42,208] DEBUG: Idle timeout reached after 15 minute(s). Turning heaters off prior to powering off plugs.
[2020-11-20 06:39:42,210] DEBUG: Heater bed already off.
[2020-11-20 06:39:42,211] DEBUG: Heater tool0 already off.
[2020-11-20 06:39:42,212] DEBUG: Heater tool0 = 28.04C
[2020-11-20 06:39:42,213] DEBUG: Heaters below temperature.
[2020-11-20 06:39:42,214] DEBUG: Checking timelapse status before shutting off power...
[2020-11-20 06:39:42,215] DEBUG: Starting abort power off timer.
[2020-11-20 06:42:25,776] DEBUG: Checking status of 192.168.1.223.
[2020-11-20 06:42:25,778] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-11-20 06:42:25,779] DEBUG: IP 192.168.1.223 is valid.
[2020-11-20 06:42:25,779] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.223
[2020-11-20 06:42:25,881] DEBUG: Could not connect to 192.168.1.223.
[2020-11-20 06:42:25,882] DEBUG: 
[2020-11-20 06:42:25,883] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}
[2020-11-20 06:44:42,306] DEBUG: Automatically powering off enabled plugs.
[2020-11-20 06:44:42,311] DEBUG: Turning off 192.168.1.223.
[2020-11-20 06:44:42,314] DEBUG: {'autoConnect': False, 'ip': '192.168.1.223', 'btnColor': '#808080', 'autoDisconnectDelay': 0, 'sysCmdOff': True, 'gcodeEnabled': False, 'gcodeOnDelay': 0, 'emeter': {'get_realtime': {}}, 'countdownOnDelay': 1, 'label': 'Printer', 'warnPrinting': False, 'thermal_runaway': True, 'gcodeOffDelay': 0, 'sysCmdOnDelay': 0, 'event_on_error': False, 'countdownOffDelay': '60', 'sysRunCmdOff': 'sudo shutdown -h now', 'sysRunCmdOn': '', 'autoDisconnect': False, 'automaticShutdownEnabled': True, 'event_on_upload': False, 'useCountdownRules': True, 'event_on_startup': False, 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': '15', 'event_on_disconnect': False, 'sysCmdOn': False, 'currentState': 'unknown', 'sysCmdOffDelay': 0}
[2020-11-20 06:44:42,316] DEBUG: IP 192.168.1.223 is valid.
[2020-11-20 06:44:42,317] DEBUG: Sending command {u'count_down': {u'delete_all_rules': None}} to 192.168.1.223
[2020-11-20 06:44:42,322] DEBUG: Could not connect to 192.168.1.223.
[2020-11-20 06:44:42,324] DEBUG: IP 192.168.1.223 is valid.
[2020-11-20 06:44:42,325] DEBUG: Sending command {u'count_down': {u'add_rule': {u'delay': 60, u'enable': 1, u'name': u'turn off', u'act': 0}}} to 192.168.1.223
[2020-11-20 06:44:42,332] DEBUG: Could not connect to 192.168.1.223.
[2020-11-20 06:44:42,334] DEBUG: None
[2020-11-20 08:09:58,519] DEBUG: abortTimeout: 300
[2020-11-20 08:09:58,524] DEBUG: powerOffWhenIdle: True
[2020-11-20 08:09:58,525] DEBUG: idleTimeout: 15
[2020-11-20 08:09:58,526] DEBUG: idleIgnoreCommands: M105
[2020-11-20 08:09:58,527] DEBUG: idleTimeoutWaitTemp: 50
[2020-11-20 08:10:20,302] DEBUG: ImmutableMultiDict([('checkStatus', u'192.168.1.223')])
[2020-11-20 08:10:20,303] DEBUG: Checking status of 192.168.1.223.
[2020-11-20 08:10:20,303] DEBUG: {'system': {'get_sysinfo': {}}}
[2020-11-20 08:10:20,304] DEBUG: IP 192.168.1.223 is valid.
[2020-11-20 08:10:20,304] DEBUG: Sending command {'system': {'get_sysinfo': {}}} to 192.168.1.223
[2020-11-20 08:10:20,571] DEBUG: Could not connect to 192.168.1.223.
[2020-11-20 08:10:20,572] DEBUG: 
[2020-11-20 08:10:20,573] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

snecklifter, 2020-11-20:
> @snecklifter below is a snipit from the log, it appears the plugin can no longer connect to the plug. 

Are you in the UK? See #218 

Even if not, can you check the app and confirm the firmware version on the plug you are running.

lokimonster82, 2020-11-20:
Yes, I'm in the UK and that is definitely my issue. I'm happy to close this ticket based on what I'm seeing.

closed 2020-11-20 by lokimonster82