jneilliii/OctoPrint-TPLinkSmartplug #174: [BUG]: Idle Target Temp shutdown not working

n2jwc, 2020-05-22:
Idle timer is working however the Idle Target Temp is not shutting down the printer.  I have screen shot and my settings and I have already turned on debug logging however not sure where it is saved.

[image]

jneilliii, 2020-05-22:
I'm assuming your triplicate post is due to github's recent issues? Could you please verify if restarting octoprint has any effect. If not please post your plugin_tplinksmartplug_debug.log file.

n2jwc, 2020-05-22:
Yes the triple post was due to github.  I will reboot.  Where can I find the debug log?

jneilliii, 2020-05-22:
in the logging section of octoprint's settings. then you can drag and drop it here in the comment you reply with.

n2jwc, 2020-05-22:
I just literally though that and was SFTPing it off.

[plugin_tplinksmartplug_debug.log.2020-05-20.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4670348/plugin_tplinksmartplug_debug.log.2020-05-20.txt)
[plugin_tplinksmartplug_debug.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4670349/plugin_tplinksmartplug_debug.txt)

There you go.  This is an Ender3 with stock board and firmware.

jneilliii, 2020-05-22:
Try turning off disconnect event monitoring and restart octoprint and let me know if that make a difference. What I'm seeing is the printer is getting powered off because of disconnect events, which may actually be bypassing the logic in the idle timeout.

n2jwc, 2020-05-23:
Hi.  I tried that and nothing changed.  I did not print anything I ran the HE and Bed to 50 and then turned them off to observe.  Nothing.  Just to be sure I tested the Thermal protection and that worked.

[2020-05-22 21:58:17,167] DEBUG: Resetting idle timer during print progress (0)...
[2020-05-22 21:59:00,320] DEBUG: Print cancelled, reseting job_power to 0
[2020-05-22 22:01:40,474] DEBUG: Max bed temp reached, shutting off plugs.
[2020-05-22 22:01:40,477] DEBUG: Turning off 192.168.244.84.

New log attached.

[plugin_tplinksmartplug_debug.log.txt](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/files/4671859/plugin_tplinksmartplug_debug.log.txt)

zhanbox, 2020-08-05:
I have the same issue, and the log shows that the plugin is not reading the actual latest temp back:

`[2020-08-05 22:10:47,843] DEBUG: Idle timeout reached after 10 minute(s). Turning heaters off prior to powering off plugs.
[2020-08-05 22:10:47,844] DEBUG: Heater bed already off.
[2020-08-05 22:10:47,845] DEBUG: Heater tool0 already off.
[2020-08-05 22:10:47,845] DEBUG: Heater tool0 = 192.73C
[2020-08-05 22:10:47,846] DEBUG: Waiting for heaters(tool0) before shutting power off...
[2020-08-05 22:10:52,850] DEBUG: Heater tool0 = 192.73C
[2020-08-05 22:10:52,851] DEBUG: Waiting for heaters(tool0) before shutting power off...
[2020-08-05 22:10:57,859] DEBUG: Heater tool0 = 192.73C
[2020-08-05 22:10:57,859] DEBUG: Waiting for heaters(tool0) before shutting power off...`

While at the same time, the terminal is showing correct temp info:

`Recv:   TT::26.3426.34  //0.000.00  BB::27.7327.73  //0.000.00  @@::00  BB@@::00
Recv: `

jneilliii, 2020-08-06:
This might be related to some random firmware temperature reporting issue that I'm not taking into account. @n2jwc does your terminal also report the temperature as `TT::` like @zhanbox report above?

khanhntran, 2020-09-14:
Just chiming in - The reported double temp issue is something that was introduced with the ender 3 pro v2 and creality cr6 se. 

Reference: https://plugins.octoprint.org/plugins/ender3v2tempfix/

n2jwc, 2020-09-14:
> This might be related to some random firmware temperature reporting issue that I'm not taking into account. @n2jwc does your terminal also report the temperature as `TT::` like @zhanbox report above?

HI. my printer is offline and packed as we are moving to a new house so I am unable to provide any information for a couple weeks.

jneilliii, 2020-09-14:
Correct me if I'm wrong @khanhntran I think if that other plugin is installed then my plugin works fine.

khanhntran, 2020-09-14:
Unfortunately, I'm not quite sure.
I actually had to open an issue on that plugin's repo. I installed the fix plugin and my temperature reading didn't change at all.
Link - https://github.com/SimplyPrint/OctoPrint-Ender3V2TempFix/issues/6

zhanbox, 2020-09-14:
I can confirm that after I installed that ender 3 v2 temp fix plugin, this plugin is automatically working. 

I did this about a week ago, I thought I already reported here and somehow my post didn't go out :)

jneilliii, 2020-09-14:
Great news, thanks @zhanbox. Hopefully that is also the issue and fix for @n2jwc. Good luck moving.

closed 2020-11-14 by jneilliii